zephyrproject-rtos/zephyr #77069: Deprecated APIs/options to be removed by 4.1

aescolar, 2024-08-14:
This issue tracks APIs/options which are meant to be removed by the 4.1 release
(after they have been deprecated for 2 releases or more)

## 4.1

* [ ] config ARM64_ENABLE_FRAME_POINTER @ycsin c87dc64
* [ ] config LEGACY_MULTI_LEVEL_TABLE_GENERATION @ycsin c5f5b96
* [ ] config DISABLE_SSBD , ENABLE_EXTENDED_IBRS @dcpleung 4c031e2
* [ ] config COUNTER_MCUX_KINETIS_LPTMR @decsny eae316ebc9ffad98b93291ebaca71a15e74fed1d
* [ ] config ETH_MCUX  @decsny fdbba0341ccc5af234cdc5a5f328fe291c1676ec
* [ ] config MCUX_KINETIS_LPTMR @decsny  90986e7e22f42a45e25da84ef3e293f4171f9886
* [ ] config MP_NUM_CPUS (6e95bdeca6479876366d24bbbff576f0d0587a01) <= 4.1
* [ ] config CPP_STATIC_INIT_GNU @aescolar 6e977ae2d54c4f5443f752ce3a88f22043dcbf07
* [ ] lib/posix/options/Kconfig.deprecated @cfriedt
* [ ] config SOC_ANDES_V5_L2C wtlee 5db259010604e989dcbbf8b891db2a65af74a88d
* [ ] config GPIO_AS_PINRESET <= 4.1
* [ ] config NFCT_PINS_AS_GPIOS <= 4.1
* [ ] config NET_BUF_DATA_POOL_SIZE @rlubos 8e2722e1cae6d7a2c65d26d4393e1bcf6550cd40
* [ ] config NET_TCP_ACK_TIMEOUT	@rlubos f211cd6345555086307da8c4998783ebe0ec0027
* [ ] config NET_SOCKETS_POSIX_NAMES @jukkar  3eeb0f0c7fac4546b5592608dc84ffb723cd44b5
* [ ] config PM_DEVICE_RUNTIME_EXCLUSIVE @ceolin 94af630b2249babd70ec1177d4c48c35ef25a1bf
* [ ] __deprecated typedef struct arch_esf z_arch_esf_t; @ycsin e54b27b9671447f7a74d6221fc753c6bd22f8f94 4.1
* [ ] BT old HCI driver @jhedberg  4.1
* [ ] BT_LE_ADV_.. @theob-pro  909ad8170c7feaae8c03ae3bc4aeafaed37d2cfa 4.1
* [ ] #define CAN_MAX_STD_ID / CAN_MAX_EXT_ID @henrikbrixandersen c6e4748deb41698edf863512172a57a4c82674d0 4.1
* [ ] can_get_min_bitrate / can_get_max_bitrate @henrikbrixandersen  8eded2f76b7e92b5c4d955c6536fd365ab40809c 4.1
* [ ] can_calc_prescaler @henrikbrixandersen 5f99e771feca8b34171796f9e83e29e80d54524d 4.1
* [ ] __deprecated void z_pm_save_idle_exit(void); @ceolin 4d85f3d91cb9e37d52e770fbf266e110aa90c296 4.1
* [ ] Global CSTD property @pdgendt  86b721c45ee7f9348e4021b9df008135a3c7a80d
* [ ] All HWMv1 board aliases in deprecated.cmake @57300 i 792be769a23487fa0ed2469f6928b8c7ce73525f (and others)
* [ ] Silicon Labs starter kits board alias in deprecated.cmake @asmellby 45e49aaca2a5eaa62186b0cc89bd97f974fb3d5c

Deprecation guidelines:
https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#deprecated

Related:
* https://github.com/zephyrproject-rtos/zephyr/issues/76943
*

fabiobaltieri, 2025-03-06:
Heads up: quite a few of these have not been removed, if you have your name on one feel free to queue up a PR for 4.2

fabiobaltieri, 2025-03-06:
Actually I'll move the missing one to the new bug and close this.

closed 2025-03-06 by fabiobaltieri