zephyrproject-rtos/zephyr #77070: Deprecated APIs/options to be removed by 4.2

aescolar, 2024-08-14:
This issue tracks APIs/options which are meant to be removed by the *4.2* release
(after they have been deprecated for 2 releases or more)

# 4.2

* [ ] config SOC_DCDC_NRF52X , SOC_DCDC_NRF52X_HV @gmarull   aaacd682cc4cc6b0106b2de8fe6a963584e4ceff
* [ ] config SOC_DCDC_NRF53X_APP, SOC_DCDC_NRF53X_NET, SOC_DCDC_NRF53X_HV @gmarull 1eddb047f98d1afe1d31ca7bcaba87a3bbba3e79
* [ ] config BOARD_ENABLE_CPUNET	@gmarull 4c84a6f658b65557bb23719e23461a9785809e42
* [ ] adafruit feather nrf52840 express board alias in deprecated.cmake @jacobw 5e789c3ae0c7cac997d03ab73b778a334e5186a0

Deprecation guidelines:
https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#deprecated

Related:
* https://github.com/zephyrproject-rtos/zephyr/issues/77069
*

valeriosetti, 2025-06-24:
> [@valeriosetti](https://github.com/valeriosetti) there is an item for the deprecation of TinyCrypt. As far as I can tell this is done now, and want to verify with you. If we're good can you check the box here?

I checked past release notes and the situation about TinyCrypt is the following:
- TC shim driver deprecated in 4.0
   - It is removed in 4.2 by @ceolin in #90113

- TC library deprecated in 4.1
    - It should be removed in 4.3 and indeed there is already a note for this in #80539.
    - I noticed that there are still users of TinyCrypt in Zephyr codebase (tests and BT) so I guess I will need to prepare some PR to remove this usage before 4.3

I think I can update TinyCrypt lines in the description of this issue and #80539 to clarify the scope in both releases

danieldegrasse, 2025-07-22:
Closing out, https://github.com/zephyrproject-rtos/zephyr/issues/80539 now tracks this

closed 2025-07-22 by danieldegrasse