zephyrproject-rtos/zephyr #77092: wifi: hostap: crypto_alt: Remove wrong AES ECB mode

krish2718, 2024-08-14:
**Describe the bug**
MbedTLS crypto wrappers in hostap are using wrong AES mode, ECB instead of CBC, PR https://github.com/zephyrproject-rtos/zephyr/pull/77091 this partially but `crypto_alt` still has ECB implementation for below
* `aes_encrypt/aes_decrypt`
* `aes_128_encrypt_block`

 `des_encrypt` is the only legacy algo that should be using ECB. (used by MSCHAP/MSCHAPv2).

ECB is unsecure, see https://en.wikipedia.org/wiki/Block_cipher_mode_of_operation#Electronic_codebook_(ECB) and shouldn't be used in production (except for DES).

**To Reproduce**
No visible impact, it's just unsecure, this was found from a code review.

**Expected behavior**
One of the secure AES modes like CBC/CTR/CCM should be used.

**Impact**
Security

**Logs and console output**
NA
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: `main`

**Additional context**
NA

fengming-ye, 2024-08-15:
This is a good point. But mbedtls cipher cmac is using ECB in mbedtls_cipher_cmac_starts. So changing to CBC will cause fail in omac1_aes_vector.
DPP may be a good tester for AES crypto.

krish2718, 2024-08-15:
>  But mbedtls cipher cmac is using ECB in mbedtls_cipher_cmac_starts. So changing to CBC will cause fail in omac1_aes_vector.

I have tested the `CRYPTO`  and it seems to work, but haven't tested the `CRYPTO_ALT`. I still wonder why ECB is used, in `mbedtls_cipher_cmac_starts` it checks the type but doesn't seem to be using it for anything? I will try and check with MbedTLS team on this, may be open a ticket.

krish2718, 2024-08-15:
ECB term is a misnomer in MbedTLS, it doesntt refer to the unsecure ECB implementation but it refers to block-cipher based `data` processing, so, this bug can be closed.

See https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/J3VXTNOM2R77V4QXVKD6C6756GTDF7NB/ for details.

closed 2024-08-15 by krish2718