zephyrproject-rtos/zephyr #77107: GNSS doesn't work with LilyGO TTGO T-Beam

jagra26, 2024-08-15:
Hi, I'm trying to port [LilyGO TTGO T-Beam](https://github.com/LilyGO/TTGO-T-Beam/tree/master) to zephyr. I'm using the 3.7.0 release and the process is almost done. But the GPS module, a [NEO-6m](https://content.u-blox.com/sites/default/files/products/documents/NEO-6_DataSheet_%28GPS.G6-HW-09005%29.pdf), isn't working with the [gnss sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/gnss). My steps until now are:

First, I configured the UART pins to be coincident with T-Beam pinout, with RX on GPIO34 and TX on GPIO12, on pinctrl.dtsi:

```
uart1_default: uart1_default {
		group1 {
			pinmux = <UART0_TX_GPIO12>;
			output-high;
		};
		group2 {
			pinmux = <UART0_RX_GPIO34>;
			bias-pull-up;
		};
	};
```

Second, I seted UART on dts, defining NEO-6m like a generic module:

```
&uart1 {
	status = "okay";
	current-speed = <9600>;
	pinctrl-0 = <&uart1_default>;
	pinctrl-names = "default";
	gnssdev: gnss-nmea-generic {
    	    	compatible = "gnss-nmea-generic";
	};
};
```

And, of course, included gnssdev in aliases:

```
aliases {
		led0 = &led_io4;
		uart-0 = &uart0;
		i2c-0 = &i2c0;
		watchdog0 = &wdt0;
		lora0 = &lora0;
		gnss = &gnssdev;
		sw0 = &button_io38;
	};
```

The main.c code and prj.conf are the same of the sample. It builds and runs with no problems, but it doesn't return any data from GPS.
I know it isn't a hardware problem, because I have a functional [arduino code](https://pastebin.com/FyPkse6T).

Maybe i configured it wrong, or NEO-6m isn't compatible with gnss driver, but i don't know  how to check this.

How to proceed?

kartben, 2024-08-15:
Why are you using `UART0_...` in the pinctrl definition? Looks like this is where your problem might be :)

bjarki-andreasen, 2024-08-15:
Also, you need to set the status of your GNSS node and UART node to "okay" :) `status = "okay";`

bjarki-andreasen, 2024-08-15:
> > Also, you need to set the status of your GNSS node and UART node to "okay" :) `status = "okay";`
> 
> Doesn't status default to okay? 🤔 I agree it can be helpful to be explicit but in this case I don't think it's strictly needed?

It seems the macros with STATUS_OKAY do treat missing status as okay :) I just looked at the base.yaml binding :) maybe we should set the default there?

kartben, 2024-08-15:
> It seems the macros with STATUS_OKAY do treat missing status as okay :) I just looked at the base.yaml binding :) maybe we should set the default there?

DT spec says that it's perfectly fine to omit the property altogether, in which case it's treated as if the node is `status="okay"`. Maybe setting a default would make sense (and would have the merit of making all the statuses explicitly show up in the generated zephyr.dts?), but this doesn't change the fact that the "treat missing as okay" behaviour always applies, even for bindings that doesn't extend base.yaml

ASerbinski, 2024-08-15:
Depending on which version of T-Beam you have, there is also a power management chip (AXP192 or AXP2101) that you need to instruct to switch on the power supply to the GPS (also LoRa if you're using that).

Zephyr appears to have a driver for the AXP192 (drivers/regulator/regulator_axp192.c), but not the AXP2101.

jagra26, 2024-08-16:
Hi guys, thanks for responses.  

@kartben 

> Why are you using `UART0_...` in the pinctrl definition? Looks like this is where your problem might be and you should replace with `UART1_...` :)

It's my fault, I've tried various combinations, and ended with UART0. But, you are correct. Now, pinctrl.dtsi looks like:

```
uart1_default: uart1_default {
		group1 {
			pinmux = <UART1_TX_GPIO12>;
			output-high;
		};
		group2 {
			pinmux = <UART1_RX_GPIO34>;
			bias-pull-up;
		};
	};
```

@bjarki-andreasen 

> Also, you need to set the status of your GNSS node and UART node to "okay" :) `status = "okay";`

I've added your suggestion and dts looks like:

```

&uart1 {
	status = "okay";
	current-speed = <9600>;
	pinctrl-0 = <&uart1_default>;
	pinctrl-names = "default";
	gnssdev: gnss-nmea-generic {
		status = "okay";
    	    	compatible = "gnss-nmea-generic";
	};
};
```

But i have the same results, compiles, run and no data.

@ASerbinski 

> Depending on which version of T-Beam you have, there is also a power management chip (AXP192 or AXP2101) that you need to instruct to switch on the power supply to the GPS (also LoRa if you're using [...]

I'm using T-Beam v1.1, and it uses the AXP192. At first, i thinked AXP192 doesn't  make difference because the arduino code dosen't manipulate anything from AXP192.  And GPS LED (see the schematic image) is blinking with the zephyr code runs.
[image]

But, after searching a lot i found in the [lilygo website](https://www.lilygo.cc/products/t-beam-softrf), a QRcode to a [AXP192 configuration](https://github.com/meshtastic/firmware/blob/7bb281d5c569870c27cdd65bcd14507148e4ca9b/src/Power.cpp#L482) (yes, lilygo dosen't help with clear documentation). And, now, I have a north to follow and configure AXP192. I will try this tomorrow and bring news.

Again, thanks for the help @kartben, @bjarki-andreasen and @ASerbinski .

kartben, 2024-08-16:
Great catch, @ASerbinski regarding the AXP192
@jagra26 check out how the couple of M5Stack boards in `boards/M5Stack` configure it, it shouldn't be too hard :) You might need to tweak init level to make sure AXP192 driver kicks in before GNSS driver 

Btw, converting this to a discussion as there is a very good chance that there is no bug :)