zephyrproject-rtos/zephyr #77118: doc: Documentation issue in 'boards/aspeed/ast1030_evb/doc/index'

igudich, 2024-08-15:
**Describe the bug**

Link [Aspeed Zephyr SDK User Guide](https://github.com/AspeedTech-BMC/zephyr/releases/download/v00.01.03/Aspeed_Zephy_SDK_User_Guide_v00.01.03.pdf) is broken.

**Environment**

* Page: `boards/aspeed/ast1030_evb/doc/index`
* Version: main
* SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

ttmut, 2024-08-15:
It seems the document has been removed from their github page.

kartben, 2024-08-17:
@kane-chen-as @stevenlee7189 can you help?

aspeeddylan, 2024-08-19:
Due to Aspeed Technology’s policy, the SDK User Guide is no longer available for download on Aspeed’s GitHub. Please contact your ASPEED representative to obtain the SDK User Guide. I will address this issue by making a commit to update the documentation and adding a description in boards/aspeed/ast1030_evb/doc/index.rst.

kartben, 2024-10-22:
link is no longer a 404 as per https://github.com/zephyrproject-rtos/zephyr/issues/77118#issuecomment-2295564528

closed 2024-10-22 by kartben