zephyrproject-rtos/zephyr #77129: NXP's IMX RT1010 EVK triggers assert on SPI Loopback test when EDMA enabled

ubieda, 2024-08-15:
**Describe the bug**

Running SPI Loopback Test on IMX RT1010 EVK triggers a failed assert when eDMA is enabled (`CONFIG_SPI_MCUX_LPSPI_DMA`).

Git Bisect points to https://github.com/zephyrproject-rtos/zephyr/commit/a5cf757c9ecbaa562a4f31b58bf133a4b978da8a (from  #61311), which indicates this could be a regression.

Running the same testsuite with `CONFIG_SPI_MCUX_LPSPI_DMA=n` works as expected.

**To Reproduce**
Using Upstream Zephyr (main branch)

Build command:
``` console
west build -b mimxrt1010_evk tests/drivers/spi/spi_loopback/ && west flash -r jlink
```

Console output:
``` console
*** Booting Zephyr OS build v3.7.0-645-gedb71f0ba176 ***
Running TESTSUITE spi_loopback
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x202040a0/0x20204080, frame size = 8, DMA enabled
I: SPI test slow config
I: Start complete multiple
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/lib/libc/picolibc/libc-hooks.c:224
E: r0/a1:  0x00000004  r1/a2:  0x000000e0  r2/a3:  0x40184000
E: r3/a4:  0x00000004 r12/ip:  0x60004f91 r14/lr:  0x60004151
E:  xpsr:  0x21000011
E: Faulting instruction address (r15/pc): 0x60008d5c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20208240 (idle)
E: Halting system
*** Booting Zephyr OS build v3.7.0-645-gedb71f0ba176 ***
Running TESTSUITE spi_loopback
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x202040a0/0x20204080, frame size = 8, DMA enabled
I: SPI test slow config
I: Start complete multiple
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/lib/libc/picolibc/libc-hooks.c:224
E: r0/a1:  0x00000004  r1/a2:  0x000000e0  r2/a3:  0x40184000
E: r3/a4:  0x00000004 r12/ip:  0x60004f91 r14/lr:  0x60004151
E:  xpsr:  0x21000011
E: Faulting instruction address (r15/pc): 0x60008d5c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20208240 (idle)
E: Halting system

```

**Expected behavior**
Test should work with or without DMA enabled.

ubieda, 2024-08-15:
CC: @dleach02 @decsny

dleach02, 2024-08-15:
@Dat-NguyenDuy your PR is apparently the bisect source. Can you look at this?

Dat-NguyenDuy, 2024-08-16:
Hi @ubieda, thanks for reporting the issue, could you run the test with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/77140 as i don't have the board in my hand

PR #77140 (2024-08-20): dts: nxp_rt1010: mark edma channel has separate interrupt entry
In SoC imxrt1010, edma channel has separate interrupt entry, not like the rest of in-tree imxrt SoC series.
~I don't have board to test so temporarily mark this PR as draft, waiting feedback from reporter~
This fixes: #77129
closed 2024-08-20 by nashif