zephyrproject-rtos/zephyr #77169: ESP32C3: Inconsistent flash erase times across different ESP32-C3-MINI-1 (ESP32-C3FN4) modules [zephyr-v3.5.0]

abdulble, 2024-08-16:
**Describe the bug**
There is a consistent variation in the time it takes to erase flash across different ESP32-C3 development kits. The erase time is stable for each individual module but differs significantly across modules.

The issue is not easily reproducable, and came up in about 10% of the tests.

**Expected behavior**
The module should not crash due to the fatal error.

**Impact**
Showstopper, since my application gets interrupted during critical operations.

**Logs and console output**
```
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:331
\u0009command opcode 0x2013 timeout with err -11
[00:00:18.935,000]  os:      a0: 00000003    t0: 00000000
[00:00:18.935,000]  os:      a1: 00000002    t1: 42043710
[00:00:18.935,000]  os:      a2: 00000008    t2: 00000020
[00:00:18.935,000]  os:      a3: 60023008    t3: 00000004
[00:00:18.935,000]  os:      a4: 00001888    t4: 0000004c
[00:00:18.935,000]  os:      a5: 00000009    t5: aaaaaaaa
[00:00:18.935,000]  os:      a6: 3fca3cac    t6: aaaaaaaa
[00:00:18.935,000]  os:      a7: 0000002e
[00:00:18.935,000]  os:      ra: 4204796c
[00:00:18.935,000]  os:    mepc: 42047974
[00:00:18.935,000]  os: mstatus: 00001880
[00:00:18.935,000]  os: 
[00:00:18.935,000]  os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:18.935,000]  os: Current thread: 0x3fc8c7f0 (unknown)
[00:00:19.019,000]  os: Halting system
```

**Environment (please complete the following information):**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.26.6
- Toolchain: Zephyr SDK 0.16.4
- Zephyr branch: zephyr-v3.5.0
- Target: esp32c3 (DevkitM-1)

Shreeyash-17, 2024-08-22:
@abdulble can you help me to how to run esp32 on zephyr from scratch?

rftafas, 2024-08-27:
> @abdulble could you please help me to how to run esp32 on zephyr from scratch?

@Shreeyash-17 this seems off-topic. For the sake of a reply, start with https://docs.zephyrproject.org/latest/develop/getting_started/index.html

This issue can be closed, as the reported problem is within component parameter range even though not close to typical values.

abdulble, 2024-08-27:
@sylvioalves Thank you for the detailed explanation. 
@rftafas Alright I will close the issue.

closed 2024-08-27 by abdulble