zephyrproject-rtos/zephyr #77185: TFM: Nordic pinctrl consumers broken

JordanYates, 2024-08-17:
**Describe the bug**

The TFM and Zephyr repositories have mismatched views on how the pinctrl uint32_t is constructed.

https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/069455be098383bf96eab73e3ff8e0c66c60fa5a/platform/ext/target/nordic_nrf/common/core/common/nrf-pinctrl.h#L9-L20

https://github.com/zephyrproject-rtos/zephyr/blob/cce7e9a706ca848f73e4ef2c435668e358b73305/include/zephyr/dt-bindings/pinctrl/nrf-pinctrl.h#L9-L20

The pinctrl values are consumed by the TFM USART driver:
https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/069455be098383bf96eab73e3ff8e0c66c60fa5a/platform/ext/target/nordic_nrf/common/core/cmsis_drivers/Driver_USART.c#L56-L74

Introduced in:
 * #73097
 * #76847

This was probably missed because `NRF_FUN_UART_TX` is 0, so the shift in bits doesn't affect this one pin.
The flow control and RX pins are currently broken.

The first PR is present in Zephyr v3.7.0, so the LTS driver does not currently work as expected.

**Expected behavior**

TFM and Zephyr need the same view on pinctrl, as TFM consumes the values created by the Zephyr build system.

**Impact**

Nordic TFM consumers of pinctrl are currently broken.

**Environment (please complete the following information):**

 - Zephyr v3.7.0

nordic-piks, 2024-09-12:
Triaged internally

JordanYates, 2025-04-04:
> Triaged internally

What does that mean? the issue is still present on the 4.1 release

nordic-piks, 2025-04-04:
@gmarull can you help?

JordanYates, 2025-04-04:
> I submitted a fix a long time ago [#77224](https://github.com/zephyrproject-rtos/zephyr/pull/77224), but it got stale as nobody seemed to care enough.

My understanding from https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/110 was that it required someone (most likely Nordic) to update TF-M upstream, and that never happened.

closed 2025-08-14 by github-actions[bot]

JordanYates, 2025-08-14:
This issue is still valid, still requiring someone from Nordic to ensure they match.
Zephyr:
https://github.com/zephyrproject-rtos/zephyr/blob/cce7e9a706ca848f73e4ef2c435668e358b73305/include/zephyr/dt-bindings/pinctrl/nrf-pinctrl.h#L9-L20
TF-M:
https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/main/platform/ext/target/nordic_nrf/common/core/common/nrf-pinctrl.h#L9-L20

carlescufi, 2025-08-14:
@bjarki-andreasen @tomi-font could you please take a look?

Vge0rge, 2025-10-23:
I am doing some local tests with the updated header and plan to send a PR in upstream TF-M soon.