zephyrproject-rtos/zephyr #77186: [Bug] Sample USB-C Sink Not Working in Zephyr v3.7.0 on weact_stm32g431_core

recalci, 2024-08-17:
**Describe the bug**

After upgrading Zephyr from v3.6.0 to v3.7.0, the project `samples/subsys/usb_c/sink` fails to work correctly on weact_stm32g431_core. The project compiles successfully, but the USB-C charging functionality does not work as expected.

**To Reproduce**

Steps to reproduce the behavior:
1. Switch to Zephyr v3.7.0.
2. Run `west update` to update dependencies.
3. Compile and flash `weact_stm32g431_core samples/subsys/usb_c/sink`.
4. The project fails to work correctly.

**Expected behavior**

The project should compile and run successfully, enabling USB-C PD sink functionality.

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v3.7.0

**Additional context**

The issue seems to be related to changes in the `hal_stm32` module between v3.6.0 and v3.7.0. Specifically, the revision `60c9634f61c697e1c310ec648d33529712806069` works correctly, while `f1317150eac951fdd8259337a47cbbc4c2e6d335` does not.

    - name: hal_stm32
      revision: f1317150eac951fdd8259337a47cbbc4c2e6d335
      path: modules/hal/stm32
      groups:
        - hal

marwaiehm-st, 2024-08-23:
Hi @recalci ,
Would it be possible for you to share the log of the test with me?

recalci, 2024-08-23:
Hi @marwaiehm-st , thanks for responses.

I have gathered some additional information that might be crucial for diagnosing the issue. 

1. The problem occurs specifically when `CONFIG_LOG_DEFAULT_LEVEL=3`. At log levels 2 or 4, the version of `hal_stm32` does not impact the functionality.

2. Additionally, when using the problematic version of `hal_stm32` with `CONFIG_LOG_DEFAULT_LEVEL=3`, there is no log output from the UART. This behavior is not observed at other log levels.

I will update the issue with these new details. 

Here is the normal log output when everything is functioning correctly:
```
[00:00:00.000,000] <err> usbc_stack: Couldn't disable vconn: -134
[00:00:00.000,000] <inf> usbc_stack: ErrorRecovery
[00:00:00.000,000] <inf> usbc_stack: Disabled
[00:00:00.000,000] <inf> usbc_stack: PE_SUSPEND
[00:00:00.000,000] <inf> usbc_stack: PRL_HR_SUSPEND
[00:00:00.000,000] <inf> usbc_stack: PRL_TX_SUSPEND
*** Booting Zephyr OS build v3.7.0-1040-g6b7558a9bf96 ***
[00:00:00.000,000] <inf> usbc_stack: ErrorRecovery
[00:00:00.245,000] <inf> usbc_stack: Unattached.SNK
[00:00:00.245,000] <inf> usbc_stack: AttachWait.SNK
[00:00:00.449,000] <inf> usbc_stack: Attached.SNK
[00:00:00.449,000] <inf> usbc_stack: PE_SNK_Startup
[00:00:00.449,000] <inf> usbc_stack: PRL_INIT
[00:00:00.449,000] <inf> usbc_stack: PRL_HR_Wait_for_Request
[00:00:00.449,000] <inf> usbc_stack: PRL_Tx_PHY_Layer_Reset
[00:00:00.449,000] <inf> usbc_stack: PRL_Tx_Wait_for_Message_Request
[00:00:00.449,000] <inf> usbc_stack: PE_SNK_Discovery
[00:00:00.449,000] <inf> usbc_stack: PE_SNK_Wait_For_Capabilities
[00:00:00.470,000] <inf> main: PWR 3A0
[00:00:00.599,000] <inf> usbc_stack: RECV 51a1/5 
[00:00:00.599,000] <inf> usbc_stack:    [0]0801912c 
[00:00:00.599,000] <inf> usbc_stack:    [1]0002d12c 
[00:00:00.599,000] <inf> usbc_stack:    [2]0003c12c 
[00:00:00.599,000] <inf> usbc_stack:    [3]0004b12c 
[00:00:00.599,000] <inf> usbc_stack:    [4]0006412c 
[00:00:00.600,000] <inf> usbc_stack: PE_SNK_Evaluate_Capability
[00:00:00.600,000] <inf> usbc_stack: PE_SNK_Select_Capability
[00:00:00.601,000] <inf> usbc_stack: PRL_Tx_Wait_for_PHY_response
[00:00:00.601,000] <inf> usbc_stack: RECV 01a1/0 
[00:00:00.601,000] <inf> usbc_stack: PRL_Tx_Wait_for_Message_Request
[00:00:00.605,000] <inf> usbc_stack: RECV 03a3/0 
[00:00:00.605,000] <inf> usbc_stack: PE_SNK_Transition_Sink
[00:00:00.716,000] <inf> usbc_stack: RECV 05a6/0 
[00:00:00.716,000] <inf> usbc_stack: PE_SNK_Ready
[00:00:01.000,000] <inf> main: Source Caps:
[00:00:01.000,000] <inf> main: PDO 0:
[00:00:01.000,000] <inf> main:  Type:              FIXED
[00:00:01.000,000] <inf> main:  Current:           3000
[00:00:01.000,000] <inf> main:  Voltage:           5000
[00:00:01.000,000] <inf> main:  Peak Current:      0
[00:00:01.000,000] <inf> main:  Uchunked Support:  0
[00:00:01.000,000] <inf> main:  Dual Role Data:    0
[00:00:01.000,000] <inf> main:  USB Comms:         0
[00:00:01.000,000] <inf> main:  Unconstrained Pwr: 1
[00:00:01.000,000] <inf> main:  USB Suspend:       0
[00:00:01.000,000] <inf> main:  Dual Role Power:   0
[00:00:01.050,000] <inf> main: PDO 1:
[00:00:01.050,000] <inf> main:  Type:              FIXED
[00:00:01.050,000] <inf> main:  Current:           3000
[00:00:01.050,000] <inf> main:  Voltage:           9000
[00:00:01.050,000] <inf> main:  Peak Current:      0
[00:00:01.050,000] <inf> main:  Uchunked Support:  0
[00:00:01.050,000] <inf> main:  Dual Role Data:    0
[00:00:01.050,000] <inf> main:  USB Comms:         0
[00:00:01.050,000] <inf> main:  Unconstrained Pwr: 0
[00:00:01.050,000] <inf> main:  USB Suspend:       0
[00:00:01.050,000] <inf> main:  Dual Role Power:   0
[00:00:01.100,000] <inf> main: PDO 2:
[00:00:01.100,000] <inf> main:  Type:              FIXED
[00:00:01.100,000] <inf> main:  Current:           3000
[00:00:01.100,000] <inf> main:  Voltage:           12000
[00:00:01.100,000] <inf> main:  Peak Current:      0
[00:00:01.100,000] <inf> main:  Uchunked Support:  0
[00:00:01.100,000] <inf> main:  Dual Role Data:    0
[00:00:01.100,000] <inf> main:  USB Comms:         0
[00:00:01.100,000] <inf> main:  Unconstrained Pwr: 0
[00:00:01.100,000] <inf> main:  USB Suspend:       0
[00:00:01.100,000] <inf> main:  Dual Role Power:   0
[00:00:01.151,000] <inf> main: PDO 3:
[00:00:01.151,000] <inf> main:  Type:              FIXED
[00:00:01.151,000] <inf> main:  Current:           3000
[00:00:01.151,000] <inf> main:  Voltage:           15000
[00:00:01.151,000] <inf> main:  Peak Current:      0
[00:00:01.151,000] <inf> main:  Uchunked Support:  0
[00:00:01.151,000] <inf> main:  Dual Role Data:    0
[00:00:01.151,000] <inf> main:  USB Comms:         0
[00:00:01.151,000] <inf> main:  Unconstrained Pwr: 0
[00:00:01.151,000] <inf> main:  USB Suspend:       0
[00:00:01.151,000] <inf> main:  Dual Role Power:   0
[00:00:01.201,000] <inf> main: PDO 4:
[00:00:01.201,000] <inf> main:  Type:              FIXED
[00:00:01.201,000] <inf> main:  Current:           3000
[00:00:01.201,000] <inf> main:  Voltage:           20000
[00:00:01.201,000] <inf> main:  Peak Current:      0
[00:00:01.201,000] <inf> main:  Uchunked Support:  0
[00:00:01.201,000] <inf> main:  Dual Role Data:    0
[00:00:01.201,000] <inf> main:  USB Comms:         0
[00:00:01.201,000] <inf> main:  Unconstrained Pwr: 0
[00:00:01.201,000] <inf> main:  USB Suspend:       0
[00:00:01.201,000] <inf> main:  Dual Role Power:   0
```

Please let me know if there is anything else I can provide to assist in resolving this issue.

marwaiehm-st, 2024-08-27:
Hi @recalci , thank you for the responses.

I reproduced the issue on the board b_g474e_dpow1, but it occurs when CONFIG_LOG_DEFAULT_LEVEL=2.

The regression is due to a modification in the function "LL_SetFlashLatency" on the file "stm32g4xx_ll_utils.c".
A condition was added to set the latency based on the BOOST Mode status, but in my case, it's disabled because the HCLKFrequency is 128MHz. According to the clock control, the frequency should be greater than 150MHz to enable the Boost Mode.
[image]

[image]

I propose, for the moment, to test with the old implementation of the function LL_SetFlashLatency  on your board "weact_stm32g431_core".
[revert_ll_setFlashLatency.txt](https://github.com/user-attachments/files/16760254/revert_ll_setFlashLatency.txt)

It's not a final solution, I'm still working on it.

recalci, 2024-08-28:
@marwaiehm-st 
I have tested the suggested configuration, and it works. Thank you for your help!
I guess this is a temporary solution. Should I close this issue?

marwaiehm-st, 2024-08-29:
Hi @recalci 
No, you should not close this issue until we find a permanent solution.

erwango, 2024-11-06:
@FRASTM, can you check in datasheet if  Sysclock @ 128MHz is compatible with use case ?

FRASTM, 2024-11-06:
For a reason I cannot explain, on my b_g474e_dpow1 stm32g474 power kit, I can see the LOG output when 
- the sysclock is 148MHz which is below the range1 for enabling the boost 
- or the sysclock is unchanged at 130MHz with a lower TICK per sec set to 4000

FRASTM, 2024-11-07:
For a reason I cannot explain, on my b_g474e_dpow1 stm32g474 power kit, I can see the LOG output when 
- the sysclock is 148MHz which is below the range1 for enabling the boost. It is like some combinations prevent the expected LOG output (sysclock @ 136MHz or 138MHz failed but 134MHz passed)
- or with a lower TICK per sec set to 4000

When failing, the code is locked in 
```
Program received signal SIGINT, Interrupt.
z_timer_expiration_handler (t=0x20000bc0 <tc_0+168>) at ~/zephyrproject/zephyr/kernel/timer.c:87
87			timer->expiry_fn(timer);
(gdb) bt
#0  z_timer_expiration_handler (t=0x20000bc0 <tc_0+168>) at ~/zephyrproject/zephyr/kernel/timer.c:87
#1  0x08005960 in sys_clock_announce (ticks=<optimized out>) at ~/zephyrproject/zephyr/kernel/timeout.c:240
#2  0x08004522 in sys_clock_isr () at ~/zephyrproject/zephyr/drivers/timer/cortex_m_systick.c:214
#3  <signal handler called>
#4  0x08006092 in __l_vfprintf ()
#5  0x08002656 in cbvprintf (out=out@entry=0x8006c1f <out_func>, ctx=ctx@entry=0x800a228 <lbu_output>, 
    fp=0x800ad9d "[%02u:%02u:%02u.%03u,%03u] ", fp@entry=0xf "\bu ", ap=..., ap@entry=...)
    at ~/zephyrproject/zephyr/lib/libc/picolibc/libc-hooks.c:99
#6  0x08001730 in print_formatted (output=output@entry=0x800a228 <lbu_output>, fmt=0x800ad9d "[%02u:%02u:%02u.%03u,%03u] ")
    at ~/zephyrproject/zephyr/subsys/logging/log_output.c:147
#7  0x08001870 in timestamp_print (timestamp=<optimized out>, flags=15, output=0x800a228 <lbu_output>)
    at ~/zephyrproject/zephyr/subsys/logging/log_output.c:313
#8  prefix_print (tid=0x0, func_on=false, level=1 '\001', source=0x800b2e6 "usbc_stack", domain=0x0, timestamp=0, flags=15, 
    output=0x800a228 <lbu_output>) at ~/zephyrproject/zephyr/subsys/logging/log_output.c:666
#9  log_output_process (output=output@entry=0x800a228 <lbu_output>, timestamp=timestamp@entry=0, domain=domain@entry=0x0, 
    source=0x800b2e6 "usbc_stack", tid=tid@entry=0x0, level=level@entry=1 '\001', package=0x20000320 <buf32+16> "\003", 
    data=data@entry=0x2000032c <buf32+28> "", data_len=0, flags=flags@entry=15)
    at ~/zephyrproject/zephyr/subsys/logging/log_output.c:707
#10 0x08006cd0 in log_output_msg_process (output=0x800a228 <lbu_output>, msg=<optimized out>, flags=15)
    at ~/zephyrproject/zephyr/subsys/logging/log_output.c:750
#11 0x08001374 in log_backend_msg_process (msg=0x20000310 <buf32>, backend=0x800a1b4 <log_backend_uart>)
    at ~/zephyrproject/zephyr/include/zephyr/logging/log_backend.h:180
#12 msg_process (msg=<optimized out>) at ~/zephyrproject/zephyr/subsys/logging/log_core.c:502
#13 z_impl_log_process () at ~/zephyrproject/zephyr/subsys/logging/log_core.c:559
#14 z_impl_log_process () at ~/zephyrproject/zephyr/subsys/logging/log_core.c:542
#15 0x0800142c in log_process () at ~/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/syscalls/log_ctrl.h:57
#16 log_process_thread_func (dummy1=<optimized out>, dummy2=<optimized out>, dummy3=<optimized out>)
    at ~/zephyrproject/zephyr/subsys/logging/log_core.c:957
#17 0x08000f9a in z_thread_entry (entry=0x80013d9 <log_process_thread_func>, p1=0x0, p2=0x0, p3=0x0)
    at ~/zephyrproject/zephyr/lib/os/thread_entry.c:48
#18 0x2020f89c in ?? ()
```

FRASTM, 2024-11-07:
set **CONFIG_LOG_MODE_IMMEDIATE=y** to get output on the console