zephyrproject-rtos/zephyr #77196: drivers: spi: stm32: Broken device power management on SPI bus

hubpav, 2024-08-17:
With PR #72991, the SPI driver does not work with PM device + runtime features enabled.

The bug was reproduced on the STM32WL device.

Steps to reproduce:

1. In `samples/subsys/lorawan/class_a` add  `CONFIG_PM_DEVICE=y` and `CONFIG_PM_DEVICE_RUNTIME=y` to `prj.conf`.

2. Build the sample: `west build -b lora_e5_mini`

3. Debug the sample (`west debug`), set the breakpoint to the `main` function `b main`, and run the code.

4. The `main` function is never hit as it is stuck in the SPI driver (`spi_ll_stm32.c:878` - in `transceive` function).

When I reverted the commit `858e8c51f1f42abb0afe62a4d701e69b7a6bd2b1` from PR #72991, the sample works as expected even with the above PM device options enabled.

erwango, 2024-08-19:
@aurel32 FYI

easyvolts, 2024-09-25:
I also met this issue and found that it's caused by SPI clock not being re-enabled during resuming SPI operation because pinctrl_apply_state function returns error for stm32wl devices (because these devices have no physical pins associated with the SPI). The fix is simple, please see the diff below:
```
diff --git zephyr/drivers/spi/spi_ll_stm32.c zephyr/drivers/spi/spi_ll_stm32.c
index 7b557ebd0..c5fab3ac5 100644
--- zephyr/drivers/spi/spi_ll_stm32.c
+++ zephyr/drivers/spi/spi_ll_stm32.c
@@ -1242,9 +1242,11 @@ static int spi_stm32_pm_action(const struct device *dev,
        switch (action) {
        case PM_DEVICE_ACTION_RESUME:
                /* Set pins to active state */
-               err = pinctrl_apply_state(config->pcfg, PINCTRL_STATE_DEFAULT);
-               if (err < 0) {
-                       return err;
+               if (!spi_stm32_is_subghzspi(dev)) {
+                       err = pinctrl_apply_state(config->pcfg, PINCTRL_STATE_DEFAULT);
+                       if (err < 0) {
+                               return err;
+                       }
                }

                /* enable clock */
@@ -1263,16 +1265,18 @@ static int spi_stm32_pm_action(const struct device *dev,
                }

                /* Move pins to sleep state */
-               err = pinctrl_apply_state(config->pcfg, PINCTRL_STATE_SLEEP);
-               if ((err < 0) && (err != -ENOENT)) {
-                       /*
-                        * If returning -ENOENT, no pins where defined for sleep mode :
-                        * Do not output on console (might sleep already) when going to sleep,
-                        * "SPI pinctrl sleep state not available"
-                        * and don't block PM suspend.
-                        * Else return the error.
-                        */
-                       return err;
+               if (!spi_stm32_is_subghzspi(dev)) {
+                       err = pinctrl_apply_state(config->pcfg, PINCTRL_STATE_SLEEP);
+                       if ((err < 0) && (err != -ENOENT)) {
+                               /*
+                               * If returning -ENOENT, no pins where defined for sleep mode :
+                               * Do not output on console (might sleep already) when going to sleep,        
+                               * "SPI pinctrl sleep state not available"
+                               * and don't block PM suspend.
+                               * Else return the error.
+                               */
+                               return err;
+                       }
                }
                break;
        default:
```

aurel32, 2024-09-25:
Hi. Thanks for the patch. However I am confused, as it looks very similar to this https://github.com/zephyrproject-rtos/zephyr/commit/18c73de93d53b1c7f81a6c9d5b1a1353cb917423 which went in around the same time than the runtime PM support for SPI. Which version are you using?

easyvolts, 2024-09-26:
> Hi. Thanks for the patch. However I am confused, as it looks very similar to this [18c73de](https://github.com/zephyrproject-rtos/zephyr/commit/18c73de93d53b1c7f81a6c9d5b1a1353cb917423) which went in around the same time than the runtime PM support for SPI. Which version are you using?

Hi, I might have overlooked an existing fix. My work is based on Zephyr OS v3.6.99-100befc70c74 version.

hubpav, 2024-11-26:
Keep open

hubpav, 2025-01-27:
Keep open

erwango, 2025-03-17:
@hubpav Did you had a test on https://github.com/zephyrproject-rtos/zephyr/pull/72703 which was merged after https://github.com/zephyrproject-rtos/zephyr/pull/72991 ?

hubpav, 2025-03-22:
@erwango The PR https://github.com/zephyrproject-rtos/zephyr/pull/72703 was merged on May 24, 2024. My report was on Aug 17, 2024; I tested it on top of the main branch at that time. But I can try again with my above-provided test instructions.

visuphi, 2025-04-22:
Hi @hubpav 

Have you already had time to test it? We currently have problems with a STM32L0 where CONFIG_PM is activated and LoRAWAN is not working properly, but we don't know yet if this is related to this issue.

zafersn, 2025-07-23:
> We were able to fix the problem by setting the clock in the dts `stm32_lp_tick_source`, but maybe it was a different problem.

Thanks, @visuphi. I already have that defined in my DTS, but the issue still persists.

Here’s what I currently have:
```
stm32_lp_tick_source: &lptim1 {
	clocks = <&rcc STM32_CLOCK_BUS_APB3 0x00000800>,
	         <&rcc STM32_SRC_LSE LPTIM1_SEL(3)>;
	status = "okay";
};
```

I also tested it with LSI instead of LSE, like this:

```
stm32_lp_tick_source: &lptim1 {
	clocks = <&rcc STM32_CLOCK_BUS_APB3 0x00000800>,
		 <&rcc STM32_SRC_LSI LPTIM1_SEL(1)>;
	status = "okay";
};
```
But unfortunately, that didn’t resolve the issue either.

erwango, 2025-07-23:
> Thanks, [@visuphi](https://github.com/visuphi). I already have that defined in my DTS, but the issue still persists.
>
> Here’s what I currently have:
>
> ```
> stm32_lp_tick_source: &lptim1 {
> 	clocks = [...]

Just to be sure. Did you enabled lsi/lse clocks as well ?
(I prefer to ask as this kind of thing happen frequently).

zafersn, 2025-07-23:
> Should be fixed by [#87246](https://github.com/zephyrproject-rtos/zephyr/pull/87246). Please have a test. Fixes the issue described on my set up.

@erwango  thanks a lot; that patch fixed my issue too. now all working

closed 2025-07-24 by cfriedt