zephyrproject-rtos/zephyr #77205: Device tree tests cannot be built with coverage enabled

Freey0, 2024-08-18:
**Describe the bug**

Device tree tests cannot be built with coverage enabled

DT_SPEC related macro tests (ADC_DT_SPEC_GET_BY_NAME, MBOX_DT_SPEC_GET)should not appear in this test because the configuration does not turn on any devices

**To Reproduce**
./scripts/twister -p native_sim -T tests/lib/devicetree --coverage-tool gcovr -i --coverage

**Expected behavior**
No Error

**Impact**

Unable to generate code base coverage

**Logs and console output**

INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-1236-gad3e941ad3ed
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report xxx/zephyr/twister-out/testplan.json
INFO    - JOBS: 22
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - native_sim                tests/lib/devicetree/api/libraries.devicetree.api   ERROR : Build failure
INFO    - xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/build.log
ERROR   - Loading Zephyr default modules (Zephyr base).

.......
Generating files from xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/zephyr.elf for board: native_sim
[96/97] Building native simulator runner, and linking final executable
FAILED: zephyr/CMakeFiles/native_runner_executable zephyr/zephyr.exe xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/CMakeFiles/native_runner_executable xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/zephyr.exe 
cd xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr && /usr/bin/make -f xxx/zephyr/scripts/native_simulator/Makefile all --warn-undefined-variables -r NSI_CONFIG_FILE=xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/NSI/nsi_config
/usr/bin/ld: xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/NSIxxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/zephyr.elf.loc_cpusw.o: in function `devicetree_api_test_io_channel_names':
xxx/zephyr/tests/lib/devicetree/api/src/main.c:1187: undefined reference to `__device_dts_ord_86'
/usr/bin/ld: xxx/zephyr/tests/lib/devicetree/api/src/main.c:1190: undefined reference to `__device_dts_ord_87'
/usr/bin/ld: xxx/zephyr/tests/lib/devicetree/api/src/main.c:1194: undefined reference to `__device_dts_ord_86'
/usr/bin/ld: xxx/zephyr/tests/lib/devicetree/api/src/main.c:1197: undefined reference to `__device_dts_ord_87'
collect2: error: ld returned 1 exit status
make: *** [xxx/zephyr/scripts/native_simulator/Makefile:131：xxx/zephyr/twister-out/native_sim/tests/lib/devicetree/api/libraries.devicetree.api/zephyr/zephyr.exe] 错误 1
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK-0.16.8
 - ad3e941ad3ed9e3e2f9032b0151558a948e99fb9

**Additional context**

Why did CI not find this BUG, because it was optimized out. Failure is allowed in coverage tests for CI.
There are always a fixed number of test failures in CI coverage tests

Freey0, 2024-08-18:
ci coverage test in native_sim platform, there are always 3 failures.

```
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/net/dhcpv6/net.dhcpv6 on native_sim error (Build failure)
INFO    - 2) tests/net/dhcpv6/net.dhcpv6.preempt on native_sim error (Build failure)
INFO    - 3) tests/lib/devicetree/api/libraries.devicetree.api on native_sim error (Build failure)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
```

1 and 2 are uninitialized variables
```
__w/zephyr/zephyr/tests/net/dhcpv6/src/main.c: In function ‘net_dhcpv6_init’:
/__w/zephyr/zephyr/subsys/net/ip/../../../subsys/net/lib/dhcpv6/dhcpv6.c:2211:26: error: ‘unspec_addr’ may be used uninitialized [-Werror=maybe-uninitialized]
 2211 |         net_ipaddr_copy(&net_sin6(&unspec_addr)->sin6_addr,
```

dkalowsk, 2024-08-20:
Assigning to the reporter as they are the author of the open PR as well.

commit 7d4e31a1 (2025-01-22): devicetree: Fix Device tree tests cannot be built with coverage enabled

DT_SPEC related macro tests (ADC_DT_SPEC_GET_BY_NAME, MBOX_DT_SPEC_GET)
should not appear in this test because the configuration does not turn
on any devices

Move DP_SPEC-related tests to api_ext and enable the related device in the
configuration

fix #77205
closed 2025-01-22 by kartben