zephyrproject-rtos/zephyr #77211: Static lightweight event publish-subscribe bus

LuoZhongYao, 2024-08-19:
## Introduction

 Ebus is a static event publish-subscribe bus, where publishers and subscribers have a many-to-many relationship. Compared to zbus, ebus is lighter. 

### Problem description

There are many scenarios where zbus is too heavyweight and cumbersome to use, and callback function registration is highly coupled. 

### Proposed change

* Zero-to-many, many-to-many, many-to-zero buses.
* Subscribers, publishers are not dependent on each other, are not aware of each other, exist independently.
* Lightweight

## Detailed RFC

Use `uint32_t id` to identify events, `id` is generated by `uint16_t namepsace` and `uint16_t event`, `ebus_id = namespace << 16 | event`. 
The `namespace` defines the namespace, the module registers the namespace using `EBUS_NAMESPACE_REGISTER` (e.g `EBUS_NAMESPACE_REGISTER(power)`, registers the `power` namespace), and the module defines the event `event` in its own namespace as needed. `. 
The user needs to declare the namespace using `EBUS_NAMESPACE_DECLARE`, (e.g `EBUS_NAMESPACE_DECLARE(power)`, declare the `power` namespace). Then use `EBUS_SUBSCRIBE(namespace, ev, callabck, userdata)`, to listen for `ev` events from `namespace`, and when the publisher publishes `ev` events from `namespace` using `EBUS_PUBLISH(namespace, ev)`, the subscriber calls `EBUS_PUBLISH(namespace, ev)`. event, the subscriber calls `callback`. Since `ebus_id` is a constant, a perfect hash table can be generated using `gperf` to speed up the process.
The initial design is as follows.
* ebus.h
```c
/*
 * Copyright (c) 2024 LuoZhongYao@gmail.com
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

#if !defined(CONFIG_EBUS)

#define EBUS_NAMESPACE_REGISTER(name)
#define EBUS_NAMESPACE_DECLARE(name)
#define EBUS_PUBLISH(_namespace, _ev, ...)
#define EBUS_SUBSCRIBE(_namespace, _ev, _callback, _userdata)

#else /* CONFIG_EBUS */

#include <stdint.h>
#include <toolchain.h>

#define __ebustext __attribute__((__section__(".TEXT.ebus")))

typedef void (*ebus_subscriber_t)(uint32_t ebus, void *userdata, void *appendix);

struct ebus_subscriber {
	void *userdata;
	ebus_subscriber_t subscriber;
};

struct ebus_slot_subscriber {
	uintptr_t ev;
	const uint8_t *namespace;
	const struct ebus_subscriber *subscriber;
} ;

struct ebus_publish {
	void *appendix;
	void (*drop)(struct ebus_publish *publish);
};

extern uint8_t __ebus_namespace_start[];
extern uint8_t __ebus_namespace_end[];
extern const struct ebus_subscriber __ebus_subscriber_start[];
extern const struct ebus_subscriber __ebus_subscriber_end[];
extern int ebus_publish(uint32_t ebus, struct ebus_publish *publish);

#define EBUS_ORIGINAL(_namespace, _ev) (((uint32_t)(&EBUS_NAMESPACE(_namespace)) << 16) | (_ev))

[... 362 lines cut ...]
EBUS_NAMESPACE_REGISTER(power);

/* subscribe.c */
#include "power.h"
static void power_on(uint32_t ebus, void *userdata, void *appendix)
{
	printk("power on: %s\n", (const char *)appendix);
}
EBUS_SUBSCRIBE(power, POWER_ON, power_on, NULL);

/* publish.c */
#include "power.h"
static struct ebus_publish msg = {"hello", NULL};
void foo(void)
{
	...
	EBUS_PUBLISH(power, POWER_ON, &pub);
	...
}

```

### Concerns and Unresolved Questions

In `user mode`, the callback runs in kernel space, which is a security risk.

pdgendt, 2024-08-20:
Not sure if there's a need for this, and I think `zbus` can be configured to be light-weight too? @rodrigopex

rodrigopex, 2024-08-24:
@pdgendt 
The lightweight zbus is already there, so use listeners and disable all the unnecessary features.

rodrigopex, 2024-08-24:
> There are many scenarios where zbus is too heavyweight and cumbersome to use, and callback function registration is highly coupled.

@LuoZhongYao, could you please give us some examples of the problem?

LuoZhongYao, 2024-08-26:
> > There are many scenarios where zbus is too heavyweight and cumbersome to use, and callback function registration is highly coupled.
> 
> @LuoZhongYao, could you please give us some examples of the problem? If possible, add them to the issue description.

I compared them carefully, and zbus's LISTENER is very similar to ebus. zbus is based on channels, and ebus is based on events. Maybe ebus is not needed.

closed 2024-08-26 by LuoZhongYao