zephyrproject-rtos/zephyr #77212: uac2_explicit_feedback example USB device does not work on Windows

prnthp, 2024-08-19:
**Describe the bug**

Using nRF Connect SDK 2.7.0 with an nRF5340 DK, the example in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/uac2_explicit_feedback fails to start as a USB Audio 2.0 device in Windows.

**To Reproduce**
1. Have nrf5340DK
2. Set up nRF Connect SDK 2.7.0 in VS Code
3. Build and flash uac2_explicit_feedback example

**Expected behavior**
USB device enumerates correctly on Windows and shows up as audio device

**Impact**
Not much, but it would be nice to be able to use USB Audio 2.0 with feedback instead of the USB Audio 1.0 examples

**Logs and console output**
N/A
[image]

**Environment (please complete the following information):**

 - OS: Windows
 - nRF Connect 2.7.0 toolchain

henrikbrixandersen, 2024-08-19:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

tmon-nordic, 2024-08-19:
The Code 10 is not telling much. I can reproduce it and the Event Log contains:
> The driver could not find a feedback endpoint for an asynchronous data OUT endpoint on device \Device\000000a9.

I wonder why it cannot find feedback endpoint (there is one in the sample).

tmon-nordic, 2024-08-19:
@mvaneerde Could you perhaps shed more light on why Microsoft UAC2 driver expects asynchronous feedback endpoint to have `wMaxPacketSize` equal 3 when device is operating at Full-Speed?

tmon-nordic, 2024-08-20:
Reported the problem via Feedback Hub: https://aka.ms/AArvnax

udoeb, 2024-08-26:
Hi, this is Udo. I wrote the UAC2 driver for MS. 

Unfortunately, at full speed, the driver does not support feedback endpoints correctly. This is an issue MS did not discover during testing as FS support was not on their list of requirements. I provided a fix after the driver was released but obviously it never made it into Windows. 

If your MCU is limited to FS, you should create an audio class 1 device. You will not gain much from HS anyway. Bandwidth and latency will not improve.

PR #86660 (2025-03-05): samples: usb: uac2: Warn about Windows driver issues
The UAC2 samples do not work with Windows UAC2 driver because:
  * driver does not support implicit feedback
  * driver expects wMaxPacketSize equal 4 on Full-Speed device

The only correct place to solve these issues is the Windows UAC2 driver.

Resolves: #77212
closed 2025-03-05 by fabiobaltieri