zephyrproject-rtos/zephyr #77287: ipv6: ICMP6, neighbor solicitation, who has 2001:db8::2

fulup-bzh, 2024-08-20:
**ipv6 neighbor solicitation not sent when calling connect(*sock, addr, addr_len);**

**Reproducing** : Use standard zephyr/sample/net/socket/http_client with IPV6 only configuration. The ```network_solicitation``` is sent when doing a ```net ping 2001:db8::2``` but not when calling C/API:connect(*sock, addr, addr_len)

**Expected**:  ```network_solicitation``` should be send when trying to reach an IPv6 not present within routing table.

**TCPDUMP Log** on server side. ```ping ff02::1:ff00:2``` froim Zephyr activates network solicitation, when without ping nothing happen, not the local link router solicitation work without ping.. 
```
15:22:07.950826 IP6 fe80::212:4b00:29b4:fce2 > ip6-allrouters: ICMP6, router solicitation, length 24
15:22:08.203505 IP6 2001:db8::1 > ff02::1:ff00:2: ICMP6, neighbor solicitation, who has 2001:db8::2, length 40
15:22:08.203832 IP6 2001:db8::2 > 2001:db8::1: ICMP6, neighbor advertisement, tgt is 2001:db8::2, length 40
```

**Environment**:
 - OS: Linux Debian on beagle board (lowpan)
 - Toolchain: Zephyr SDK-0.16.8, ...)
 - Zephyr: current 3.7 stable
 - target: beagle connect freedom

**Additional context**
```ip router``` does not return anything (even after ping and ip discovery). ```net ipv6``` shows a mask of /128 when /64 would be expected.
```
uart:~$ net ipv6
IPv6 support                              : enabled
IPv6 fragmentation support                : disabled
Multicast Listener Discovery support      : enabled
Neighbor cache support                    : enabled
Neighbor discovery support                : enabled
Duplicate address detection (DAD) support : enabled
Router advertisement RDNSS option support : disabled
6lo header compression support            : enabled
Privacy extension support                 : disabled
Max number of IPv6 network interfaces in the system          : 1
Max number of unicast IPv6 addresses per network interface   : 2
Max number of multicast IPv6 addresses per network interface : 3
Max number of IPv6 prefixes per network interface            : 2

IPv6 addresses for interface 1 (0x20001578) (IEEE 802.15.4)
=========================================================
Type      	State    	Lifetime (sec)	Ref	Address
autoconf  	preferred	      infinite	2	fe80::212:4b00:29b4:fce2/128
manual  	preferred	      infinite	1	2001:db8::1/128
```

jukkar, 2024-08-20:
That sounds really weird, I just tried the http-client sample and it definitely sends neighbor solicitation to `2001:db8::2`.
I do not have the same board as you but used native-sim to verify this, but this should not matter as the IPv6 stack is board agnostic.

fulup-bzh, 2024-08-20:
Note that I tested it with beagleconnect with lowpan network interface. Following @Ayush1325 on Discord I added following config to the sample. I ordered an NXP board to make the same test on a more standard Ethernet interface, I should received it within few days. This behavior may come from some IEEE802154 specificity.

```
CONFIG_NET_L2_IEEE802154=y
CONFIG_NET_L2_IEEE802154_SHELL=y
CONFIG_NET_L2_IEEE802154_LOG_LEVEL_INF=y

# Uncomment for 868 MHz
#CONFIG_NET_CONFIG_IEEE802154_CHANNEL=0
# Uncomment for 906 MHz:
CONFIG_NET_CONFIG_IEEE802154_CHANNEL=1
CONFIG_NET_MGMT_EVENT_STACK_SIZE=2048
```

jukkar, 2024-08-20:
@fgrandel any idea what might cause this in 802.15.4?

fgrandel, 2024-08-21:
@fulup-bzh Thanks for your bug report.

> @fgrandel any idea what might cause this in 802.15.4?

 I agree that NS should be sent in the connect() scenario. I'll try to reproduce later today.

fulup-bzh, 2024-08-21:
I confirm that the same config without NET_CONFIG_IEEE802154 on NXP FRDM-K64F send a  neighbour solicitation (same SDK & Zephyr)
```
15:46:30.617750 IP6 2001:db8::1 > ff02::1:ff00:2: ICMP6, neighbor solicitation, who has 2001:db8::2, length 32
```

fgrandel, 2024-08-21:
@fulup-bzh

I debugged the issue and on my side at least the root cause was missing resources on the HTTP client. This meant that NS packets could not be created. Probably this does not occur with ETH interfaces because they are much faster and won't have to buffer so many packets.

I tested with both, the CC1352P1 2.4GHz and SubG drivers.

As the CC1352P1 has fewer resources as the P7, I had to (temporarily) optimize the server RAM size. But this is unrelated to your issue. I also introduced a few debugging utils/improvements in case you still experience the problem so that you could send us more detailed logs.

See #77370.

The PR is draft as it contains changes that should not be merged. I only included them for full reproducability of what I tested locally.

Can you please try to apply the fixes and re-try?

fulup-bzh, 2024-08-22:
Much better, but not clean yet.

Initial HTTP/GET complete on server side but zephyr fail to read http body ```[00:00:01.141,540] <err> net_ieee802154_6lo_fragment: Could not get a cache entry```. Here after 
* zephyr log 
* script used on beagleplay as httpd responder.

**Zephyr LOG on Beagleconnect**
```
*** Booting Zephyr OS build v3.7.0-49-ga58f40736714 ***
[00:00:00.044,067] <inf> net_config: Initializing network
[00:00:00.175,292] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.184,631] <dbg> net_http_client_sample: main: Init http_client.
[00:00:00.400,146] <dbg> net_http_client: http_flush_data: Data to send
                                          47 45 54 20 2f 20 48 54  54 50 2f 31 2e 31 0d 0a |GET / HT TP/1.1..
                                          48 6f 73 74 3a 20 32 30  30 31 3a 64 62 38 3a 3a |Host: 20 01:db8::
                                          32 0d 0a 0d 0a                                   |2....            
[00:00:00.553,192] <dbg> net_http_client: http_client_req: (main): Sent 37 bytes
[00:00:00.564,666] <dbg> net_http_client: on_message_begin: (main): -- HTTP GET response (headers) --
[00:00:00.578,033] <dbg> net_http_client: on_status: (main): HTTP response status 200 OK
[00:00:00.589,843] <dbg> net_http_client: print_header_field: (main): [12] Content-Type
[00:00:00.601,531] <dbg> net_http_client: print_header_field: (main): [10] text/plain
[00:00:00.612,976] <dbg> net_http_client: on_headers_complete: (main): Headers complete
[00:00:00.624,572] <dbg> net_http_client: on_body: (main): Processed 65 length 65
[00:00:01.141,540] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
[00:00:01.152,954] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
[00:00:02.037,597] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
[00:00:02.049,011] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
[00:00:03.399,810] <dbg> net_http_client: http_wait_data: Timeout
[00:00:03.409,332] <dbg> net_http_client: http_wait_data: Connection error (-116)
[00:00:03.420,379] <dbg> net_http_client: http_client_req: (main): Wait data failure (-116)
[00:00:03.504,486] <err> net_http_client_sample: http_client exit
[00:00:03.829,559] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
[00:00:03.841,003] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
```

**HTTPd server on BeaglePlay*** (do not forget '-6' to map lowpan0 interface. 
```
#!/bin/sh
   socat \
    -6 -v -d -d \
    TCP-LISTEN:1234,crlf,reuseaddr,fork \
    SYSTEM:"
        echo HTTP/1.1 200 OK;
        echo Content-Type\: text/plain;
        echo;
        echo \"Server: \$SOCAT_SOCKADDR:\$SOCAT_SOCKPORT\";
        echo \"Client: \$SOCAT_PEERADDR:\$SOCAT_PEERPORT\";
    "
```

fulup-bzh, 2024-08-22:
Note:
 * I modified sample main.c to keep shell even when client die
 * the config I added to http_client config sample
 
**main.c should not kill Zephyr shell on error **
 ```
 		ret = run_queries();
		if (ret < 0) {
            LOG_ERR("http_client exit");
			k_sleep(K_FOREVER);
		}
```

**sample config addon** 
```
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=n
CONFIG_NET_CONFIG_MY_IPV4_ADDR=""
CONFIG_NET_CONFIG_PEER_IPV4_ADDR=""

# HTTP
CONFIG_HTTP_CLIENT=y
CONFIG_NET_SOCKETS_SOCKOPT_TLS=n

# Ressources
CONFIG_MAIN_STACK_SIZE=3072
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128
CONFIG_NET_CONTEXT_NET_PKT_POOL=y
```

fgrandel, 2024-08-22:
> Initial HTTP/GET complete on server side but zephyr fail to read http body `[00:00:01.141,540] <err> net_ieee802154_6lo_fragment: Could not get a cache entry`. Here after
>```
> [00:00:01.141,540] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
> [00:00:01.152,954] <err> net_ieee802154_6lo_fragment: Could not get a cache entry
> ```

Please try increasing `CONFIG_NET_L2_IEEE802154_FRAGMENT_REASS_CACHE_SIZE` and see whether this helps. Let me know about your results so that I can include the fix in the config overlay. I'll also make the error message more informative.

The background is: IPv6 default L3 MTU is 1280 bytes while IEEE 802.15.4 supports an L2 "MTU" of 125 bytes (127 bytes PHY packet size - 2 bytes checksum). Of these 125 bytes only a fraction can be used for actual L3 payload as IEEE 802.15.4 headers, compressed IP headers and fragment headers also need to be transported.

If you're considering a professional application, you should sniff the packets and check whether they can be further optimized (using IEEE 802.15.4 short addresses, stateful 6LoWPAN compression, etc.). You could also consider contributing optimizations to Zephyr (e.g. enabling larger L2 MTUs which is possible in SubG bands). If you're interested, let me know.

Please be aware that TCP and HTTP are not an optimal fit for IEEE 802.15.4. Raw IPv6 is ok for routing support but payloads should be optimized for IEEE 802.15.4 like in the Thread protocol. If you need point-to-point (star topology) only, consider DGRAM sockets (i.e. L2 only).

fulup-bzh, 2024-08-22:
Increasing FRAGMENT_CACHE_SIZE remove initial error, but http request still fail in timeout.
```
```

fulup-bzh, 2024-08-22:
> > Increasing FRAGMENT_CACHE_SIZE remove initial error, but http request still fail in timeout.
> 
> Please include your log... :-)

Hoop!, This is the output of Zephyr HTTP-client Sample when using 6lowpan and increasing CONFIG_NET_L2_IEEE802154_FRAGMENT_REASS_CACHE_SIZE. HTTP_GET still fail.

fgrandel, 2024-08-22:
Please increase the timeout of your http_client_req() call. You may also consider switching to 2.4 G band as it has faster throughput. HTTP + TCP are really a stretch for SubG bands. ;-)

fulup-bzh, 2024-08-22:
@fgrandel  > If you're considering a professional application, you should sniff the packets and check whether they can be further optimized (using IEEE 802.15.4 short addresses, stateful 6LoWPAN compression, etc.). You could also consider contributing optimizations to Zephyr (e.g. enabling larger L2 MTUs which is possible in SubG bands). If you're interested, let me know

We are investigating 6lowpan in the context of EV-charger to check if this could be a valid option to transfer ISO15118 package. ISO15118 packet are typically 30-100bytes, but few as certificates update could potentially be few KB. The speed is not an issue, but if should still be able to update certificate & firmware update on the air. In both case stability/reliability is a must have.

fulup-bzh, 2024-08-22:
@fgrandel I already increase timeout to 10s, it does not change anything. In fact digging further, I see that Tcpdump shows many retransmission error from Zephyr(2001:db8::1)->Linux(2001:db8::2)
[image]
Full tcpdump log: [zephyr-6lowpan.tcpdump.zip](https://github.com/user-attachments/files/16710700/zephyr-6lowpan.tcpdump.zip)

Unfortunately even if config modification improve sample behavior, we are still far from a working model.

fgrandel, 2024-08-22:
HTTP + TCP do not seem the right solution then. Plain IEEE 802.15.4 (with or w/o 6LoWPAN) sounds like a good match, though, especially if you want SubG support for larger range and less interference:
* Consider using UDP or raw IPv6 sockets and ideally let higher layers ensure that collisions are avoided and missed packages will be re-sent.
* Do you need routing or could your border router reach your field devices in a single hop (star topology)? In the latter case you should use DGRAM sockets. This will considerably optimize throughput and/or duty cycle on the SubG band (no IP header). You're probably aware that in the SubG band duty cycle restrictions apply in some legislations.
* IEEE 802.15.4 reliability can be optimized with appropriate ACKs, resends and custom CSMA/CA backoff configuration.
* You may consider upgrading the driver to support larger MTUs for OTA and certificate transport. I'll help you with that.

In any case I suspect that if you want to build a professional application over SubG you'll have to dig into the lower protocols to get a reliable solution.

fgrandel, 2024-08-22:
> In fact digging further, I see that Tcpdump shows many retransmission error

Could be anything between bad antennae over collisions on the physical channel to timing errors or simply not enough bandwidth for parallel requests. Are you sniffing on the air or are you attaching wireshark on one side? If you want to know what's going on you'll have to tcpdump on both sides plus sniff in between.

Maybe I find time to reproduce this on the weekend but it's not really a use case that matters much for the community (see my comments re HTTP/TCP over IEEE 802.15.4).

fulup-bzh, 2024-08-22:
> In any case I suspect that if you want to build a professional application over SubG you'll have to dig into the lower protocols to get a reliable solution.

I unfortunately agree with your analyze, this effort would require more resources that I currently have. 

As DTLS is not really popular on EV side, moving ISO15118 out of TCP would probably require to tunnel everything through UDP with Wireshark and even then fragmentation will probably still be an issue.

Thank a lot for your support. If needed I would be more than happy to do more testing or ship you boards if you not have enough for testing. Unfortunately for this project [tux-evse](https://github.com/tux-evse) for the time being, I will probably rollback to more traditional network infrastructure.

PR #77370 (2024-08-26): Fix #77287: HTTP Client/Server: No NS over IEEE 802.15.4
Fixes #77287 by adding additional resources to the HTTP client sample.

Also adds a few minor debuggability improvements:
- improved logging for IEEE 802.15.4 net management and fragmentation
- support net capture for IPv6/v4 only

Adds IEEE 802.15.4 support to the HTTP client/server samples.
commit bc1b85c4 (2024-08-26): samples: net: http_client: increase resource requirements

Low throughput links (e.g. IEEE 802.15.4) require additional stack and
buffering resources. Therefore some stack and buffer pool sizes were
increased to cater for over-the-air scenarios.

Fixes: #77287
closed 2024-08-26 by carlescufi