zephyrproject-rtos/zephyr #77303: net: socket: recvmsg() doesn't update msg_controllen accordingly

axelnxp, 2024-08-20:
**Describe the bug**

The bug is exactly the same than it was with #68352.
This was fixed at the time, but I think the changes from #73713 introduced the regression.
`msg_controllen` is still set to the initial length, while `csmg_len` is set to 0.

If I bypass this added code, I don't reproduce the issue.

Maybe the problem comes from `add_timestamping` ? Does it update the `csmg_len` accordingly (if applicable) ?

**To Reproduce**
One can reuse the matter function where the issue is observed:
https://github.com/project-chip/connectedhomeip/blob/master/src/inet/UDPEndPointImplSockets.cpp#L611
And make sure no control data is received.

**Expected behavior**
`msg_controllen` field shall be updated by `recvmsg` according to the control data received.

**Impact**
Impact is high, in the current state, any Matter application over Zephyr would be impacted, likely unable to even perform a commissioning.

axelnxp, 2024-08-20:
cc @jukkar

jukkar, 2024-08-20:
Cc: @awojasinski please take a look, seems to be related to the ptp pr

axelnxp, 2024-08-21:
I've been able to get more info on the root cause.
In my case, `add_timestamping` returns `-ENOTSUP` because `net_context_get_option` sets `timestamping` to 0.
In this case, `csmg_len` is not updated, so `msg_controllen` should be cleared, as no pkt info has been added.

So the problem comes from the `clear_controllen` logic.

jukkar, 2024-08-21:
@axelnxp looks good, please send a PR for it.

axelnxp, 2024-08-21:
@jukkar done, refer to #77345

PR #77345 (2024-08-23): net: sockets: Update msg_controllen in recvmsg properly
If adding any control data like timestamping or pkt_info fails, the msg_controllen wasn't properly cleared.
This commit aims to fix this by clearing the msg_controllen only when adding control data is successful.

Fixes #77303

Author: @axelnxp 
(submitted on behalf because of some nxp github policies delaying his ability to post a PR)
commit 5d643f4b (2024-08-23): net: sockets: Update msg_controllen in recvmsg properly

According to recvmsg man page, msg_controllen should be set to the sum
of the length of all control messages in the buffer.
This is missing from the current recvmsg implementation.
This commit aims to fix this by updating msg_controllen each time control
data are added to the buffer.
This commit also fixes cases where the msg_controllen is cleared
incorrectly.

Fixes #77303
PR #77487 (2024-08-31): [Backport v3.7-branch] net: sockets: Update msg_controllen in recvmsg properly
Backport 179c85aa06d670ca48844a2e0d009fb3863a424e~2..179c85aa06d670ca48844a2e0d009fb3863a424e from #77345.

Fixes #77303
commit ae03e45b (2024-08-31): net: sockets: Update msg_controllen in recvmsg properly

According to recvmsg man page, msg_controllen should be set to the sum
of the length of all control messages in the buffer.
This is missing from the current recvmsg implementation.
This commit aims to fix this by updating msg_controllen each time control
data are added to the buffer.
This commit also fixes cases where the msg_controllen is cleared
incorrectly.

Fixes #77303

(cherry picked from commit 5d643f4b00da1572cb10494357197548c6f28883)
closed 2024-08-23 by fabiobaltieri