zephyrproject-rtos/zephyr #77325: net: ip: Frequent 4G/5G switching triggers rs_timeout infinite loop, causing WDT exception

AndyKauo, 2024-08-21:
**Is your enhancement proposal related to a problem? Please describe.**

The process of frequently switching between 4G and 5G triggers network interface up and down events, as well as initiates the RSRA flow of IPv6. When the network interface goes up, the function [net_if_start_rs()](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1489) appends `rs_node` to the list using [sys_slist_append](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1510). However, it may add the same node to the linked list multiple times (`rs_node = rs_node.next`), because `Z_GENLIST_APPEND` does not check if the same `rs_node` is already added, resulting in an infinite loop in [rs_timeout()](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1464) in `net_if.c`.

**Links to relevant code:**
- `net_if_start_rs()` [net_if.c: L1489](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1489)
- `sys_slist_append` in `net_if_start_rs()` [net_if.c: L1510](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1510)
- `Z_GENLIST_APPEND` [list_gen.h: L95](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/list_gen.h#L95)
- `rs_timeout()` [net_if.c: L1464](https://github.com/zephyrproject-rtos/zephyr/blob/7e65299a7e917fb2787b2333495e0a2c2b44caed/subsys/net/ip/net_if.c#L1464)

**Describe the solution you'd like**

There are two solutions as follows:
1. Modify the network flow of [notify_iface_down()](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_if.c#L5240) to add `net_if_stop_rs(iface)` at the end. This ensures the corresponding node is removed.

```c
static void notify_iface_down(struct net_if *iface)
{
    struct net_if_link_cb *link, *tmp;

    SYS_SLIST_FOR_EACH_CONTAINER_SAFE(&link_callbacks, link, tmp, node) {
        if (link->cb->iface_down) {
            link->cb->iface_down(iface, link->user_data);
        }
    }

+   net_if_stop_rs(iface); // Ensure the corresponding node is removed
}
```

2. Modify [Z_GENLIST_APPEND ()](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/list_gen.h#L95) to include a check for the new node's validity. We also found that Linux performs the same check in [list.h](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/include/linux/list.h#n150) on kernel.org as well.

```c
#define Z_GENLIST_APPEND(__lname, __nname)				\
	static inline void						\
	sys_ ## __lname ## _append(sys_ ## __lname ## _t *list,		\
				   sys_ ## __nname ## _t *node)		\
	{								\
		z_ ## __nname ## _next_set(node, NULL);			\
									\
		if (sys_ ## __lname ## _peek_tail(list) == NULL) {	\
			z_ ## __lname ## _tail_set(list, node);		\
			z_ ## __lname ## _head_set(list, node);		\
+		} else if (sys_ ## __lname ## _peek_tail(list) != node) { \
			z_ ## __nname ## _next_set(			\
				sys_ ## __lname ## _peek_tail(list),	\
				node);					\
			z_ ## __lname ## _tail_set(list, node);		\
		}							\
	}
```

jukkar, 2024-08-21:
> 2\. Modify [Z_GENLIST_APPEND ()](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/list_gen.h#L95) to include a check for the new node's validity. We also found that Linux [...]

Could you please send a PR having this solution as it is a more general one.

AndyKauo, 2024-08-21:
Thank you for your feedback and for accepting my suggestion. I am eager to submit a pull request with the proposed changes. However, due to firewall restrictions on my company's network, I am currently unable to create a PR. I will need to wait until the weekend to upload the changes from home.

AndyKauo, 2024-08-23:
Hi @jukkar @rlubos 
Would it be possible for you to help submit this PR? I'm still experiencing firewall issues and would greatly appreciate your assistance. Thank you very much.

```c
#define Z_GENLIST_APPEND(__lname, __nname)				\
	static inline void						\
	sys_ ## __lname ## _append(sys_ ## __lname ## _t *list,		\
				   sys_ ## __nname ## _t *node)		\
	{								\
		z_ ## __nname ## _next_set(node, NULL);			\
									\
		if (sys_ ## __lname ## _peek_tail(list) == NULL) {	\
			z_ ## __lname ## _tail_set(list, node);		\
			z_ ## __lname ## _head_set(list, node);		\
+		} else if (sys_ ## __lname ## _peek_tail(list) != node) { \
			z_ ## __nname ## _next_set(			\
				sys_ ## __lname ## _peek_tail(list),	\
				node);					\
			z_ ## __lname ## _tail_set(list, node);		\
		}							\
	}
```

jukkar, 2024-08-23:
Can't you submit the PR using HTTP endpoint in github i.e., having https://github.com/zephyrproject-rtos/zephyr.git as a git remote repo address? You can access github web UI so that URL should work ok.

As a last resort, we can certainly help, but it would be really good if you could do this yourself.

PR #88957 (2025-04-25): Avoid adding IPv6 RS multiple times to active timer list
Adding RS timer to active timers list multiple times can cause issues when traversing the list.

Fixes #77325
closed 2025-04-25 by kartben