zephyrproject-rtos/zephyr #77330: net: shell: conn_mgr shell compilation error

WilliamGFish, 2024-08-21:
**Description of bug**
Compilation error when using Net Shell

**To Reproduce**

**Expected behavior**
No compilation warnings

**Impact**
Compilation error, appears to still function correctly

**Logs and console output**

> [268/437] Building C object zephyr/subsys/net/CMakeFiles/subsys__net.dir/lib/shell/cm.c.obj
> ../zephyrNew/subsys/net/lib/shell/cm.c: In function 'cm_get_iface_info':
> ../zephyrNew/subsys/net/lib/shell/cm.c:371:2: warning: implicit declaration of function 'snprintf' [-Wimplicit-function-declaration]
>   371 |  snprintf(buf, len, "%d (%p - %s - %s)", net_if_get_by_iface(iface), iface, name,
>       |  ^~~~~~~~
> ../zephyrNew/subsys/net/lib/shell/cm.c:371:2: warning: incompatible implicit declaration of built-in function 'snprintf'
> ../zephyrNew/subsys/net/lib/shell/cm.c:21:1: note: include '<stdio.h>' or provide a declaration of 'snprintf'
>    20 | #include <zephyr/net/conn_mgr_monitor.h>
>   +++ |+#include <stdio.h>

**Environment (please complete the following information):**

-- CMake version: 3.25.0-rc2
-- Found Python3: C:\Python311\python.exe (found suitable version "3.11.9", minimum required is "3.8") found components: Interpreter
-- Zephyr version: 3.7.99 (/zephyrNew)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: particle_argon, qualifiers: nrf52840
-- Found toolchain: gnuarmemb (C:\gnu_arm_embedded_10)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")

**Additional context**

jukkar, 2024-08-21:
@glarsennordic please take a look

glarsennordic, 2024-08-21:
I'm having some difficulty reproducing, @WilliamGFish Are you building a specific sample by any chance that I can also build?

glarsennordic, 2024-08-21:
I do notice that `cm.c` appears to be an outlier for using `snprintf` as opposed to `snprintk`. So I will submit a patch to switch to that, and I'll try to figure out whatever header file needs to be included as well. It unfortunate that I cannot replicate the build warning myself, however

WilliamGFish, 2024-08-21:
Excellent you resolved before I'd read the original response. I'll try the fix to double check it. Billy..

PR #77364 (2024-08-24): net: shell: cm: Fix snprintf warnings on NEWLIBC
Building with NEWLIBC triggers warnings about
snprintf since stdio is no longer automatically
included by printk.h

This PR switches to using snprintk to avoid these
warnings.

Fixes #77330
commit a900d176 (2024-08-24): net: shell: cm: Fix snprintf warnings on NEWLIBC

Building with NEWLIBC triggers warnings about
snprintf since stdio is no longer automatically
included by printk.h

This PR switches to using snprintk to avoid these
warnings.

Fixes #77330
closed 2024-08-24 by nashif