zephyrproject-rtos/zephyr #77354: RTC_DS1307 driver zeroes-out seconds when initialising

clodnut, 2024-08-21:
**Describe the bug**

ds1307_init() zeroes-out the "SECONDS" field of the time in the DS1307 device, corrupting the time.

**Details**

ds1307_init() correctly clears the DS1307's "Clock Halt" bit which will be set when power is first applied to the DS1307.

However, it does this by [wiping out the entire DS1307_REG_SECONDS register](https://github.com/zephyrproject-rtos/zephyr/blob/34206f5505b582942ae4c393d7133a5ea6c849bd/drivers/rtc/rtc_ds1307.c#L142), which unconditionally zeroes-out the "seconds" field of the time:

```
	/* Make clock halt = 0 */
	err = i2c_reg_write_byte_dt(&config->i2c_bus, DS1307_REG_SECONDS, 0x00);
	if (err < 0) {
		LOG_ERR("Error: Set clock halt bit:%d\n", err);
	}
```

This operation should be a read/modify/write to clear only bit 7 ("CH") and only if that bit is set (so the time cannot change between read and write).

This would avoid corrupting the "seconds" field(s).

I suggest:

```
	/* Ensure Clock Halt bit is 0 */
	uint8_t reg = 0;
	err = i2c_reg_read_byte_dt(&config->i2c_bus, DS1307_REG_SECONDS, &reg);
	if (err < 0) {
		LOG_ERR("Error: Read SECONDS/Clock Halt register:%d", err);
	}
	if (reg & ~SECONDS_BITS){
		/* Clock Halt bit is set */
		err = i2c_reg_write_byte_dt(&config->i2c_bus, DS1307_REG_SECONDS, reg & SECONDS_BITS);
		if (err < 0) {
			LOG_ERR("Error: Write SECONDS/Clock Halt register:%d", err);
		}
	}
```

Tested informally and seems to work.

kartben, 2024-08-21:
Thanks for the report! Since you seem to have identified a fix, could you please submit a pull request? Thanks!

clodnut, 2024-10-27:
> This issue has been marked as stale...

Please remove the stale label.  The PR review is still underway.

PR #77727 (2024-11-08): Preserve SECONDS fields when clearing Clock Halt bit at initialisation (#77354)
drivers: rtc: rtc_1307.c

Read/modify/write the CH/SECONDS register at initialisation to clear only the Clock Halt bit and only if it is set.

This prevents the SECONDS field(s) from being wiped out unconditionally at initialisation.

Fixes #77354.
commit a2e920cf (2024-11-08): drivers: rtc: rtc_ds1307: Fix corruption of SECONDS register

We read/modify/write the CH/SECONDS register at initialisation to clear
only the Clock Halt bit and only if it is set.

The previous implementation zeroes the entire register unconditionally at
initialisation, which wipes the SECONDS fields.

Fixes #77354.
closed 2024-11-08 by mmahadevan108