zephyrproject-rtos/zephyr #77381: Flash area erase doesn't support full erase

srkapil, 2024-08-22:
**Describe the bug**

We have a custom sequence in our bootloader (mcuboot) that programs to the primary flash partition using a USB mass storage device. It does a flash erase by calling `flash_area_erase(fa, fa->fa_off, fa->fa_size);` where `fa->fa_off` is 0x40000 and `fa->fa_size` is 0x300000 (3MB slots) in our case. This should delete the whole partition but instead, It fails on the `is_in_flash_area_bounds` function. This is because `fa->fa_size` is being used to ensure that the area is within bounds but this should be `off+fa->fa_size` as that's where the end address of the flash area is. 

```c
static inline bool is_in_flash_area_bounds(const struct flash_area *fa,
					   off_t off, size_t len)
{
	return (off >= 0) && ((off + len) <= fa->fa_size);
}
```
> See `zephyr/subsys/storage/flash_map/flash_map.c` for `flash_area_erase` function definition. 

**To Reproduce**

Steps to reproduce the behavior:

1) Have flash with partitions in your DTS. 
1) Open the flash primary partition and erase it completely as below: 

```c
  const struct flash_area* fap;
  int ret = flash_area_open(1, &fap);
  if (ret != 0)
  {
    BOOT_LOG_INF("flash_open fail(%d)", ret);
    return ret;
  }

  ret = flash_area_erase(fap, fap->fa_off, fap->fa_size); // Erase the partition completely
  if (ret != 0) //fails 
  {
    BOOT_LOG_INF("Erase fail(%d)", ret);
    return ret;
  }
```

**Expected behavior**
The expected behaviour is `flash_area_erase` should delete the whole partition.

**Impact**
Unable to delete the flash area beyond the fa->fa_size.

**Environment (please complete the following information):**

 - OS:  Zephyr 3.7

nordicjm, 2024-08-22:
You're using the wrong parameters, this isn't a bug, the erase offset is the offset from the flash area, fa_off is the internal offset from the start of the flash device, so if you have a 3MiB flash device and partition starting at 1MiB of 1MiB then fa_off will be 1MiB, that doesn't mean you pass 1MiB into the erase function, it will rightfully fail because you are trying to erase at offset 1MiB from the partition, which would be 2MiB from the base flash device

closed 2024-08-22 by nordicjm

srkapil, 2024-08-22:
@nordicjm  Thanks, that makes sense. Sorry for the misunderstanding.