zephyrproject-rtos/zephyr #77387: Bluetooth: HCI: Disconnecting when ISO data path isn't set when there is a TX

Thalley, 2024-08-22:
**Describe the bug**
This was uncovered by https://github.com/zephyrproject-rtos/zephyr/pull/75549 and discussed in Discord. Below is the discussion from Discord:

[1:41 PM]Emil Gydesen (Thalley): @Johan Hedberg @Jonathan Rico @Aleksander Wasaznik @Vinayak Kariappa Chettimada 

I encountered a race condition / TX issue with the changes from https://github.com/zephyrproject-rtos/zephyr/pull/75549

This PR effectively makes setting up and removing the ISO data paths an API, so that the upper layers have better control of them. 

Initially this looked like a race condition, but after looking further into it, it looks like a HCI driver issue. 

First I'll explain the initial issue that I saw which is I though was a race condition: 

The race condition happens when

1) App connects CIS
2) App setups TX data path for CIS
3) App sends some ISO data that is then enqueued
4) App removes the data path
5)  bt_conn_tx_processor is then called but fails due to:

d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_ctlr_hci: hci_iso_handle: Input data path not set for HCI
d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_conn: send_buf: Unable to send to driver (err -22)
d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_conn: bt_conn_tx_processor: Fatal error (-5). Disconnecting 0x8258240
d_00: @00:00:15.694387  [00:00:15.694,366] <wrn> bt_conn: bt_conn_tx_processor: conn 0x8258240: not connected

This triggers a disconnection of the CIS. 

At first it looked like it would be possible and make sense to simply clear all pending data for the CIS when removing the data path, but that does not prevent applications from sending new data. 

In fact the issue is as simple as:
1) App connects CIS
2) App sends some ISO data (without setting up the data path). 

And if that happens, the CIS is then immediately disconnected, which seems a bit drastic for a simple rejection. 

When we send ISO data over HCI on async driver (e.g. IPC) then the controller can actually not reject the data at all, it can at best ignore it and send a command complete event, but if we send it over a synchronous driver like we have in this case, then it triggers a disconnect
GitHub
[Bluetooth: ISO: Make setting ISO data explicit by Thalley · Pull Re...](https://github.com/zephyrproject-rtos/zephyr/pull/75549)
The stack will no longer implicitly set the data path for ISO channel, and the responsibility for doing that is now for the upper layers/applications.
This provides additional flexibility for the h...
[1:45 PM]Emil Gydesen (Thalley): That is a huge difference in behavior based on the type of driver. 

We could, in the host, keep track of the ISO data paths to ensure that we never attempt to send anything to the controller, as well as clearing all data when removing the data path, but that's a significant change, and the state of the ISO data paths in the controller is non-trivial as it depends on what type of ISO channel it is (different behavior for Central CIS, Peripheral CIS, Broadcast source BIS and sync receiver BIS). 

The current driver and TX path seems to assume that all connections are equal, and that the data path is always there, which is an assumption that isn't true for ISO channels. 

I'm not sure what the right solution is, but IMO it makes sense to have similar behavior regardless of the driver and definitely not disconnecting the channel when removing the data path. 

Please share your thoughts
[2:28 PM]Aleksander Wasaznik: I agree, the integrated Controller's HCI driver has to behave the same as a normal HCI driver. It is not OK for bt_send to return any error value from the Controller. It's only for transport errors. Errors from the Controller have to come though a Command Status event or similar. As you say, for ACL/ISO data there is no error response available, so errors must be ignored.
[2:41 PM]Emil Gydesen (Thalley): @Aleksander Wasaznik will you or @Jonathan Rico take a look at fixing this? Would be nice if we can get that fixed before the 3.7 release if we consider it a bug
[8:22 AM]Aleksander Wasaznik: @Vinayak Kariappa Chettimada Do you agree the driver is at fault here?
[10:14 AM]Aleksander Wasaznik: The error is generated here in the Controller:

https://github.com/zephyrproject-rtos/zephyr/blob/9b305d5d72d1be6a648a82cc491f9806f34e6848/subsys/bluetooth/controller/hci/hci.c#L5882-L5885

I think we cannot just ignore/remove the error, since the Host is expecting a NumComplete back. Will it be generated despite the early return at the above-linked code site?
[10:25 AM]Vinayak Kariappa Chettimada: What does the spec say about num completes for non-existent data path? why is ISO data supplied without data path?
[10:37 AM]Aleksander Wasaznik: I think even if the Host violates some rule here, the Controller must do something to free the Controller buffer for the Host, in a way that works over HCI. The only options for this are a normal NumComplete, DisconnectionComplete, or HardwareError. Otherwise the buffer leaks.
[10:40 AM]Emil Gydesen (Thalley): I don't think a missing data path should be considered different than any other invalid data from the host (e.g. if sending ISO data with bad sequence number or flags)
[10:43 AM]Aleksander Wasaznik:
What does the spec say about num completes for non-existent data path? 

This is a good question. I can't find anything after looking for a few minutes. Still I think HCI buffer management should be a separate concern anyway, and the configured data path should not affect this.

As to the contents of the buffer, I suspect it's not specified, in which case it's up to us if we accept the data and send it despite no (or even otherwise configured) data path or discard it. 
[1:37 PM]Aleksander Wasaznik:
why is ISO data supplied without data path?
Using our Host API, the application can queue ISO data, and then de-configure the data path. The Host treats the two operations asynchronously, and can serialize them in opposite order onto HCI. It arguable whether this is OK; maybe the Host should not let this happen. But we think this uncovered a flaw in ZLL's HCI implementation.
[... 59 lines cut ...]
d_00: @00:00:15.694387  [00:00:15.694,366] <dbg> bt_bap_unicast_client: unicast_client_ep_qos_state: dir sink cig 0x00 cis 0x00 codec 0x06 interval 10000 framing 0x00 phy 0x02 rtn 2 latency 10 pd 40000
d_00: @00:00:15.694387  QoS set stream 0x84b0200
d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_ctlr_hci: hci_iso_handle: Input data path not set for HCI
d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_conn: send_buf: Unable to send to driver (err -22)
d_00: @00:00:15.694387  [00:00:15.694,366] <err> bt_conn: bt_conn_tx_processor: Fatal error (-5). Disconnecting 0x84f5020
d_00: @00:00:15.694387  [00:00:15.694,366] <wrn> bt_conn: bt_conn_tx_processor: conn 0x84f5020: not connected
d_00: @00:00:15.694825  [00:00:15.694,824] <dbg> bt_bap_stream: bt_bap_stream_disable: stream 0x84b0244
d_00: @00:00:15.694825  [00:00:15.694,824] <dbg> bt_bap_unicast_client: bt_bap_unicast_client_disable: stream 0x84b0244
d_00: @00:00:15.694825  [00:00:15.694,824] <dbg> bt_bap_unicast_client: unicast_client_ep_disable: ep 0x85206c4 buf 0x8520cfc
d_00: @00:00:15.694825  [00:00:15.694,824] <dbg> bt_bap_unicast_client: unicast_client_ep_disable: id 0x03
d_00: @00:00:15.694825  [00:00:15.694,824] <dbg> bt_bap_unicast_client: bt_bap_unicast_client_ep_send: conn 0x84e2e40 ep 0x85206c4 buf 0x8520cfc len 3
d_00: @00:00:15.741608  [00:00:15.741,607] <dbg> bt_iso: bt_iso_disconnected: 0x84f5020
d_00: @00:00:15.741608  [00:00:15.741,607] <dbg> bt_iso: bt_iso_chan_disconnected: 0x851ed20, reason 0x16
d_00: @00:00:15.741608  [00:00:15.741,607] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x851ed20 iso 0x84f5020 connected -> disconnected
d_00: @00:00:15.741608  [00:00:15.741,607] <dbg> bt_bap_unicast_client: unicast_client_ep_iso_disconnected: stream 0x84b0244 ep 0x85206c4 reason 0x16
d_00: @00:00:15.741608  Disconnected stream 0x84b0244 with reason 22
d_00: @00:00:15.741608  [00:00:15.741,607] <dbg> bt_bap_unicast_client: unicast_client_ep_iso_disconnected: stream 0x84b0200 ep 0x852031c reason 0x16
d_00: @00:00:15.741608  Disconnected stream 0x84b0200 with reason 22
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 34206f5505b582942ae4c393d7133a5ea6c849bd

**Additional context**
See also https://discordapp.com/channels/720317445772017664/1259836843374149784

jori-nordic, 2024-08-22:
WG:
- @alwa-nordic : All drivers should behave the same way. Since it's impossible for e.g. a UART-based HCI driver to return this error, the integrated drivers should also not raise it.
- @alwa-nordic : Fragmentation will make this situation worse. E.g. send a first fragment, data path is torn down, what do we do with the other fragments?
- @Thalley : What about changing the datapath when fragmenting. What happens then?
- @jori-nordic : We could test this situation
- @alwa-nordic : On `-EIO` (ie transport error), we should reset the driver, not just disconnect the link
- @alwa-nordic : On boot, the controller should send a HW error HCI event for synchronization

Thalley, 2025-01-21:
@alwa-nordic @cvinayak @jhedberg did something happen in the TX path that I am unaware of? My PR, https://github.com/zephyrproject-rtos/zephyr/pull/75549, is now passing CI :s

Thalley, 2025-02-05:
> [@alwa-nordic](https://github.com/alwa-nordic) [@cvinayak](https://github.com/cvinayak) [@jhedberg](https://github.com/jhedberg) did something happen in the TX path that I am unaware of? My PR, [#75549](https://github.com/zephyrproject-rtos/zephyr/pull/75549), is now passing CI :s

@alwa-nordic @cvinayak @jhedberg Would it make sense to put https://github.com/zephyrproject-rtos/zephyr/pull/75549 into review? It's passing CI now, even though I'm not sure _why_. Previously it was failing when the ISO data path was removed while the host had outstanding packets, but now it doesn't. Did something change, or just timing?

Thalley, 2025-02-05:
> It is ok to put the review ready, the failure will show up again, but that is for another day.

My worry is that it'll end up being a volatile change that will break CI once in a while until this has been properly fixed :/

Thalley, 2025-10-07:
Issue resurfaced https://github.com/zephyrproject-rtos/zephyr/actions/runs/18318328205/job/52164584619?pr=96800