zephyrproject-rtos/zephyr #77402: Enabling IPv6-forwarding on VLAN-based interface triggers deadloop & hang

hakehuang, 2024-08-22:
[Background]
In order to measure the network IP forwarding performance, we enabled IPv6 forwarding by setting the CONFIG_NET_ROUTING=y. And, reuse & customize the sample/net/vlan for testing because the on-hand HW board only has 1 Ethernet port. The customization is only adding 1 static route in vlan/src/main.c pointing to a gateway located in VLAN.200 virtual interface. The traffics will be injected from VLAN.100 with a DIP matched to the static route, and then should be forwarded to VLAN.200 (same physical port). Thus, we can use a Spirent TestCenter and use 1 port to Tx/Rx packets for measurement.

[Steps]
1)	Enable CONFIG_NET_ROUTING
------------------------------------------------------------------------------------------
# in file subsys/net/ip/Kconfig: (enabled NET_ROUTING by adding 1 line below)
```
 config NET_ROUTING
        bool
        depends on NET_ROUTE
+       default y if NET_ROUTE
        help
          Allow IPv6 routing between different network interfaces and
	------------------------------------------------------------------------------------------ 
```
2)	Add 1 static route in sample/net/vlan/src/main.c
-----------------------------------------------------------------------------------------
```
#include "ipv6.h"
#include "nbr.h"
#include "route.h"

/* Generic address that we are using to generate some more addresses */
static struct in6_addr netaddr = { { { 0x20, 0x01, 0x0d, 0xb8, 0x01, 0x01, 0, 0,
                                            0, 0, 0, 0, 0, 0, 0, 0 } } };

static struct in6_addr gw_addr = { { { 0x20, 0x01, 0x0d, 0xb8, 0x02, 0, 0, 0,
                                            0, 0, 0, 0, 0, 0, 0, 2 } } };

static char mac_addr[6] = {0x70, 0xb5, 0xe8, 0x65, 0x33, 0x05};

static int init_route(struct net_if *iface)
{
        struct net_route_entry *route_entry;
        struct net_nbr *nbr;
        struct net_linkaddr ll_addr = {0};

        ll_addr.len = 6U;
        ll_addr.type = NET_LINK_ETHERNET;
        ll_addr.addr = &mac_addr[0];

        nbr = net_ipv6_nbr_add(iface,
                               &gw_addr,
                               &ll_addr,
                               true,
                               NET_IPV6_NBR_STATE_REACHABLE);

        if (!nbr) {
                LOG_ERR("failed to add nbr.");
                return -1;
        }

        route_entry = net_route_add(iface,
                                    &netaddr, 64,
                                    &gw_addr,
                                    NET_IPV6_ND_INFINITE_LIFETIME,
                                    NET_ROUTE_PREFERENCE_LOW);

        if (!route_entry) {
                LOG_ERR("failed to add route.");
                //net_neighbor_remove(nbr);
                return -1;
        }

        LOG_INF("init route done.");

        return 0;
}

[... 124 lines cut ...]
//              net_if_tx(iface, pkt);
                return;
        }
```
	------------------------------------------------------------------------------------------
For VLAN interfaces, the data-send will loop 1st the virtual iface (VLAN-iface) and 2nd the physical iface with which VLAN attached. And the 2nd time data-send will actually send out the packets. However, ‘net_ipv6_prepare_for_send()’ code always changes the pkt->iface back to the ‘nbr->iface’ which is the virtual iface. So, the packets will be looped infinitely between the 1st time and the 2nd time, and system will hang.

[Suggested Fixes]
Two possible solutions:
a)	Simply change the ‘net_if_tx()’ 1st parameter from ‘net_pkt_iface(pkt)’ to ‘iface’,
b)	Give a precondition for calling ‘net_ipv6_prepare_for_send()’ – if no lladdr_dst.

-----------------------------------------------------------------------------------------
```
        /* If the ll dst address is not set check if it is present in the nbr
         * cache.
         */
        if (IS_ENABLED(CONFIG_NET_IPV6) && net_pkt_family(pkt) == AF_INET6) {
                if (!net_pkt_lladdr_dst(pkt)->addr)
                        verdict = net_ipv6_prepare_for_send(pkt);
        }
```
------------------------------------------------------------------------------------------ 

We’ve tried both can work well.

hakehuang, 2024-08-22:
add @shrek-wang who initial identify this issue and provide detailed analysis and fix suggestion.

hakehuang, 2024-08-22:
related to #77399 and #77401

carlescufi, 2024-09-04:
@hakehuang I checked with @jukkar and @rlubos and they have no plans to work on this. Are you planning to do so?

hakehuang, 2024-09-05:
> @hakehuang I checked with @jukkar and @rlubos and they have no plans to work on this. Are you planning to do so?

@shrek-wang will follow up.

shrek-wang, 2024-09-06:
I will prepare 1 PR for review soon.

PR #78375 (2024-09-18): net: Add onlink and forwarding check to IPv6-prepare
Deadloop happens when CONFIG_NET_ROUTING and VLAN are enabled. In function net_ipv6_prepare_for_send(), pkt->iface will be updated with net_pkt_set_iface(pkt, iface) in 2 scenarios:
1. ip_hdr->dst is onlink
2. check_route or nbr_lookup
 
VLAN is virtual-iface which attaches to a physical-iface. Each time a packet being sent to a VLAN port will invoke twice of the net_send_data(). The 1st time, pkt->iface is set to virtual iface and the 2nd time to physical iface. However in above 2 scenarios, at the 2nd time of calling the net_send_data(), t [... cut]
commit 0b24b960 (2024-09-18): net: Add onlink and forwarding check to IPv6-prepare

Deadloop happens when CONFIG_NET_ROUTING and VLAN are enabled.
In function net_ipv6_prepare_for_send(), pkt->iface will be
updated with net_pkt_set_iface(pkt, iface) in 2 scenarios:
1. ip_hdr->dst is onlink
2. check_route or nbr_lookup
VLAN is virtual-iface which attaches to a physical-iface. Each
time a packet being sent to a VLAN port will invoke twice of
the net_send_data(). The 1st time, pkt->iface is set to virtual
iface and the 2nd time to physical iface.
However in above 2 scenarios, at the 2nd time of calling the
net_send_data(), th [... cut]
closed 2024-09-18 by nashif