zephyrproject-rtos/zephyr #77404: build_only tests breed TestCases with NONE statuses

LukaszMrugala, 2024-08-22:
**Describe the bug**

When launching a `build_only` `TestSuite` via Twister, `TestCase`s of that suite will be marked with `TwisterStatus.NONE` (None), rather than expected `TwisterStatus.SKIP` (skipped). Tests must use `ztest` or `test` Harness.

**Platform:** QEMU_x86_64 in my tests, should be reproduced on any

**To Reproduce**
Steps to reproduce the behavior:
1. Run a `build_only` `TestSuite` that uses a `ztest` or `test` Harness.
2. Verify resultant `TestCase` statuses

**Expected behavior**
`TestCase`s have 'skipped' status.

**Impact**
Our test result database is filled with garbage data, making it harder for us to analyse Twister's changes.

**Logs and console output**
I have used a custom test that consists of one `build_only` `TestSuite` and one `TestCase` that always passes.
```JSON
{
    "environment":{
        "os":"posix",
        "zephyr_version":"v3.7.0-1378-g2b994656fe1d",
        "toolchain":"zephyr",
        "commit_date":"2024-08-21T11:24:51+00:00",
        "run_date":"2024-08-22T10:49:20+00:00",
        "options":{
            "testsuite_root":[
                "<ZEPHYR_HOME>/tests/ztest/temp/subgroup_skip"
            ],
            "platform":[
                "qemu_x86_64"
            ],
            "retry_failed":1,
            "verbose":1
        }
    },
    "testsuites":[
        {
            "name":"tests/ztest/temp/subgroup_skip/testing.ztest.temp.build_only_skip",
            "arch":"x86",
            "platform":"qemu_x86_64",
            "path":"tests/ztest/temp/subgroup_skip",
            "run_id":"960a0c290578706c50f55a60a8b871df",
            "runnable":false,
            "retries":0,
            "status":"passed",
            "execution_time":"0.00",
            "build_time":"17.19",
            "testcases":[
                {
                    "identifier":"testing.ztest.temp.build_only_skip.skip",
                    "execution_time":"0.00",
                    "status":"None"
                }
            ]
        }
    ]
}
```

**Environment (please complete the following information):**

 - **OS:** Linux
 - **Toolchain:** Zephyr SDK
 - **Commit SHA or Version used:** v3.7.0-1378-g2b994656fe1d (modified for inline `TestCase` status printing)

**Additional context**
Basic problem lies in these two code parts:
https://github.com/zephyrproject-rtos/zephyr/blob/67b7c993e19837335392c406e225217d4a5b3e06/scripts/pylib/twister/twisterlib/runner.py#L646-L675
https://github.com/zephyrproject-rtos/zephyr/blob/67b7c993e19837335392c406e225217d4a5b3e06/scripts/pylib/twister/twisterlib/runner.py#L729-L766

In `runner.py`, `process()` method uses `self.build()`. Inside there, Twister checks whether the `TestInstance` was made to `run` and if not, it will assign the `SKIP` status to all its `TestCase`s that didn't have proper statuses yet via `add_missing_case_status()`.

However, after that assignment, if we go into the `determine_testcases()` line, we forgo all that work to instead delete and recreate those `TestCase`s. Those recreations have no statuses and will shamble on with `NONE` until Twister terminates.

nashif, 2024-08-27:
@LukaszMrugala  this was introduced in your PR introducing the status class...

LukaszMrugala, 2024-08-27:
> @LukaszMrugala this was introduced in your PR introducing the status class...

@nashif Which of my commits would have introduced this?

We encounter NONE/None statuses in our validation since 2023-09-23, almost a year ago. Long before the Status class PR. I see that the same `determine_testcases()` logic is kept since its introduction in 94f2d2437f17203bd658ba19a4ea70170dd3841d, two years ago.

nashif, 2024-08-27:
Look at  
[image]

what was merged at that time? https://github.com/zephyrproject-rtos/zephyr/pull/71401

I still need to look what exactly causes this, but that is a good indicator...

LukaszMrugala, 2024-08-28:
Is this a chart of `'None'` statuses? Of course those started since my Isolate Status Class PR. Previously they were `None` statuses (note the `str` vs `NoneType` type difference) and my PR brought type consistency in that area.

Is this chart showing null values correctly?

nashif, 2024-08-28:
> Is this chart showing null values correctly? Additionally, this Issue talks about `TestCase` statuses, not `TestSuite` and the chart axes talk about `TestSuite`s.

chart is correct, showing testcase statuses, despite the label.

I just looked in details, this is not a bug, the move from null to None just exposes the status differently. The goal was to improve this as part of the overall status overhaul, see https://github.com/zephyrproject-rtos/zephyr/issues/58070. Plan is to introduce a new status, in the proposal it is called 'NOT RUN' to distinguish between a testcase that was skipped intentionally, i.e. it was decided at run time that 1 test out of many was skipped and something that was never run and potentially could run but for some reason it can't.

arikgreen, 2024-08-28:
> Look at [...]

which index includes this data in our kibana?

nashif, 2024-08-28:
our kibana? 
This is upstream data: https://kibana.zephyrproject.io/app/dashboards#/view/25cd6690-9e6f-11ed-9e05-812a086fdb7e?_g=(filters:!(),refreshInterval:(pause:!t,value:0),time:(from:now-30d%2Fd,to:now))

arikgreen, 2024-08-28:
> our kibana? 
http://reports-fmos.hf.intel.com/

closed 2024-10-17 by carlescufi