zephyrproject-rtos/zephyr #77421: ci: check_compliance is broken if run it inside the zephyr\scripts\ci directory

butok, 2024-08-22:
**Describe the bug**

check_compliance.py is broken if run it inside the zephyr\scripts\ci directory on Windows

**To Reproduce**

Steps to reproduce the behavior:
1. Go to the zephyr\scripts\ci directory
2. Run in command line: python check_compliance.py
3. The KeepSorted test module raises a Python exception.

**Expected behavior**

No Python exceptions.

**Impact**
It is not possible to run the compliance tests in the zephyr\scripts\ci directory on Windows.

**Logs and console output**
```
C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci>python check_compliance.py
Running Identity         tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running GitDiffCheck     tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running KconfigHWMv2     tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running Kconfig          tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running ClangFormat      tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Checkpatch       tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Gitlint          tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Pylint           tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running ModulesMaintainers tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running ImageSize        tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running BinaryFiles      tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Nits             tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running KconfigBasic     tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running DevicetreeBindings tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running BoardYml         tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running KeepSorted       tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Python exception in `C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py`:

Traceback (most recent call last):
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1769, in main
    n_fails = _main(args)
              ^^^^^^^^^^^
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1704, in _main
    test.run()
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1548, in run
    with open(file, "r") as fp:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'scripts/ci/coverage/coverage_analysis.py'

Traceback (most recent call last):
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1796, in <module>
    main(sys.argv[1:])
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1769, in main
    n_fails = _main(args)
              ^^^^^^^^^^^
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1704, in _main
    test.run()
  File "C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr\scripts\ci\check_compliance.py", line 1548, in run
    with open(file, "r") as fp:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'scripts/ci/coverage/coverage_analysis.py'
```

**Environment (please complete the following information):**

OS: Windows
SHA-1: 65051c8a242e4bb5dd2c0625756d468c716b4616

**Additional context**
If to run it from the top directory, no errors:

```
C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr>python ./scripts/ci/check_compliance.py
Running Nits             tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Gitlint          tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running ClangFormat      tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running MaintainersFormat tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running KeepSorted       tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running KconfigBasicNoModules tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running KconfigBasic     tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running Identity         tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running ModulesMaintainers tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running GitDiffCheck     tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running DevicetreeBindings tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running KconfigHWMv2     tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running Checkpatch       tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Pylint           tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running Kconfig          tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running BoardYml         tests in C:\_ddm\ZEPHYR\review\butok_gh_3\zephyr ...
Running YAMLLint         tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running ImageSize        tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...
Running BinaryFiles      tests in C:/_ddm/ZEPHYR/review/butok_gh_3/zephyr ...

Complete results in compliance.xml
```

butok, 2024-08-23:
> According to the documentation the correct way to use it is to run it from the top directory.
> 
> https://docs.zephyrproject.org/latest/contribute/guidelines.html#check-compliance-py
> 
> > ./scripts/ci/check_compliance.py -c upstream/main..

Understand. However
- it does not claim that this is the only correct way.
- only the KeepSorted test module is problematic. All other modules handle this case correctly.

henrikbrixandersen, 2024-08-23:
> Understand. However
> 
> * it does not claim that this is the only correct way.
> * only the KeepSorted test module is problematic. All other modules handle this case correctly.

So since this works as intended (when following the instructions), this is an enhancement, not a bug?

butok, 2024-08-23:
> > Understand. However
> > 
> > * it does not claim that this is the only correct way.
> > * only the KeepSorted test module is problematic. All other modules handle this case correctly.
> 
> So since this works as intended (when following the instructions), this is an enhancement, not a bug?

OK. It can be marked as a place for improvement.

nashif, 2025-06-18:
i am not sure what this will be enhancing, and why we should be tracking this as enhancement, if you have a usecase, just submit a PR adding this functionality.

closed 2025-06-18 by nashif

kartben, 2025-06-18:
> Understand. However
> - it does not claim that this is the only correct way.
> - only the KeepSorted test module is problematic. All other modules handle this case correctly.

FWIW there are many other modules than KeepSorted that will give all kind of bad errors when you run from anywhere but the top folder, no matter which OS you're using

butok, 2025-06-19:
It's not critical, we have other higher priority tasks ;)