zephyrproject-rtos/zephyr #77436: NXP S32 CANXL the calculating initial timing is not as accurate as expected

congnguyenhuu, 2024-08-23:
**Describe the bug**

NXP S32 CANXL the calculating initial timing was not as accurate as expected.
Incase there is using different parameters bitrate and sample point to calculate the timing.
The calculating prescaler of the nominal bit timing and the data phase bit timing are not same, see logs and console output below (this is not as expectation).
Due to the hardware CANXL peripheral use the same bit time quanta prescaler for the nominal bit timing and the data phase bit timing.

**To Reproduce**

Run `tests/drivers/can/api` with `CONFIG_CAN_LOG_LEVEL_DBG=y`
```
west build -p -b s32z270dc2_rtu0_r52 tests\drivers\can\api -DCONFIG_CAN_LOG_LEVEL_DBG=y
...
west flash -r trace32
```

**Expected behavior**

The prescaler of calculating initial timing should be same for both timing.

**Impact**

The calculating initial timing was not as accurate as expected. It is possible to impact to the CANXL functionality when running external traffic. 

User need to calculate-self the timing and use  CAN APIs `set_timing ` and `set_timing_data` to re-configure the timing before starting to operate CAN, run application.

**Logs and console output**

The prescaler of the data phase bit timing (`= 4`) is not equal this one of the nominal bit timing (`= 40`).
```
D: Setting CAN bitrate 125000:
D: propSeg: 6, phase_seg1: 7, phase_seg2: 2, prescaler: 40, sjw: 1
D: Setting CAN FD bitrate 1000000:
D: propSeg: 7, phase_seg1: 7, phase_seg2: 5, prescaler: 4, sjw: 2
*** Booting Zephyr OS build v3.7.0-1481-g20665c28d7d1 ***
Running TESTSUITE can_classic
===================================================================
...
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used Zephyr 3.7.0

**Proposal**

Option 1: create a function that calculates the data phase bit timing so that the prescaler of the data phase bit timing must be equal this one of the nominal bit timing.

Option 2: revert commit 0c13f3888 `dts: bindings: can: remove deprecated properties for initial timing` that allows user calculate-self the timing and set timing over dts properties.

manuargue, 2024-10-14:
Assigning to CAN area maintainer

henrikbrixandersen, 2024-10-15:
Thank you for reporting this. I have lowered priority to "low" as this affects only one driver on one platform.

henrikbrixandersen, 2025-01-22:
I have not forgotten about this issue. I am currently monitoring the outcome of a similar discussion for the Linux kernel CAN subsystem: https://lore.kernel.org/linux-can/20250102213631.765503-2-stefan.maetje@esd.eu/