zephyrproject-rtos/zephyr #77452: ESP32S3 crashes when sending image over UDP

vojtechj, 2024-08-23:
**Describe the bug**
ESP32S3 crashes when sending image over UDP, but the ESP32C3 works well.

**To Reproduce**
add the following lines to `samples/subsys/mgmt/mcumgr/smp_svr/prj.conf` to be able to connect to wifi:
```
CONFIG_WIFI=y
CONFIG_INIT_STACKS=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SHELL=y
CONFIG_NET_L2_WIFI_SHELL=y
```

in `boards/esp32s3_devkitc_procpu.overlay` enable `usb_serial` and `wifi`:
```
/ {
    chosen {
        zephyr,console = &usb_serial;
        zephyr,shell-uart = &usb_serial;
      };
};

&usb_serial {
    status = "okay";
};

&wifi {
	status = "okay";
};
```

build for board `esp32s3_devkitc/esp32s3/procpu` and add UDP conf: `-DEXTRA_CONF_FILE=overlay-udp.conf`

when built and assuming mcuboot flashed and connected to wifi:
flash via UDP: `newtmgr.exe --conntype udp --connstring=[192.168.0.100]:1337 image upload build\zephyr\zephyr.signed.confirmed.bin`

**Expected behavior**
to be successfully flashed via UDP like it is possible with esp32c3_devkitm

**Logs and console output**
```
uart:~$ Connected
uart:~$ [00:01:10.603,000] <err> os:  ** FATAL EXCEPTION
uart:~$ [00:01:10.603,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
uart:~$ [00:01:10.603,000] <err> os:  **  PC 0x40378868 VADDR 0xffffffef
0x40378868: xtensa_restore_high_regs at D:/Downloaded/programming/zp/zephyr/arch/xtensa/core/xtensa_asm2_util.S:109

uart:~$ [00:01:10.603,000] <err> os:  **  PS 0x60920
uart:~$ [00:01:10.603,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:9 CALLINC:2)
uart:~$ [00:01:10.603,000] <err> os:  **  A0 0x40378894  SP 0x3fc9999c  A2 0xffffffef  A3 0xffffffff
uart:~$ [00:01:10.603,000] <err> os:  **  A4 0  A5 0  A6 0  A7 0x3fcb309c
uart:~$ [00:01:10.603,000] <err> os:  **  A8 0x8037abb7  A9 0x3fc9a920 A10 0 A11 0x3fcaf3d8
uart:~$ [00:01:10.603,000] <err> os:  ** A12 0x3fcb391c A13 0x3fc9a95c A14 0 A15 0x420155e4
0x40378894: _restore_context at D:/Downloaded/programming/zp/zephyr/arch/xtensa/core/xtensa_asm2_util.S:154

0x420155e4: zsock_received_cb at D:/Downloaded/programming/zp/zephyr/subsys/net/lib/sockets/sockets.c:405

uart:~$ [00:01:10.603,000] <err> os:  ** LBEG 0x4202024c LEND 0x42020290 LCOUNT 0x5
0x4202024c: net_context_is_used at D:/Downloaded/programming/zp/zephyr/include/zephyr/net/net_context.h:410
 (inlined by) find_context at D:/Downloaded/programming/zp/zephyr/subsys/net/ip/net_context.c:1057
 (inlined by) net_context_packet_received at D:/Downloaded/programming/zp/zephyr/subsys/net/ip/net_context.c:2407

0x42020290: find_context at D:/Downloaded/programming/zp/zephyr/subsys/net/ip/net_context.c:1056
 (inlined by) net_context_packet_received at D:/Downloaded/programming/zp/zephyr/subsys/net/ip/net_context.c:2407

uart:~$ [00:01:10.603,000] <err> os:  ** SAR 0xd
uart:~$ [00:01:10.603,000] <err> os:  **  THREADPTR 0x3fcb2d24
uart:~$ [00:01:10.603,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
uart:~$ [00:01:10.603,000] <err> os: Current thread: 0x3fcaf3d8 (unknown)
uart:~$ [00:01:10.694,000] <err> os: Halting system
```

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.8
 - Version: 3.7.0

sylvioalves, 2024-09-27:
@vojtechj I submmited a fix to solve the issue. However, it is also necessary/recommended that you add the following to your app build (values could be optimized..):

```
# increase UDP stack size
CONFIG_MCUMGR_TRANSPORT_UDP_STACK_SIZE=2048

#increase sys workq size
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

# increase mcumgr net buffer size
CONFIG_MCUMGR_TRANSPORT_NETBUF_SIZE=4096
CONFIG_MCUMGR_TRANSPORT_NETBUF_COUNT=4
CONFIG_MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE=32

CONFIG_NET_DHCPV4=y
CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y
CONFIG_NETWORKING=y
```

vojtechj, 2024-09-28:
Great, thank you a lot.

vojtechj, 2024-09-30:
> @vojtechj I submmited a fix to solve the issue. However, it is also necessary/recommended that you add the following to your app build (values could be optimized..):
>
> ```
> # increase UDP stack size [...]

do these config recommendations apply for esp32 c3 as well? and where can I watch the submitted fix?

vojtechj, 2024-10-22:
@sylvioalves do these config recommendations apply for esp32 c3 as well? and where can I watch the submitted fix?

sylvioalves, 2024-10-22:
@vojtechj humm, this issue should be closed already. Not sure why it is still opened. Fix is already merged, both in v3.7-branch and main. Have you tried already? Recommendations follows as is for any ESP32 model.

vojtechj, 2024-10-22:
Thank you for the fix. A will try it.

closed 2024-10-22 by vojtechj