zephyrproject-rtos/zephyr #77475: Legacy pairing with OOB pairing request wrong MITM  flag set

DevAljosa, 2024-08-23:
**Describe the bug**
We're using Zephyr version 2.6.0 with NCS (It's same with latest Zephyr version on the date 08/23/2024). Due to some technological limitations we had to develop BLE 4.1 peripheral device which supports only legacy pairing with OOB.

To debug connection and pairing procedure we're using Wireshark where we noticed following behaviour. When device connects we try to elevate security level to Level 3. Legacy pairing flag is set to true and we're enforcing MITM. However the log in wireshark shows that our requests never sets MITM flag. 

**To Reproduce**

We set CONFIG_BT_SMP_ENFORCE_MITM=y in our prj.conf, conn->required_sec_level is set to BT_SECURITY_L3 by call of bt_conn_set_security(conn, BT_SECURITY_L3), and make sure that appropriate callbacks for pairing are set/unset to result in BT_SMP_IO_NO_INPUT_OUTPUT.

Inside original Zepyhr function get_auth in file smp.c there are lines:

```	
       if ((get_io_capa(smp) == BT_SMP_IO_NO_INPUT_OUTPUT) ||
	    (!IS_ENABLED(CONFIG_BT_SMP_ENFORCE_MITM) &&
	    (conn->required_sec_level < BT_SECURITY_L3))) {
		auth &= ~(BT_SMP_AUTH_MITM);
	} else {
		auth |= BT_SMP_AUTH_MITM;
	}
```

**Expected behavior**
In our case get_io_capa(smp) is BT_SMP_IO_NO_INPUT_OUTPUT, CONFIG_BT_SMP_ENFORCE_MITM is set to TRUE and conn->required_sec_level=BT_SECURITY_L3 which we expect that it would result in setting auth to auth |= BT_SMP_AUTH_MITM;

However this code results in auth &= ~(BT_SMP_AUTH_MITM);

As we understand the code the control expression is like X || Y && Z.

We believe there are missing parenthesis to make this expression like  (X || Y)  && Z so that the fixed code would become

```	
       if (((get_io_capa(smp) == BT_SMP_IO_NO_INPUT_OUTPUT) ||
	    (!IS_ENABLED(CONFIG_BT_SMP_ENFORCE_MITM)) &&
	    (conn->required_sec_level < BT_SECURITY_L3))) {
		auth &= ~(BT_SMP_AUTH_MITM);
	} else {
		auth |= BT_SMP_AUTH_MITM;
	}
```

In that case the result would be expected - MITM flag set according to the BLE standard.

**Impact**
It wrongly sets MITM flag which might cause pairing issues with central. It's hard to bypass it or make a workaround without modifying the Zephyr.

sjanc, 2024-08-23:
Hi,

I think this works as expected, CONFIG_BT_SMP_ENFORCE_MITM=y enforce MITM flag only when IO capabilities allows for authenticated pairing (which is not the case with NoInputNoOutput)

DevAljosa, 2024-08-24:
Thank you for your fast reply. Now I understand and I agree with that. One thing I forgot to mention is that we are using legacy pairing with OOB. That's the reason why we have NoInputNoOutput. Therfore this will always evaluate to cleared MITM flag.  But since we use OOB this is considered as safe for MITM therfore to my understanding MITM flag should be set but there's no check for OOB in use. 

I assume there should be additional check prior this to see if OOB is used  and the original code would become something like this:

```
if (is_oob_pairing_used(smp)) {
    auth |= BT_SMP_AUTH_MITM;     // Set the MITM flag for OOB pairing
} else if (((get_io_capa(smp) == BT_SMP_IO_NO_INPUT_OUTPUT) ||
            (!IS_ENABLED(CONFIG_BT_SMP_ENFORCE_MITM)) &&
            (conn->required_sec_level < BT_SECURITY_L3))) {
    auth &= ~(BT_SMP_AUTH_MITM);  // Clear the MITM flag
} else {
    auth |= BT_SMP_AUTH_MITM;     // Set the MITM flag
}
```

sjanc, 2024-08-26:
Hi,

MITM flag is used to select in-band pairing method based on IO capabilities. When both sides have OOB bit set than OOB pairing method is used regardless of MITM flag.  And if OOB channel is secure from eavesdropping, the key generated with OOB pairing is assumed to be authenticated.

Please check Core Specification 5. Vol3 Part H "2.3.5.1 Selecting key generation method"

DevAljosa, 2024-08-26:
Hi,

All clear now. Thank you for your help and explanation and I am sorry for bothering you!

Case closed.
Thanks!

jhedberg, 2024-08-26:
@sjanc @DevAljosa thanks for bringing the matter to a mutual understanding. I'll close it now.

closed 2024-08-26 by jhedberg