zephyrproject-rtos/zephyr #77482: MIC Failures during ISO streaming

fredrikdanebjer, 2024-08-23:
**Describe the bug**

When starting to stream unicast audio on a audio shell test app the endpoint will get disconnected almost immediatly after reaching streaming state. The forwarded error code is 0x3d, or MIC Failure.

Looking at Ellisys traces the error is not sent from the Central, but generated inside the audio shell test app.

Changing the controller away from Zephyr solves the problem, even when running same application.

I have tried various commit between the tip of Zephyr back to commit bde663f4840f79601d10dd74a7d11ffc84e9c0a7, all fails consistently with the same issue.

Platform being used is a NRF53 Audio Dev Kit, and a Pixel 8 with Android 15 (AP31.240517.031) (although several models have been tested)

**To Reproduce**
On tip of Zephyr main:

1. Build controller and app as such:
west build -b nrf5340_audio_dk_nrf5340_cpuapp -d build/nrf5340audiodk/audio_shell tests/bluetooth/shell/ --pristine -- -DCONF_FILE=audio.conf -DOVERLAY_CONFIG=boards/nrf5340_audio_dk_nrf5340_cpuapp.conf

west build -b nrf5340_audio_dk_nrf5340_cpunet -d build/nrf5340audiodk/hci_ipc samples/bluetooth/hci_ipc --pristine -- -DCONF_FILE=nrf5340_cpunet_iso-bt_ll_sw_split.conf
 
2. Flash both controller and app.

3. Make sure the phone is not bonded with the target device

4. In shell application issue the following commands:
bt init
bt clear all
bap init
bt advertise on

5. On the phone
Connect to shell
Enable LE Audio
Start streaming through youtube

6. Notice that as soon as you enter streaming state, the endpoint will be disconnected due to reason 0x3d

**Expected behavior**
Streaming goes into streaming state, iso packets are received, perpetually until the stream is shut down by user.

**Impact**
Annoyance, and hindrance in development.

**Logs and console output**
Relevant log:
[00:03:16.477,478] <dbg> bt_bap_stream: bt_bap_stream_start: stream 0x2001e5a8 ep 0x2003b1fc
[00:03:17.074,707] <dbg> bt_ascs: ascs_iso_accept: conn 0x2000a948
[00:03:17.074,707] <dbg> bt_ascs: ascs_iso_accept: iso_chan 0x20043e0c
[00:03:17.216,033] <dbg> bt_ascs: ascs_ep_iso_connected: stream 0x2001e5a8 ep 0x2003b1fc dir sink
[00:03:17.216,094] <dbg> bt_ascs: ascs_ep_get_status: ep 0x2003b1fc id 0x01 state streaming
[00:03:17.216,125] <dbg> bt_ascs: ascs_ep_get_status_enable: dir sink cig 0x01 cis 0x00
[00:03:17.216,217] <dbg> bt_ascs: state_transition_work_handler: ase 0x2003b1f8 ep 0x2003b1fc id 0x01 enabling -> streaming
Stream 0x2001e5a8 started
[00:03:17.336,395] <dbg> bt_ascs: ascs_ep_iso_disconnected: stream 0x2001e5a8 ep 0x2003b1fc state streaming reason 0x3d
[00:03:17.336,456] <dbg> bt_ascs: ascs_ep_get_status: ep 0x2003b1fc id 0x01 state qos-configured
[00:03:17.336,486] <dbg> bt_ascs: ascs_ep_get_status_qos: dir sink codec id 0x06 interval 10000 framing 0x00 phy 0x02 rtn 0 latency 60 pd 10000
[00:03:17.336,578] <dbg> bt_ascs: state_transition_work_handler: ase 0x2003b1f8 ep 0x2003b1fc id 0x01 streaming -> qos-configured
Stream 0x2001e5a8 stopped with reason 0x3D
Release: stream 0x2001e5a8
Stream 0x2001e5a8 released

cvinayak, 2024-08-26:
I am guessing the phone is using interleaved packing, and there is no support for it in Zephyr Controller.

mmahadevan108, 2024-08-27:
@cvinayak, should this be a feature request instead of a bug?

cvinayak, 2024-08-27:
@fredrikdanebjer could you please tag this as enhancement request if the failure is due to use of interleaved packing?

fredrikdanebjer, 2024-08-27:
@cvinayak I will try to find out during the week if this actually is the case, and change accordingly then, if that's ok.

fredrikdanebjer, 2024-08-28:
@cvinayak we are fairly certain that this issue is not due to interleaved packing. We see this issue using only a single one-way CIS, where interleaving should be irrelevant.

Actually I think that the above described steps will always produce a one-way CIS. However, it has also been tested with a shell app where the source ase has been removed, and it only exposes a single sink ase.

PR #78096 (2024-10-11): Bluetooth: Controller: Fix CIS payload count under skipped events
Fix CIS payload count calculation under skipped events that caused CIS disconnection with reason MIC failure.

Fixes #77482.
closed 2024-10-11 by nashif