zephyrproject-rtos/zephyr #77496: tests: kernel.timer.timer : all NXP kinetis platform build failure

hakehuang, 2024-08-23:
**Describe the bug**

tests/kernel/timer/timer_behavior build failure for all Kinetis platforms
frdm_kl25z/frdm_kw41z/frdm_k22f/frdm_k82f/frdm_ke15z/frdm_k64/frdm_ke17z512/frdm_ke17z

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - > frdm_kl25z/frdm_kw41z/frdm_k22f/frdm_k82f/frdm_ke15z/frdm_k64/frdm_ke17z512/frdm_ke17z
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 ->

```
f58a4c4b1ca3b2378c3c5fe2ad0dbce777bdf9b0 is the first bad commit
commit f58a4c4b1ca3b2378c3c5fe2ad0dbce777bdf9b0
Author: Arkadiusz Cholewinski <arkadiuszx.cholewinski@intel.com>
Date:   Tue Aug 13 12:14:52 2024 +0200

    ztest: expand ztest functionality.
    
    Expanding the ZTEST shell to include a command that
    accepts additional parameters, such as ztest run_testcase
    <suite:test> -p any_value.
    
    Adding the new ZTEST_P macro, to create the
    parameterized test.
    
    Signed-off-by: Arkadiusz Cholewinski <arkadiuszx.cholewinski@intel.com>

```

**To Reproduce**

scripts/twister -p frdm_k64f -T tests/kernel/timer/timer_behavior  --build-only

**Expected behavior**

build pass

**Impact**

kernel time API

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/kernel/timer/timer_behavior/src/main.c: In function 'test_main':
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test.h:589:9: error: too few arguments to function 'z_ztest_run_test_suite'
  589 |         z_ztest_run_test_suite(STRINGIFY(suite), shuffle, suite_iter, case_iter)
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/kernel/timer/timer_behavior/src/main.c:11:9: note: in expansion of macro 'ztest_run_test_suite'
   11 |         ztest_run_test_suite(timer_jitter_drift, false, 1, 1);
      |         ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test.h:348:5: note: declared here
  348 | int z_ztest_run_test_suite(const char *name, bool shuffle, int suite_iter,
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test.h:589:9: error: too few arguments to function 'z_ztest_run_test_suite'
  589 |         z_ztest_run_test_suite(STRINGIFY(suite), shuffle, suite_iter, case_iter)
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/tests/kernel/timer/timer_behavior/src/main.c:13:9: note: in expansion of macro 'ztest_run_test_suite'
   13 |         ztest_run_test_suite(timer_tick_train, false, 1, 1);
      |         ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/subsys/testsuite/ztest/include/zephyr/ztest_test.h:348:5: note: declared here
  348 | int z_ztest_run_test_suite(const char *name, bool shuffle, int suite_iter,
      |     ^~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-1593-g143b14bb4fb7

golowanow, 2024-08-23:
observed on CI
https://github.com/zephyrproject-rtos/zephyr/actions/runs/10529406791/job/29179330904?pr=77498

@gbarkadiusz, this might be caused by recently merged #75457

avolkov-1221, 2024-08-23:
I confirm that this bug has also broken my PR.

https://github.com/zephyrproject-rtos/zephyr/actions/runs/10533304984/job/29189179084?pr=75740.

nordic-piks, 2024-08-26:
Observed also for NRF platforms (all).

gbarkadiusz, 2024-08-26:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/77508

mmahadevan108, 2024-08-27:
@hakehuang, can you please retest with the latest code and confirm if this issue has been fixed.

hakehuang, 2024-08-28:
> @hakehuang, can you please retest with the latest code and confirm if this issue has been fixed.

@mmahadevan108, retesting, issue is fixed

closed 2024-08-28 by hakehuang