zephyrproject-rtos/zephyr #77516: stm32f7: rtc not keeping time on power off with battery

paradajz, 2024-08-24:
I am using `nucleo_f767zi` board. I have an external RTC clock and I'm trying to use it for the RTC:

overlay:
```
&clk_lse {
    status = "okay";
};

&rtc {
    status = "okay";
    clocks = <&rcc STM32_CLOCK_BUS_APB1 0x10000000>,
             <&rcc STM32_SRC_LSE RTC_SEL(2)>;

    backup_regs {
        status = "okay";
    };
};
```

(side note: I have no idea what RTC_SEL does and if it's even setup correctly).

config:
```
CONFIG_RTC=y
```
Everything seems to work fine during runtime. I can read the time without issues, and also when I reset the board, the time is kept. However once I remove the power supply and then bring it back on, the `rtc_get_time` is unable to retrieve the time so I have to set it again. I am not sure if I am missing anything obvious here.

Zephyr version: v3.7.0

erwango, 2024-08-26:
> However once I remove the power supply and then bring it back on, the rtc_get_time is unable to retrieve the time so I have to set it again. I am not sure if I am missing anything obvious here.

This is expected, see ref man (RM0385, §29.1 Introduction)
```
[...]
As long as the supply voltage remains in the operating range, the RTC never stops,
regardless of the device status (Run mode, low-power mode or under reset).
```

erwango, 2024-08-26:
Just read the "with battery" in the title.

FRASTM, 2024-08-29:
1. RTC_SEL() is the RTC clock source selection as described in the RCC backup domain control register (RCC_BDCR bit[9:8]) 
If you choose to clock the RTC by an external LSE clock, then Device Tree DTS must be `RTC_SEl(1) ` and the LSE must be 32768Hz

2. to keep the calendar during power-off, the hardware must supply the RTC by Vbat with "VBAT pin can be connected to an optional standby voltage supplied by a battery or by another source."

paradajz, 2024-09-02:
Okay, I've set RTC_SEC to 1, but the issue remains. I'm using Epson Q13FC13500003 crystal with 1p caps.

[image]

I've also tried to set different levels of `driving-capability` to no avail.

erwango, 2024-11-06:
To be tested first with nucleo_wb55rg which provides native HW battery support

closed 2025-01-21 by github-actions[bot]