zephyrproject-rtos/zephyr #77523: Bluetooth: extended advertising: excessive aux offsets

JordanYates, 2024-08-25:
**Describe the bug**

Running the standard `samples/bluetooth/extended_adv/advertiser` sample on a `nrf52840dk/nrf52840` results in some pretty extreme timing differences in the scheduling of extended advertising packets. The Aux Offset with the Zephyr controller tends to be anywhere from 3ms all the way up to 108ms. e.g.
[image]

By contrast, running the same sample with the Nordic Softdevice on v2.7.0 is consistently 450us on the last primary advertising channel. The Zephyr controller values seem way too large to be correct.

Below is a wireshark .pcap trace of the Zephyr controller delays (rename the .txt to .pcap):
[ext_adv_delay.txt](https://github.com/user-attachments/files/16739881/ext_adv_delay.txt)

**To Reproduce**

Steps to reproduce the behavior:
1. `west build -b nrf52840dk/nrf52840 zephyr/samples/bluetooth/extended_adv/advertiser/ -p`
2. Observe AUX offsets with Wireshark

**Expected behavior**

AUX offsets should not be this large.

**Impact**

Scanning for extended advertising packets with such large offsets seems to be very problematic.
I have only a cursory understanding of scheduling in the controller, so this may be entirely incorrect.
The scheduling of the receive here always returns `TICKER_STATUS_BUSY` when given such large aux offsets.
https://github.com/zephyrproject-rtos/zephyr/blob/f29377a12cb5dd2ee845f21b8319c0c4ee2cca40/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c#L784

**Environment (please complete the following information):**

 - Zephyr v3.7.0

cvinayak, 2024-08-26:
> AUX offsets should not be this large.

BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 6, Part B, Section 2.3.4.5 AuxPtr field

> The Aux Offset shall be at least the length of
the packet plus T_MAFS

The upper limit to the aux offset value is restricted by only the field/units to represent it, and hence large aux offsets are conformant to the specifications.

> The scheduling of the receive here always returns TICKER_STATUS_BUSY when given such large aux offsets.

This is acceptable as per Zephyr Controller design, and if any failure to receive should have been accordingly handled (release buffers reserved stashed/reserved etc.) in https://github.com/zephyrproject-rtos/zephyr/blob/f29377a12cb5dd2ee845f21b8319c0c4ee2cca40/subsys/bluetooth/controller/ll_sw/ull_scan_aux.c#L1405

What are the practical symptoms you are observing? Is the Observer having difficulty receiving advertising reports for transmissions with high aux offsets? Are there reduced advertising reports while there are advertisings with large aux offsets?

To be able to receive multiple interleaved extended advertisings' advertising reports, you can try increasing `CONFIG_BT_CTLR_SCAN_AUX_SET` value.

JordanYates, 2024-08-26:
The practical symptoms were that I was unable to receive **any** extended advertising reports on the Observer, despite seeing most of them through Wireshark. Returning to the Softdevice controller immediately solved the issue without any changes to my Bluetooth logic (with the downside that i am once again unable to connect through iOS devices).

I'm not arguing that the large values are not conforming to the spec, more that when I was looking at the wildly varying and constantly increasing (to a point) offsets, it looked to me like some  sort of pathological case of resource contention and backoffs. I couldn't track down where in the controller the desired offset was coming from to confirm whether this was the case though.

I am used to most things in Bluetooth land being on the microseconds scale, so when a standard sample starts generating offsets of 110ms (when the desired advertising interval is smaller than that), I smell a problem.

cvinayak, 2024-08-26:
[image]

It is not necessary to have one-to-one aux offset between primary and auxiliary channel PDUs, and the aux offsets can be larger than the advertising intervals.

> The practical symptoms were that I was unable to receive any extended advertising reports on the Observer.

Is the Observer also Zephyr Controller based? or which phone is this?

JordanYates, 2024-08-26:
The observer and advertisers were both Zephyr devices (nRF52840) running v3.7.0.

Again, I understand that the values are compliant with the specs, but they were so far away from what I expected that it indicated to me a problem. Is there somewhere in the controller logic where you would expect offsets of >100ms to be generated while running `samples/bluetooth/extended_adv/advertiser`?

cvinayak, 2024-08-26:
The offset > 100 ms can be (recently) generated from this PR change: https://github.com/zephyrproject-rtos/zephyr/pull/76132/files#diff-2cdf3e8b2d143b776f14fe1e226fa7c85ed0c09c9f0d0b930626a074df895fa9R3262

JordanYates, 2024-08-26:
> The offset > 100 ms can be (recently) generated from this PR change: https://github.com/zephyrproject-rtos/zephyr/pull/76132/files#diff-2cdf3e8b2d143b776f14fe1e226fa7c85ed0c09c9f0d0b930626a074df895fa9R3262

I'm relatively sure that I was testing without that PR, I can double check tomorrow

JordanYates, 2024-08-27:
> The offset > 100 ms can be (recently) generated from this PR change: https://github.com/zephyrproject-rtos/zephyr/pull/76132/files#diff-2cdf3e8b2d143b776f14fe1e226fa7c85ed0c09c9f0d0b930626a074df895fa9R3262

Can confirm that the large offsets are present on the `v3.7.0` tag, which does not contain the linked PR.

cvinayak, 2024-08-27:
Aux offsets of upto 110 ms for advertising intervals of 100 ms are possible in v3.7.0 too, the PR just makes it more likely.

I am feeling you have too low Rx buffers and hence, by the time (~110 ms) to receive the auxiliary PDU then legacy advertising PDUs received utpto then consume the Rx buffers. You could increase the Rx buffer counts in the Controller and check if it improves the reception of extended advertising reports with large aux offsets. If this needs improvement, then I will need to think around pre-allocating/reserving Rx buffers in anticipation of a auxiliary PDU reception.

`CONFIG_BT_CTLR_RX_BUFFERS`

cvinayak, 2024-09-03:
Did you get a chance to try with increased Rx buffer count?

closed 2024-11-17 by github-actions[bot]