zephyrproject-rtos/zephyr #77538: C++ constructor can't run before driver initialization

yperess, 2024-08-26:
In a downstream project we're using C++ to write drivers, but the init function can only run in POST_KERNEL which is before the constructors are run. This means that all our data fields haven't been initialized yet and we have to schedule our own initialization via a k_work. We need a way to either set a Kconfig that will run the C++ constructors earlier OR bring back the APPLICATION init phase so we can initialize the drivers later.

henrikbrixandersen, 2024-08-26:
Technically, as per the discussions in recent TSC meetings, Zephyr supports writing _applications_ in languages other than C, not kernel code such as drivers.

stephanosio, 2024-08-26:
From "C++ Language Support" page:

> [Do not use C++ for kernel, driver, or system initialization code.](https://docs.zephyrproject.org/latest/develop/languages/cpp/index.html)

C++ support, in its current form, aims to provide the underlying language and runtime support for porting existing C++ applications to Zephyr; not to provide writing core Zephyr components in C++.

For this reason, I have marked this as an enhancement.

yperess, 2024-08-26:
> From "C++ Language Support" page:
> 
> > [Do not use C++ for kernel, driver, or system initialization code.](https://docs.zephyrproject.org/latest/develop/languages/cpp/index.html)
> 
> C++ support, in its current form, aims to provide the underlying language and runtime support for porting existing C++ applications to Zephyr; not to provide the facilities necessary for writing core Zephyr components in C++.
> 
> For this reason, I have marked this as an enhancement.

Thanks for the clarification. I always assumed that section to be meant for "don't do this upstream". I didn't realize it applied downstream too. I'm happy with this being an enhancement. We have a few internal projects that want to use Zephyr's device model but require C++ support and having the constructors run before the device initialization. Right now we return 0 and punt to the work queue, but you can imagine that's not ideal.