zephyrproject-rtos/zephyr #77556: I want to enable secure boot in ESP32-S3, but I checked in menuconfig, and it doesn't support it. How can I enable it?

SanjayEIC, 2024-08-26:
**Is your enhancement proposal related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

rftafas, 2024-08-26:
Hi.

Without a bootloader, this won't be available. You need MCUboot. But just using "regular" MCUboot won't work either. To get security features, you need to use Espressif Port of MCUboot.

- Instructions here: https://docs.mcuboot.com/readme-espressif.html
- Yes, you need ESP-IDF installed for now.
- Yes, you generate 2 images independently.
- No, this is not a Fork. It is just a different port within MCUboot main repo.
- We are working to improve Zephyr and MCUboot-Espressif Port relation, but not there yet.
- This is not a bug neither a missing feature. We (Espressif team) do not have plans to port security features to Zephyr Port of MCUboot.

This can be closed or, better yet, moved to discussion.

SanjayEIC, 2024-08-26:
Thank you so much for update.