zephyrproject-rtos/zephyr #77612: The NUCLEO-H745ZI-Q, NUCLEO-H755ZI-Q and NUCLEO-H7A3ZI-Q share the same board reference MB1363 but implement different common parts

bene42, 2024-08-27:
The NUCLEO-H745ZI-Q, NUCLEO-H755ZI-Q and NUCLEO-H7A3ZI-Q share the same board reference MB1363 i.e. use the same PCB with different SoCs. Furthermore the SoC share a wide varity of common IPs.

The difference between the boards are in Crypto support and Ethernet.

 - NUCLEO-H7A3ZI-Q: No Crypto and no Ethernet
 - NUCLEO-H745ZI-Q: No Crypto, but Ethernet
 - NUCLEO-H755ZI-Q: Crypto and Ethernet

The board ports of all 3 boards differ in common features: LEDs, different tests, different default configuration.
Fixes and enhancements to one of these boards are not ported to the other boards and therefor the boards have diverged.

**Expected behavior**
All features, tests, settings except Crypto and Ethernet behave the same on all these boards. 

**Impact**
Applications compiled for these boards behave different in common parts.

**Notes**
As far as I see, the new Zephyr current hardware model takes into account:
 - Support for multi-core, multi-arch AMP (Asymmetrical Multi Processing) SoCs
 - Support for multi-SoC boards
 - Support for reusing the SoC and board Kconfig trees outside of the Zephyr build system
 - Support for advanced use cases with Sysbuild (System build)

but not different SoC on a common PCB, like above.

Regards
    Benedikt Spranger

nordicjm, 2024-08-27:
They could be put into one folder and have the same base files then just include the common dts file, .conf files would need to be separate.

erwango, 2024-09-17:
> They could be put into one folder and have the same base files then just include the common dts file, .conf files would need to be separate.

IMO, this proposal is only not viable as doesn't take into account whole board aspects like documentation and .yaml files.
Removing the `STM32` label as this is a generic problem to the hardware model, also switching from `Bug` to `Enhancement`.

nordicjm, 2024-09-17:
> IMO, this proposal is only not viable as doesn't take into account whole board aspects like documentation and .yaml files. Removing the `STM32` label as this is a generic problem to the hardware [...]

yaml files would be the same, documentation would just be multiple files in the doc folder, there are boards that do this already e.g. some adafruit boards

nordicjm, 2024-09-17:
> > there are boards that do this already e.g. some adafruit boards
> 
> Would you mind pointing me to an example. I haven't found any accross adafruit. Other vendor maybe ?
> 
> > yaml files would be the same
> 
> I meant `board.yml` file. To me, only one board could be specified, but I could be wrong.

The adafruit one was reworked last week, another example here of multiple boards: https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/st/stm32l1_disco

erwango, 2024-09-20:
So indeed, there should be some room to factorize the PCB part of nucleo boards as usually, a same pcb is used between several nucleo boards (see 3 boards mentioned above).

I see 3 challenges or things to take care when doing this:
- Preserve boards visibility in https://docs.zephyrproject.org/latest/boards/index.html (not the case today with stm32l1_disco)
- Avoid a flat directory with tens of files (MB1136 reference board is used across 18 nucleo references, not counting that some can have variants, revisions, ...) > this would be quickly a mess
- We should maintain script'ability. Today it's easy to do massive changes across boards, this should be preserved
- People should understand easily how things are organized. For instance replacing nucleo_foo folders by mbxxx folders is likely to cause trouble to a majority of users.

Note also that there are actually some differences across nucleo references using the same pcb ref. 
For instance on MB1136:
- BOOT0 Pin: Present on all boards, but some boards have additional configurations (e.g., NUCLEO-F091RC can configure BOOT0 as PF11).
- Analog Pins (A4 and A5): Vary between PC1/PB9 and PC0/PB8 across different boards.
- ST morpho Connector: Generally consistent but with specific variations in pin configurations and additional options for certain boards.

nordicjm, 2024-09-24:
> Preserve boards visibility in https://docs.zephyrproject.org/latest/boards/index.html (not the case today with stm32l1_disco)

This is just because of that board only having one doc file, there can be multiple doc files in a single directory and it will have multiple entries, under arm check `Arm FVP BaseR AEMv8-R AArch32` and `Arm FVP BaseR AEMv8-R` - these are both in the same directory but have distinct entries there.