zephyrproject-rtos/zephyr #77613: Zbus heap increase

chirambaht, 2024-08-27:
**Is your feature request related to a problem? Please describe.**
When running a code with a few message subscribers and listeners, an error started to appear described as follows:
```txt
ASSERTION FAIL [buf != ((void *)0)] @ WEST_TOPDIR/deps/zephyr/subsys/zbus/zbus.c:133
    net_buf zbus_msg_subscribers_pool is unavailable or heap is full
@ WEST_TOPDIR/deps/zephyr/lib/os/assert.c:43
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x80bad00 (test_thread)
[00:00:00.000,000] <err> os: Halting system
```
This is run with no **CONFIG_HEAP_MEM_POOL_SIZE** option defined to use the auto heap calculation introduced recently in Zephyr.

**Describe the solution you'd like**
Adding a Kconfig allowing Zbus users to add heap to their projects as is needed. This is done by setting adding the following lines to the `zephyr/subsys/zbus/Kconfig` file:
```kconfig
config HEAP_MEM_POOL_ADD_SIZE_ZBUS
    int "The size of heap to add to the memory pool on behalf of ZBUS."
    default 0
    depends on ZBUS_MSG_SUBSCRIBER_BUF_ALLOC_DYNAMIC
```

rodrigopex, 2024-08-27:
@chirambaht, thanks for pointing out this improvement on zbus. I have added a PR for adding that. I added a default value of 1024 for the heap size for zbus. 

I hope, soon, you will see that available in the main.

rodrigopex, 2024-11-02:
Solve by #77631

PR #77631 (2024-10-10): zbus: add HEAP_MEM_POOL_ADD_SIZE_ZBUS
ZBus currently does not have a `HEAP_MEM_POOL_ADD_SIZE_ZBUS` when using message subscribers or runtime observers, forcing the developer to set that manually. Adding the configuration option to zbus would improve its usability and make zbus easier to use. With this feature, zbus aligns with the other Zephyr subsystems' heap memory allocation approach.

Fixes #77613
closed 2024-11-02 by rodrigopex