zephyrproject-rtos/zephyr #77615: SPI: NRF52DK/NRF52832 fails spi_controller_peripheral tests

matt-wood-ct, 2024-08-27:
**Describe the bug**
I'm investigating an issue using SPI between an IMXRT and NRF52 MCUs and as part of that I went back to basics and was validating the SPI peripherals behaviours on both ICs, to do this I ran the `tests/drivers/spi/spi_controller_peripheral` test on both MCUs and I got this strange result on the NRF

```
------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [spi_controller_peripheral]: pass = 5, fail = 5, skip = 4, total = 14 duration = 0.198 seconds
 - PASS - [spi_controller_peripheral.test_basic] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_basic_async] duration = 0.026 seconds
 - PASS - [spi_controller_peripheral.test_basic_zero_len] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_basic_zero_len_async] duration = 0.026 seconds
 - PASS - [spi_controller_peripheral.test_only_rx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_only_rx_async] duration = 0.023 seconds
 - SKIP - [spi_controller_peripheral.test_only_rx_in_chunks] duration = 0.004 seconds
 - SKIP - [spi_controller_peripheral.test_only_rx_in_chunks_async] duration = 0.004 seconds
 - PASS - [spi_controller_peripheral.test_only_tx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_only_tx_async] duration = 0.026 seconds
 - SKIP - [spi_controller_peripheral.test_only_tx_in_chunks] duration = 0.004 seconds
 - SKIP - [spi_controller_peripheral.test_only_tx_in_chunks_async] duration = 0.004 seconds
 - PASS - [spi_controller_peripheral.test_short_rx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_short_rx_async] duration = 0.026 seconds

------ TESTSUITE SUMMARY END ------
```

It seems like all the Async tests fail on this platform, is this right?

**To Reproduce**
Setup a dev kit with some loop back jumpers from

- P0.31 -> P0.11
- P0.30 -> P0.12
- P0.29 -> P0.13
- P0.28 -> P0.14

With the attached overlay file 
[nrf52dk_nrf52832.overlay.txt](https://github.com/user-attachments/files/16759951/nrf52dk_nrf52832.overlay.txt)

Build and run

**Expected behavior**
Full set of passes

**Impact**
Unclear, may be unrelated or may be the cause of my issues

**Logs and console output**
Full output:
<details>
  <summary>Full log</summary>

  ```
*** Booting Zephyr OS build v3.7.0-1712-geaf92a1494cc ***
Running TESTSUITE spi_controller_peripheral
===================================================================
START - test_basic
 PASS - test_basic in 0.011 seconds
===================================================================
START - test_basic_async

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_controller_peripheral/src/main.c:253: run_test: (periph_rv not equal to srx_len)
Got: 0 but expected:16
 FAIL - test_basic_async in 0.026 seconds
===================================================================
START - test_basic_zero_len
 PASS - test_basic_zero_len in 0.011 seconds
===================================================================
START - test_basic_zero_len_async

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_controller_peripheral/src/main.c:253: run_test: (periph_rv not equal to srx_len)
Got: 0 but expected:8
 FAIL - test_basic_zero_len_async in 0.026 seconds
===================================================================
START - test_only_rx
 PASS - test_only_rx in 0.011 seconds
===================================================================
START - test_only_rx_async

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_controller_peripheral/src/main.c:259: run_test: rv not equal to 0

 FAIL - test_only_rx_async in 0.023 seconds
===================================================================
START - test_only_rx_in_chunks
E: Scattered buffers are not supported
 SKIP - test_only_rx_in_chunks in 0.004 seconds
===================================================================
START - test_only_rx_in_chunks_async
E: Scattered buffers are not supported
 SKIP - test_only_rx_in_chunks_async in 0.004 seconds
===================================================================
START - test_only_tx
 PASS - test_only_tx in 0.011 seconds
===================================================================
START - test_only_tx_async

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_controller_peripheral/src/main.c:253: run_test: (periph_rv not equal to srx_len)
Got: 0 but expected:16
 FAIL - test_only_tx_async in 0.026 seconds
===================================================================
START - test_only_tx_in_chunks
E: Scattered buffers are not supported
 SKIP - test_only_tx_in_chunks in 0.004 seconds
===================================================================
START - test_only_tx_in_chunks_async
E: Scattered buffers are not supported
 SKIP - test_only_tx_in_chunks_async in 0.004 seconds
===================================================================
START - test_short_rx
 PASS - test_short_rx in 0.011 seconds
===================================================================
START - test_short_rx_async

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_controller_peripheral/src/main.c:253: run_test: (periph_rv not equal to srx_len)
Got: 0 but expected:12
 FAIL - test_short_rx_async in 0.026 seconds
===================================================================
TESTSUITE spi_controller_peripheral failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [spi_controller_peripheral]: pass = 5, fail = 5, skip = 4, total = 14 duration = 0.198 seconds
 - PASS - [spi_controller_peripheral.test_basic] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_basic_async] duration = 0.026 seconds
 - PASS - [spi_controller_peripheral.test_basic_zero_len] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_basic_zero_len_async] duration = 0.026 seconds
 - PASS - [spi_controller_peripheral.test_only_rx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_only_rx_async] duration = 0.023 seconds
 - SKIP - [spi_controller_peripheral.test_only_rx_in_chunks] duration = 0.004 seconds
 - SKIP - [spi_controller_peripheral.test_only_rx_in_chunks_async] duration = 0.004 seconds
 - PASS - [spi_controller_peripheral.test_only_tx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_only_tx_async] duration = 0.026 seconds
 - SKIP - [spi_controller_peripheral.test_only_tx_in_chunks] duration = 0.004 seconds
 - SKIP - [spi_controller_peripheral.test_only_tx_in_chunks_async] duration = 0.004 seconds
 - PASS - [spi_controller_peripheral.test_short_rx] duration = 0.011 seconds
 - FAIL - [spi_controller_peripheral.test_short_rx_async] duration = 0.026 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED

  ```
</details>

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain Zephyr SDK
 - eaf92a1494cc

nordicjm, 2024-08-27:
@bjarki-andreasen aren't you doing something around this?

bjarki-andreasen, 2024-08-27:
@nordicjm no :)

closed 2026-03-14 by github-actions[bot]