zephyrproject-rtos/zephyr #77626: Teensy41 does not run with SPI enabled

matt-wood-ct, 2024-08-27:
**Describe the bug**

I enabled `CONFIG_SPI` in prj.conf in the hello world sample and it doesn't output anything. Tested with the stock empty prj.conf and it works perfectly, I suspect something in the SPI init is getting stuck in a polling loop or faulting.

**To Reproduce**
Steps to reproduce the behaviour:
1. Open hello world sample
2. Add `CONFIG_SPI=y` to the prj.conf
3. west build
4. flash & reboot with teensy loader

**Expected behavior**
Normal hello world output

**Impact**
SPI unusable on DK

**Logs and console output**

No output when SPI enabled, checked with a logic analyser on pins 0 and 1 to eliminate baud / pin swap issues. D1 comes high but no serial output.

**Environment (please complete the following information):**

 - Windows
 - Zephyr SDK
 - eaf92a1494cc0e245f14f22ad1d1d42f363a07b3

matt-wood-ct, 2024-08-27:
Tested the same setup on the `mimxrt1040_evk` board and it runs without issue, so its not a generic IMXRT issue on this version of zephyr, it is specific to the teensy

nordicjm, 2024-08-27:
@bdkrae are you able to investigate?

bdkrae, 2024-08-27:
From looking at the code I see that there is no SPI peripheral activated by default in the devicetree of `teensy4x` (as opposed to `mimxrt1040_evk`). This is intended as the teensy board doesn´t have too many pins and they can be mapped to different peripherals project dependent.

So please add a devicetree-overlay file to your project and insert e.g.
```
&lpspi3 {
  status = "okay";
};
```
to activate the SPI3. This should solve the issue.

matt-wood-ct, 2024-08-28:
> From looking at the code I see that there is no SPI peripheral activated by default in the devicetree of `teensy4x` (as opposed to `mimxrt1040_evk`). This is intended as the teensy board doesn´t [...]

Logic analyser capture:

[image]

Also I checked the DT for the teensy41, `lpspi3` is enabled in the 41 version

`lpspi3` is enabled on the teensy41:
https://github.com/zephyrproject-rtos/zephyr/blob/ad70ff7b9d5dbd342f53e4abd7f5cb13bf8c111d/boards/pjrc/teensy4/teensy41.dts#L37-L39

mmahadevan108, 2024-08-28:
You would also need `pinmux` setup in the `pinctrl` file.

matt-wood-ct, 2024-08-29:
@mmahadevan108 I don't think you do, the device tree stacking for this part is 

1. `teensy41.dts`
2. `teensy40.dts`
3. NXP base SOC files

1 has
```
 	status = "okay"; 
```

2 has 
```
	pinctrl-0 = <&pinmux_lpspi3>;
	pinctrl-names = "default";
```
and
```
	/* LPSPI3 MISO, MOSI, SCK, CS on Teensy-Pins 39/26/27/38 */
	pinmux_lpspi3: pinmux_lpspi3 {
		group0 {
			pinmux = <&iomuxc_gpio_ad_b1_12_lpspi3_pcs0>,
				<&iomuxc_gpio_ad_b0_00_lpspi3_sck>,
				<&iomuxc_gpio_ad_b0_02_lpspi3_sdi>,
				<&iomuxc_gpio_ad_b0_01_lpspi3_sdo>;
			drive-strength = "r0-6";
			slew-rate = "slow";
			nxp,speed = "100-mhz";
		};
	};
```

The resultant DT used for the build is 
```
&lpspi3 {
 	status = "okay"; 
	pinctrl-0 = <&pinmux_lpspi3>;
	pinctrl-names = "default";
        /* NXP compat, ISR and CLK */
};
```

So as far as I can tell the SPI is configured correctly, the fact I'm not getting anything out leads me to believe one of the following things is happening:  
1. The driver is stuck in a hard loop due to pin conditions (I've seen this on NXPs UART driver, when using flow control several times)
2. The firmware is in a hardfault -> reboot loop

bdkrae, 2024-09-01:
I was able to reproduce.

It´s a bug in `teensy4-pinctrl.dtsi`. LPSPI3 was mapped to the same pins as LPUART6. Corrected lines 132-134 are:
```
/* LPSPI3 MISO, MOSI, SCK, CS on Teensy-Pins 39/26/27/38 */
	pinmux_lpspi3: pinmux_lpspi3 {
		group0 {
			pinmux = <&iomuxc_gpio_ad_b1_12_lpspi3_pcs0>,
				<&iomuxc_gpio_ad_b1_15_lpspi3_sck>, // this line was interfering with lpuart6
				<&iomuxc_gpio_ad_b1_13_lpspi3_sdi>, // this line was interfering with lpuart6
				<&iomuxc_gpio_ad_b1_14_lpspi3_sdo>;  // this line was interfering with lpuart6
			drive-strength = "r0-6";
			slew-rate = "slow";
			nxp,speed = "100-mhz";
		};
	};
```

I´ll prepare a PR next days

matt-wood-ct, 2024-09-09:
@nordicjm The pull request @bdkrae wrote appears to fix the issue, can we get it pulled?

PR #77889 (2024-09-10): boards: teensy4: Fix pinmux settings for LPSPI3
Was interfering with LPUART6, which is the chosen terminal output.

Resolves #77626
closed 2024-09-10 by nashif