zephyrproject-rtos/zephyr #77639: cpuflpr target broken on custom nRF54L15 boards using sysbuild

gmarull, 2024-08-27:
**Describe the bug**

When creating a custom board based on nRF54L15, the build is broken for `cpuflpr` target if using sysbuild. This is likely because the flpr snippets are not populated for such board, but can hardly be achieved downstream (where custom boards usually reside).

**To Reproduce**

- Create a custom board for nRF54L15, e.g. with: `west ncs-genboard -o boards -e "myvnd" -b myboard -d "My Board" -s nrf54l15 -v qfaa -n "2.7.0"` from https://github.com/nrfconnect/sdk-nrf/pull/16861 
- Run `west build -b myboard/nrf54l15/cpuflpr samples/hello_world -p --sysbuild`
- Observe the failures

```
   **********************************
   * Running CMake for vpr_launcher *
   **********************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/gmarull/ws/nordic/zephyrproject/zephyr/samples/basic/minimal
-- CMake version: 3.28.2
-- Found Python3: /home/gmarull/ws/nordic/zephyrproject/.venv/bin/python (found suitable version "3.12.4", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/gmarull/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/gmarull/ws/nordic/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: myboard, qualifiers: nrf54l15/cpuapp
-- Snippet(s): nordic-flpr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/gmarull/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/gmarull/zephyr-sdk-0.16.8)
-- Found Dtc: /home/gmarull/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/gmarull/ws/nordic/zephyrproject/zephyr/boards/myvnd/myboard/myboard_nrf54l15_cpuapp.dts
-- Found devicetree overlay: /home/gmarull/ws/nordic/zephyrproject/zephyr/snippets/nordic-flpr/nordic-flpr.overlay
-- In: /home/gmarull/ws/nordic/zephyrproject/zephyr/build/vpr_launcher/zephyr, command: /home/gmarull/ws/nordic/zephyrproject/.venv/bin/python;/home/gmarull/ws/nordic/zephyrproject/zephyr/scripts/dts/gen_defines.py;--dts;/home/gmarull/ws/nordic/zephyrproject/zephyr/build/vpr_launcher/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/home/gmarull/ws/nordic/zephyrproject/zephyr/dts/bindings;--header-out;/home/gmarull/ws/nordic/zephyrproject/zephyr/build/vpr_launcher/zephyr/include/generated/zephyr/devicetree_generated.h.new;--dts-out;/home/gmarull/ws/nordic/zephyrproject/zephyr/build/vpr_launcher/zephyr/zephyr.dts.new;--edt-pickle-out;/home/gmarull/ws/nordic/zephyrproject/zephyr/build/vpr_launcher/zephyr/edt.pickle;--vendor-prefixes;/home/gmarull/ws/nordic/zephyrproject/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /home/gmarull/ws/nordic/zephyrproject/zephyr/cmake/modules/dts.cmake:311 (message):
  gen_defines.py failed with result code: 1 - stderr contents:

  devicetree error: 'execution-memory' is marked as required in 'properties:'
  in
  /home/gmarull/ws/nordic/zephyrproject/zephyr/dts/bindings/riscv/nordic,nrf-vpr-coprocessor.yaml,
  but does not appear in <Node /soc/peripheral@50000000/vpr@4c000 in
  '/home/gmarull/ws/nordic/zephyrproject/zephyr/misc/empty_file.c'>

Call Stack (most recent call first):
  /home/gmarull/ws/nordic/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/gmarull/ws/nordic/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/gmarull/ws/nordic/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:486 (message):
  CMake configure failed for Zephyr project: vpr_launcher

  Location:
  /home/gmarull/ws/nordic/zephyrproject/zephyr/samples/basic/minimal
Call Stack (most recent call first):
  cmake/modules/sysbuild_images.cmake:20 (ExternalZephyrProject_Cmake)
  cmake/modules/sysbuild_default.cmake:20 (include)
  /home/gmarull/ws/nordic/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/gmarull/ws/nordic/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  /home/gmarull/ws/nordic/zephyrproject/zephyr/share/sysbuild-package/cmake/SysbuildConfig.cmake:8 (include)
  template/CMakeLists.txt:10 (find_package)
```

**Expected behavior**

Out of tree boards should work out of the box, or have a mechanism to make them work.

**Impact**

Blocker for out of tree nRF54L15 based boards.

**Logs and console output**

See above.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used 7b4e7ff9eda0227762666a39636b5f7729fdfd51

**Additional context**

N/A

gmarull, 2024-08-27:
Note that wildcards in snippet may be a solution if the overlay can be applied universally to all nRF54L15 boards.

nordic-piks, 2024-09-12:
@barsok Please have a look, this is probably a new feature request.

57300, 2024-10-02:
I believe one of the goals for snippets was to support extending them out of tree, which would allow this to be solved generically. From #51834:

> I additionally want to be able to extend existing "snippets" defined elsewhere. For example, if the zephyr repository defines a standard `cdc-acm-console` snippet, I want to be able to extend it with support for my board by adding files to a module I control, without any changes to the zephyr repository itself.

PR #80522 (2024-11-16): snippets: nordic-*: support custom boards
Apply overlays for custom boards, based on already supported SoCs.

Fixes #77639.
commit 26973bd0 (2024-11-16): snippets: nordic-*: support custom boards

Apply overlays for custom boards, based on already supported SoCs.

Fixes #77639.
closed 2024-11-16 by nashif