zephyrproject-rtos/zephyr #77681: hostap os_time_t definition as long long

fengming-ye, 2024-08-28:
Hostap origin code is defining os_time_t as long. Why does zephyr have to define os_time_t as long long?
And this change needs many following changes of PRINTF format of os_time_t from %ld to %lld.
That will cause many warnings when we enable some macros which are disabled before.
And it is hard for us to always change PRINTF format.
So can we stick to os_time_t as long and revert the PRINTF format changes?

fengming-ye, 2024-08-28:
@krish2718 pls find os_time_t issue here

krish2718, 2024-08-28:
Thanks. The original commit (downstream) is https://github.com/nrfconnect/sdk-hostap/pull/21/commits/aad5576ca51989b002d919b1be5b3168ca46c586 and as you can see it was reported by coverity, see below information (all 3 CIDs have same information).

```
176118 [Use of 32-bit time_t](https://coverity:8443/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY)
The time value stored in this integer will represent a different, but possibly valid, time.

In os_get_time: A 64-bit time_t value is stored in a smaller width integer. ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))
```

fengming-ye, 2024-08-29:
Hi @krish2718 

Thanks for the info.
I got the point that os_get_time is converting uint64_t to uint32_t in 32-bit machine.
And also because this 32-bit machine is defining time_t as uint64_t.
And time restored in uint32_t can persist until 2038.
So if hostap is hard to change all the PRINTF %ld stuff, I'd propose an option like this:

int os_get_time(struct os_time *t)
{
	int res;
	struct timeval tv;

	res = gettimeofday(&tv, NULL);
	t->sec = (long)tv.tv_sec;
	t->usec = (long)tv.tv_usec;
	return res;
}

krish2718, 2024-08-29:
Sure, either approach is fine, do you want to take a stab and send either one to the hostap ML?

fengming-ye, 2024-09-02:
I can go with changing os_get_time and it is in driver_zephyr.
Shall I create a PR and you can check if coverity is fixed?

fengming-ye, 2024-09-02:
> Sure, either approach is fine, do you want to take a stab and send either one to the hostap ML?

I can go with changing os_get_time and it is in driver_zephyr.
Shall I create a PR and you can check if coverity is fixed?

krish2718, 2024-09-02:
> > Sure, either approach is fine, do you want to take a stab and send either one to the hostap ML?
> 
> I can go with changing os_get_time and it is in os_zephyr. Shall I create a PR and you can check if coverity is fixed?

Sure, but I would suggest sending this to the hostap ML (Mailing list) first to see which solution is better.

fengming-ye, 2024-09-02:
> Sure, but I would suggest sending this to the hostap ML (Mailing list) first to see which solution is better.

Fix:
https://github.com/zephyrproject-rtos/hostap/pull/34/commits
West:
https://github.com/zephyrproject-rtos/zephyr/pull/77864

This PR only reverts what Zephyr has changed hostap core code, and change os_zephyr.c, which is Zephyr special.
So it does not need to go to hostap ML.

krish2718, 2024-09-02:
> This PR only reverts what Zephyr has changed hostap core code, and change os_zephyr.c, which is Zephyr special.
So it does not need to go to hostap ML.

No, it changes `os.h`, so, need to submit to hostap ML.

fengming-ye, 2024-09-02:
> > This PR only reverts what Zephyr has changed hostap core code, and change os_zephyr.c, which is Zephyr special.
> > So it does not need to go to hostap ML.
> 
> No, it changes `os.h`, so, need to submit to hostap ML.

hostap os.h is using
typedef long os_time_t;

Zephyr changes it to
typedef long long os_time_t;

So this is reverting Zephyr special change.

krish2718, 2024-09-02:
Ah ok, understood, but I would still recommend to submit to the ML because this is warning is independent of the OS.

fengming-ye, 2024-09-02:
> Ah ok, understood, but I would still recommend to submit to the ML because this is warning is independent of the OS.

I'm still confused why this has to go to the ML. Because part 1 is changing code back to original hostap core code. And part 2 is changing os_zephyr.c which is not on hostap original code. And I'm not going to change other OS because 1. time_t is highly coupled archs and libs implementation. 2. I do not have environment to verify them.

krish2718, 2024-09-02:
>  And I'm not going to change other OS because 1. time_t is highly coupled archs and libs implementation. 2. I do not have environment to verify them.

I was suggesting upstreaming for other OSes, as this is now a `noup` change it's completely up to you to volunteer for other OSes.

closed 2024-10-17 by fengming-ye