zephyrproject-rtos/zephyr #77698: Configuring gpio pin multiple times as disconnected can stop other pins to react on stm32

jbr-ia, 2024-08-28:
**Describe the bug**
When configuring a gpio-pin by using ```gpio_pin_configure_dt(..)``` multiple times as GPIO_DISCONNECTED, the other pins on the same port can also stop working. This seems to happen with an stm32 when PM_DEVICE_RUNTIME is enabled (I have seen it with nucleo_wb55rg and nucleo_u575zi_q).

**To Reproduce**
Use the basic blinky sample (for example with nucleo_wb55rg board). When reproducing, it can be seen that disconnecting LED1 too often, affects the blinking of LED0.
1) Add:
```
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
```
to the prj.conf

2) Configure another pin than the default led, but on the same port as the LED (when using the nucleo_wb55rg, pin PB5 is used for the default (blue)LED, the green led (led1) is on PB0, so add the green led). To do that, add to the main.c above the main-function:
```
#define LED1_NODE DT_ALIAS(led1)
static const struct gpio_dt_spec led1 = GPIO_DT_SPEC_GET(LED1_NODE, gpios);
```
In the main-function just before the while-loop, add:
```ret = gpio_pin_configure_dt(&led1, GPIO_DISCONNECTED); ```
3) Build the sample for the nucleo_wb55rg (``` west build -b nucleo_wb55rg ./samples/basic/blinky ```) and flash it to the nucleo
3) See that the blue LED(led0) does not blink. 
4) Now add ```ret = gpio_pin_configure_dt(&led1, GPIO_OUTPUT_ACTIVE); ```  just before the GPIO_DISCONNECTED-line. (so the configuration of led1 to disconnected is still the last action before the while-loop)
5) Compile, and flash to the board. See that the Blue LED(led0)-blinking now does work.
6) Change the code, and see that if you first configure it as input or output multiple times (for example 5 times), that you can configure it as disconnected also up to 5 times with the blinking still working. If you configure it as disconnected more times, the blinking stops working.

The changes to reproduce are also here: https://github.com/jbr-ia/zephyr/commit/d28944e7c22902c6638199b739955be4ff1d4ff8

**Expected behavior**
Putting a pin as GPIO_DISCONNECTED should not affect other pins on the same port

**Impact**
When an pin is configured as disconnected multiple times, some other IO can unexpected stop working. 

**Environment:**
 - OS: Ubuntu
 - Zephyr sdk 0.16.8
 - zephyr SHA: 386263cc668b10a2546b93a2a3e7444d75b7b05d

**Additional context**
It seems to be caused by pm_device_runtime_put and pm_device_runtime_get which are called in gpio_stm32_config https://github.com/zephyrproject-rtos/zephyr/blob/ad70ff7b9d5dbd342f53e4abd7f5cb13bf8c111d/drivers/gpio/gpio_stm32.c#L424

Any time this function is called with the pin defined as disconnected, pm_device_runtime_put is called. Every time this function is called when defining a pin as input or output, pm_device_runtime_get is called. Because the pm_device_runtime_put/get functions need to be balanced, the calls of the config must also be called balanced. So also if a gpio-pin is switched between input and output multiple times, in order to disable the port, it must be called with disconnect the same number of times.

erwango, 2024-08-28:
Thanks for the detailed description and provided sample.

According to discussion on that matter ended with the following comment https://github.com/zephyrproject-rtos/zephyr/issues/32220#issuecomment-783411288, `GPIO_DISCONNECTED` is indeed the official way according to the GPIO API to track the state of a pin and hence pin configuration should be balanced with pin "release" using `GPIO_DISCONNECTED`.

(Note that the issue has been re-open for a side point on GPIO IRQ handling which is not linked with the issue you mention)

Hence, STM32 GPIO driver is behaving as expected. Please have a look to this discussion and then I'd suggest to close this issue related to GPIO STM32 driver behavior.
If you feel this is should be reviewed, I'd suggest opening a new issue targeting GPIO API directly.

jbr-ia, 2024-08-30:
Thanks for the response. I see in that issue that GPIO_DISCONNECTED can be used to indicate a pin is no longer used, and if all pins are disconnected, the clock can be disabled. 
But I do not see there that the configuration-calls are expected to be be balanced per pin. The problem I am having is that if for example the following actions are executed:
- gpio_pin_configure PA1 as GPIO_OUTPUT_ACTIVE
- gpio_pin_configure PA2 as GPIO_OUTPUT_ACTIVE
- gpio_pin_configure PA2 as GPIO_DISCONNECTED
- gpio_pin_configure PA2 as GPIO_DISCONNECTED

After these actions, the clock of gpioa is disabled, and therefore PA1 is not working anymore.
I had assumed PA1 should not be influenced by configuring PA2 multiple times as GPIO_DISCONNECTED.

If I keep track of which pins want to use the clock, and do not call pm_device_runtime_put/get multiple times for the same pin, than my problem seems to be gone. The way I kept track of which pins should still be used is here: https://github.com/zephyrproject-rtos/zephyr/commit/f84c96f4d8add84fc5acad419622431cb4c4b247

erwango, 2024-09-17:
@henrikbrixandersen I'd welcome your thoughts here. Should we consider that it's application responsibility to track state of each pin configuration or should this be done in drivers ?
I'm fine either ways, but fixing stm32 driver might result in wrong applications behavior which would then have impact on other gpio drivers in the end. So I prefer we're clear on responsibility split before starting.

henrikbrixandersen, 2024-09-17:
> **Expected behavior**
> Putting a pin as GPIO_DISCONNECTED should not affect other pins on the same port

I would expect that as well.

> @henrikbrixandersen I'd welcome your thoughts here. Should we consider that it's application responsibility to track state of each pin configuration or should this be done in drivers ? I'm fine [...]

I can't see how the application can track this always, as individual GPIOs may be used by other drivers and subsystems as well. I would say fixing this in the STM32 GPIO driver is the best approach.

erwango, 2024-09-18:
@jbr-ia If we're to fix this, your solution sounds good, would you want to push it yourself ?

jbr-ia, 2024-09-18:
@erwango  Okay, I will try to create a pull request. Probably tomorrow.

commit cf837dd3 (2024-09-23): drivers: gpio: stm32: keep track of clock enabling per pin

The driver enables the clock of a gpio-port if any of the
pins use the port. This is done by calling pm_device_runtime_get
when a pin is used and pm_device_runtime_put when the pin is not
used anymore.
These calls needs to be balanced. But if a single pin was configured
as GPIO_DISCONNECTED multiple times, every time pm_device_runtime_put
was called.
This caused the clock of the port to be stopped and therefore also
other pins on the same port stopped working.

This commit fixes this by keeping track of which pin on a port
has request [... cut]
closed 2024-09-23 by mmahadevan108