zephyrproject-rtos/zephyr #77751: NXP INPUTMUX not configured on MIMXRT595-EVK for Flexcomm12

DerekSnell, 2024-08-29:
The test [spi_loopback](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/spi/spi_loopback) is configured to use Flexcomm16 (`hs_spi1`) on the `mimxrt595_evk`.  Trying to enable this test on Flexcomm12 exposed a bug related to the INPUTMUX peripheral, and this test fails on that Flexcomm.  INPUTMUX is a peripheral that connects signals from different peripherals in the SOC.  The INPUTMUX registers `DMAC1_CHMUX_SEL15` and `DMAC1_CHMUX_SEL14` must be configured to connect the Flexcomm12 DMA requests to the DMA peripheral.  There is a bug with the current drivers since they do not configure the INPUTMUX when needed.

The ideal solution is to introduce a new INPUTMUX driver, and configure this peripheral with devicetree settings.  @VitekST submitted  https://github.com/zephyrproject-rtos/zephyr/pull/77674 to add this driver.

To replicate the issue, the spi_loopback test is run on the `mimxrt595_evk` board, and the overlay below configures to use `flexcomm12`:
```
	 &hs_lspi1{
		status = "disabled";

	 };

	 &pinctrl{
		pinmux_flexcomm12_spi: pinmux_flexcomm12_spi {
			group0 {
				pinmux = <IOPCTL_MUX(157, 6)>,
						<FC12_TXD_SCL_MISO_PIO4_30>,
						<FC12_RXD_SDA_MOSI_PIO4_31>,
						<FC12_CTS_SDA_SSELN0_PIO5_0>;
				input-enable;
				slew-rate = "normal";
				drive-strength = "normal";
			};
		};
	};

	arduino_spi: &flexcomm12 {
		compatible = "nxp,lpc-spi";
		/delete-property/ current-speed;
		pinctrl-0 = <&pinmux_flexcomm12_spi>;
		pinctrl-names = "default";
		#address-cells = <1>;
		#size-cells = <0>;
		dmas = <&dma0 34>, <&dma0 35>;
		dma-names = "rx", "tx";
		status = "okay";
	};

	&arduino_spi {
		slow@0 {
			compatible = "test-spi-loopback-slow";
			reg = <0>;
			spi-max-frequency = <500000>;
		};
		fast@0 {
			compatible = "test-spi-loopback-fast";
			reg = <0>;
			spi-max-frequency = <16000000>;
		};
	};
```
The pins below are used for this SPI from the Auduino header.   To run the test, short the MOSI and MISO signals at J27 p1 and p2.  With the FC12 changes above, this test fails.  It will pass if DMA is disabled, which skips the SPI ASYNC mode test.
```
	FC12_SCK_PIO4_29                  FXIO_D9   J28-p2
	FC12_TXD_SCL_MISO_PIO4_30         FXIO_10   J27-p2
	FC12_RXD_SDA_MOSI_PIO4_31         FXIO_11   J27-p1
	FC12_CTS_SDA_SSELN0_PIO5_0        FXIO_12   J28-p3
```

A workaround to resolve this issue is to configure the INPUTMUX in the application, before using the SPI driver.  This code snippet configures the INPUTMUX to connect the FC12 DMA request signals to the DMA:
```
#ifdef CONFIG_SPI_MCUX_FLEXCOMM_DMA
	#include "fsl_inputmux.h"

	/* By default, DMAC channel is not enabled for Flexcomm12 */
	INPUTMUX_Init(INPUTMUX);
	INPUTMUX_EnableSignal(INPUTMUX, kINPUTMUX_Flexcomm12RxToDmac0Ch34RequestEna, true);
	INPUTMUX_EnableSignal(INPUTMUX, kINPUTMUX_Flexcomm12TxToDmac0Ch35RequestEna, true);
	INPUTMUX_Deinit(INPUTMUX);
#endif
```

VitekST, 2024-12-18:
@DerekSnell After discussions with @decsny and @dleach02, I decided to abandon the `nxp,inputmux` driver, as the way it handles configuration in the DT, I have to admit, is far from ideal. The interrupt muxing config can be done statically (although I am against that in the long term), but I don't have an immediately available solution for configuring other signals, such as the DMA trigger signals, which this issue discusses. 

Declan suggested to use this binding for describing some of the INPUTMUX registers: https://www.kernel.org/doc/Documentation/devicetree/bindings/mux/reg-mux.txt The description, IMO, could be acted upon by the `nxp,lpc-dma` driver depending on the configuration supplied by the application with `dma_mcux_lpc_configure`.

decsny, 2024-12-18:
I am actually still a bit skeptical about the linux muxing system and DT description being used for zephyr and use cases like input mux, it would take some time to evaluate if there is value to doing that and how would be the best way if so. It could allow for there to at least be a generic zephyr API for muxing required mux resources for situations like this issue

In the meantime I think a reasonable way to address this issue is to introduce a board kconfig