zephyrproject-rtos/zephyr #77768: smf: state history tracking

brandon-exact, 2024-08-29:
**Is your feature request related to a problem? Please describe.**

Add a optional feature to SMF that tracks a history of states and its reason for transition. Add a shell command to dump the history to console in a nice way.

**Describe the solution you'd like**

One idea would be to add a ring buffer to `struct smf_ctx` that tracks the last X states and its reason for the new state. The ring buffer would be of a type that contains

a index that increases by 1 each time a transition happens
New State
User Data (transition reason, fixed size)

keith-zephyr, 2024-08-30:
There was a previous effort to add some tracing https://github.com/zephyrproject-rtos/zephyr/pull/52161 to the SMF.

However, that specific proposal was rejected as it did not use Zephyr's existing [tracing framework](https://docs.zephyrproject.org/latest/services/tracing/index.html).

@wysman - did you make any progress with converting your SMF specific tracing mechanism to Zephyr's tracing framework?

closed 2024-09-24 by brandon-exact

brandon-exact, 2025-01-29:
I never attempted implementing it