zephyrproject-rtos/zephyr #77773: spi_dw driver bug

ruiyinghou, 2024-08-30:
**Describe the bug**

There is a bug in the api dw.c file that is missing the configuration of the ser register (Select a register from the device) , as follows：
`write_ser(dev, 1 << config->slave);`
This line of code should be added to line 260 of the spi dw config() function as follows：
```c
if (!spi_dw_is_slave(spi)) {
		/* Baud rate and Slave select, for master only */
		write_baudr(dev, SPI_DW_CLK_DIVIDER(info->clock_frequency,
						    config->frequency));
		write_ser(dev, 1 << config->slave);
}
```
**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

henrikbrixandersen, 2024-09-02:
You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md - leaving the fields blank doesn't help anyone.

tbursztyka, 2024-09-02:
This is not a bug. See line 436. (and line 77 when everything is done).

closed 2024-09-02 by tbursztyka