zephyrproject-rtos/zephyr #77779: Significant increase in boot-up time of ESP32C3 with USB_Serial and SYSBUILD

rana-roy, 2024-08-30:
**Describe the bug**
In any Zephyr sample, flashing the default console and shell-uart configuration i.e. using uart0 of the esp32c3_devkitm board along with sysbuild, boots up the board immediately. However, when I set the configurations to usb_serial, the bootup time increases to approximately 20 seconds. 

- Target Platform: ESP32C3-DevkitM
- Target Board: esp32c3_devkitm
- Zephyr version: 3.7.0 (LTS)

**To Reproduce**
Change the configurations in the .dts file as:
        chosen {
		        zephyr,sram = &sram0;
		        zephyr,console = &usb_serial;//&uart0;
		        zephyr,shell-uart = &usb_serial;//&uart0;
		        zephyr,flash = &flash0;
		        zephyr,code-partition = &slot0_partition;
	        };

Build command: west build --sysbuild -b esp32c3_devkitm .

**Expected behavior**
The board should boot up immediately.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8

rana-roy, 2024-09-10:
Hi @sylvioalves,
I want to know if you get a chance to look into this bug.

sylvioalves, 2024-09-12:
@rana-roy sorry for the delay, will check this today/tomorrow.

sylvioalves, 2024-09-15:
@rana-roy fix is here: https://github.com/zephyrproject-rtos/hal_espressif/pull/340
Once merged I'll update Zephyr main accordingly. You can use in the meanwhile if it fits you.

sylvioalves, 2024-09-23:
Already fixed.

closed 2024-09-23 by sylvioalves