zephyrproject-rtos/zephyr #77794: Network connection unreliable on nucleo_f767zi

jkrautmacher, 2024-08-30:
**Describe the bug**

While working on a quite minimal Zephyr firmware for the `nucleo_f767zi` board I noticed that the network connection is unreliable. As shown below this is reproducible with the `net/telnet` sample of Zephyr `v3.7.0`.

The bug can be noticed by the following indicators:

- ICMP and any IP-based communication (IPv4 and IPv6) does not work
- only the orange, not the green LED on the RJ45 connector of the board is lighting up
- the `stm_eth` thread has significantly lower stack usage (see shell output of `kernel stacks`)

The bug appears after roughly 50 % of the boot processes. It was so far not observed that network communication breaks later if it was operational directly after booting the board.

**To Reproduce**

Steps to reproduce the behavior:
1. connect the `nucleo_f767zi` board with an Ethernet cable to a Linux PC
2. connect the PC also via USB to the onboard ST-Link
3. configure the static IPv4 address `192.0.2.2/24` on the PC
4. install the `st-info` tool (see e.g. [source repo](https://github.com/stlink-org/stlink))
5. build and flash `samples/net/telnet` from `zephyr v3.7.0` as described in the [Getting Started Guide](https://docs.zephyrproject.org/3.7.0/develop/getting_started/index.html)
6. execute the Python script below without any arguments (sorry, GitHub did not allow me to attach a Python file)

```
#!/usr/bin/python3

import argparse
import subprocess
import time

HELP = """This script can be used to debug an issue where the Zephyr-based
firmware of a target microcontroller is not able to communicate via the network
after roughly every second boot process. The Ethernet-capable microcontroller
has to be connected directly to the Linux-based host PC via an Ethernet cable
and a ST-Link. Reboots are triggered via the Open Source st-utils tool
(https://github.com/stlink-org/stlink). Connectivity is checked with ping."""

def main():
    args = parse_args()

    ok = 0
    not_ok = 0

    for i in range(args.iterations):
        print(f"Iteration #{i+1}")
        subprocess.run(["st-info", "--connect-under-reset"], check=True)
        time.sleep(args.delay)
        try:
            subprocess.run(
                ["ping", "-c", "1", args.address], check=True, stdout=subprocess.DEVNULL
            )
            print("ok")
            ok += 1
        except:
            print("not ok")
            not_ok += 1

    print(
        f"Success rate is {round(100 * ok / (ok + not_ok))} % ({ok} ok and {not_ok} failed)"
    )

def parse_args():
    parser = argparse.ArgumentParser(description=HELP)

    parser.add_argument(
        "-i",
        "--iterations",
        default=100,
        help="how often the test should be executed",
        type=int,
    )

    parser.add_argument(
        "-d",
        "--delay",
        default=10,
        help="how long to wait after reset for the ICMP request",
        type=int,
    )

    parser.add_argument(
        "-a",
        "--address",
        default="192.0.2.1",
        help="address of the microcontroller / target of ICMP request",
        type=str,
    )

    return parser.parse_args()

if __name__ == "__main__":
    main()
```

**Expected behavior**

It is expected that the used script reports 100 % success rate.

**Impact**

This bug is a showstopper. A firmware with such an unreliable network connection is useless.

**Logs and console output**

The script summarizes a test with 100 iterations on my setup with:

```
Success rate is 44 % (44 ok and 56 failed)
```

**Environment (please complete the following information):**

 - OS: Arch Linux
 - Toolchain: Zephyr SDK `0.16.5`
 - Zephyr version: `v3.7.0`

jkrautmacher, 2024-08-30:
Updated bug report after it turned out that Zephyr `v3.6.0` is affected too.

FRASTM, 2024-09-18:
@jkrautmacher can you please confirm that ?

jkrautmacher, 2024-09-18:
I would be glad to help but unsure what to confirm. As far as I understood the current theory is that the power supply of the PHY on `nucleo_f767zi` might not be stable enough during init so that initialization fails.

If this is correct my next debugging step would be to connect an oscilloscope to the supply voltage of the PHY and observe it during init. That together with a debug GPIO from the kernel code which toggles right before and after Ethernet init should validate this theory or not. The next step would be to either fix the hardware design or to add a workaround to the Zephyr kernel (longer delays or similar).

I am lacking two things to verify the theory:

- an oscilloscope at home
- detailed schematics of `nucleo_f767zi`

Is there maybe more public information about the board than in UM1974 Rev 10 I overlooked? The oscilloscope situation I could maybe improve but it would likely be way faster if you could do that at ST if this is an option.

jkrautmacher, 2024-11-27:
Since it was easy to test for me I checked how the board / firmware behaves with an external power supply instead of powering it via USB. I moved the jumper on JP3 from U5V to VIN-5V and provided 12 VDC to VIN and GND on CN8.

Result is: `Success rate is 39 % (39 ok and 61 failed)`

So this did not fix it. Updated the initial bug report accordingly.

leonardo-munoz-pragmafw, 2025-04-21:
I tried to reproduce this issue using the same SDK version and both Zephyr versions mentioned. I tested it on the nucleo_f429zi and disco_h745i boards, and consistently obtained a 100% success rate:

`Success rate is 100% (100 OK, 0 failed)`

@jkrautmacher — is this still an issue on your side? Were you able to resolve it?

leonardo-munoz-pragmafw, 2025-04-22:
@jkrautmacher Quick update on my previous comment:

I managed to get a brand new `nucleo_f767zi` board and ran the same tests to try to reproduce the issue. Once again, I got a 100% success rate.

At this point, it seems likely that the issue might have been related to the board hardware itself (e.g., a faulty connector or similar).

jkrautmacher, 2025-04-28:
@leonardo-munoz-pragmafw Thank you very much for testing on a brand new `nucleo_f767zi`. Nice to hear that it works on that one.

Apart from a hardware defect I could also imagine a wrong hardware configuration. Below are pictures of my board from both sides just after it returned with a success rate of 41 % based on Zephyr v4.1.0 (sorry for the questionable focus).

Could you maybe compare that with the jumper settings of the board with 100 % success rate? Maybe I just misconfigured something.

Apart from that: Is it normal that only four of the five checkboxes of the sticker are marked? The `QA` field is not checked which might indicate missing quality assurance tests.

The MAC label is from myself. Apart from that it should basically be like it was shipped.

I wonder about this jumper configuration topic since @marwaiehm-st was able to reproduce it similarly on a `nucleo_f767zi` if I understood [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/77794#issuecomment-2328859627) correctly.

[image]
[image]

jkrautmacher, 2025-05-09:
Thank you for the pictures.

If I see this correctly we can summarize that it works on a fresh `nucleo_f767zi` board in the same hardware configuration to 100 %. Two other boards show this behavior.

While I would be curious what the actual issue is I think it would be quite tedious to find out. Furthermore Zephyr proved on the fresh board that it works in general on this piece of hardware.

If the other participants do not disagree I would suggest to close the issue.

erwango, 2025-05-12:
> If the other participants do not disagree I would suggest to close the issue.

@jkrautmacher I agree. Thanks for the efforts on catching the issue though.

FRASTM, 2025-06-09:
OK for closing

closed 2025-06-09 by FRASTM