zephyrproject-rtos/zephyr #77810: Bluetooth: Controller: ISO: Assert when using BAP broadcast preset 441_1_1 as ISO receiver

Thalley, 2024-08-30:
**Describe the bug**
When using the 441_1_1 preset with Audio Configuration 14 configuration from BAP we hit an assert

```
ASSERTION FAIL [err == ((isoal_status_t) 0x00) || err == ((isoal_status_t) 0x04)]
```
from https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/hci/hci_driver.c#L513-L514

This only happens when using the nRF530 BSIM and not the nRF52840, and only for Audio Configuration where the SDU size is double that of Audio Configuration 12 and 13. 

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout https://github.com/zephyrproject-rtos/zephyr/pull/77746 (if not merged)
2. Build and run https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/audio/test_scripts/cap_broadcast_ac_14.sh
3. See error

**Expected behavior**
Expect the configuration to work on the nRF52 and nRF53 series in babblesim. In either case, it should not assert but return an error if unsupported. 

**Impact**
Unlikely to hit any users, so quite minor, but may prevent us from qualifying BAP. 

**Logs and console output**
```
d_01: @00:00:01.382354  (CPU:0): Received BASE with 1 subgroups from broadcast sink 0x85112a0
d_01: @00:00:01.382354  (CPU:0): Metadata updated
d_01: @00:00:01.382354  (CPU:0): type 2 len 2
d_01: @00:00:01.395604  (CPU:0): [00:00:01.395,599] <dbg> bt_iso: hci_le_big_sync_established: BIG[0] 0x84ddb80 sync established, status 0x00 
d_01: @00:00:01.395604  (CPU:0): [00:00:01.395,599] <dbg> bt_iso: bt_iso_connected: 0x84dd800
d_01: @00:00:01.395604  (CPU:0): [00:00:01.395,599] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x8507480 iso 0x84dd800 connecting -> connected
d_01: @00:00:01.395604  (CPU:0): [00:00:01.395,599] <dbg> bt_bap_broadcast_sink: broadcast_sink_iso_connected: stream 0x8493740
d_01: @00:00:01.395604  (CPU:0): [00:00:01.395,599] <dbg> bt_bap_broadcast_sink: broadcast_sink_set_ep_state: ep 0x8510fc0 id 0x00 qos-configured -> streaming
d_01: @00:00:01.395604  (CPU:0): Stream 0x8493740 started
d_01: @00:00:01.406037  (CPU:1): ASSERTION FAIL [err == ((isoal_status_t) 0x00) || err == ((isoal_status_t) 0x04)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:513
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 494f88070f6bb909389630264e4f89c71072ae53

**Additional context**

[image]
[image]

Thalley, 2025-05-05:
Confirmed to still be an issue

dsseng, 2025-07-21:
I wanted to take a look, but the file got moved: https://github.com/zephyrproject-rtos/zephyr/commit/60e9f9958297fcf9eb55b30b62af247738500ab3#diff-2f19bbc1440ba9d46172b6ec22ed6cfaf18f1dfd81dc0dffd3cfa170557bef31

However, these new tests do not seem to fail on 87917a10874dea20da24f9cd6324d61d5e11ce4b:

```
tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_1_1.sh PASSED
tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_1_2.sh PASSED
tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_1.sh PASSED
tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_2.sh PASSED
```

Thalley, 2025-07-28:
> Edit: `./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_1.sh` now runs the affected test, the test run command there to run.
> old
> 
> the file got moved: [60e9f99#diff-2f19bbc1440ba9d46172b6ec22ed6cfaf18f1dfd81dc0dffd3cfa170557bef31](https://github.com/zephyrproject-rtos/zephyr/commit/60e9f9958297fcf9eb55b30b62af247738500ab3#diff-2f19bbc1440ba9d46172b6ec22ed6cfaf18f1dfd81dc0dffd3cfa170557bef31)
> 
> However, these new tests do not seem to fail on [87917a1](https://github.com/zephyrproject-rtos/zephyr/commit/87917a10874dea20da24f9cd6324d61d5e11ce4b):
> 
> ```
> Attempting to run 32 cases (logging to  /tmp/zephyr/RunResults.xml)
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_16_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_16_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_16_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_16_2_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_24_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_24_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_24_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_24_2_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_32_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_32_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_32_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_32_2_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_2_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_3_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_3_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_4_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_4_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_5_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_5_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_6_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_48_6_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_8_1_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_8_1_2.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_8_2_1.sh PASSED
> ./tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_8_2_2.sh PASSED
> ```

Thanks - You are correct that zephyr/tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_1_1.sh now passes. I tried to run zephyr/tests/bsim/bluetooth/audio/test_scripts/audio_config_tests/cap_ac_14_441_2_1.sh that refer to the same issue, and for that one it still exists, so the issue isn't fixed yet